sarxos/webcam-capture #288: com.github.sarxos.webcam.WebcamException: Cannot execute task && com.github.sarxos.webcam.WebcamException: Cannot start native grabber!

therapia, 2015-01-03:
Hi,

Thank you for your work, but I have some trouble making it work with my project. I just want to display the webcam in a JFrame and when the user clicks a button it takes a picture. All my problems : 
- first of all my webcam is on when i launch eclipse (even if I do not compile the project)
- i have to bypass the locking on my webcam or i just can't use it (Exception in thread "main" com.github.sarxos.webcam.WebcamLockException: Webcam ASUS USB2.0 Webcam 0 has already been locked)
- I tried your example in WebcamPanelExample.java and imported all jars needed and I still get these errors.

[main] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
[main] INFO com.github.sarxos.webcam.WebcamLock - Locking mechanism has been disabled in Webcam ASUS USB2.0 Webcam 0
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam ASUS USB2.0 Webcam 0
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:271)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:232)
	at com.github.sarxos.webcam.WebcamPanel.start(WebcamPanel.java:796)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:743)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:703)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:693)
	at bob.TakePictureExample.main(TakePictureExample.java:19)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:330)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

this stack comes from the same code of your example with just a webcam.getLock().disable();

sarxos, 2015-01-03:
Hi @therapia,

Thank you for the report. In regards to your problems:

> my webcam is on when i launch eclipse (even if I do not compile the project)

I have no idea why this is happening. Are you sure you didn't turn on the webcam before you start eclipse?

> i have to bypass the locking on my webcam or i just can't use it

That means another thread has already locked the webcam and use it. The best for all the users is to not mess with the lock at all. The locking feature has been designed especially for Windows to prevent situations like described in your ticket (exceptions from native grabber).

I would suggest you to carefully investigate why your webcam is open when you turn on the eclipse and what process is locking it (maybe the applet example in the web browser) etc. The lock is just a normal file in a temporary directory (please see the [source code](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/WebcamLock.java) for more details), so the owning process can be easily traced with [filemon](http://technet.microsoft.com/en-us/sysinternals/bb896642.aspx).

therapia, 2015-01-03:
Thank you for being so fast, I guess I found out why my webcam is on, it actually turns on when I open the file written by virtual swing. The GUI preview shows the webcam next to my buttons and I can see myself moving, but when I compile I just get these errors. Maybe you do know how to turn off the preview so it doesn't turn my webcam on before compilation.
I wanted to ask, because I didn't read through all your work, did you work on face recognition and/or comparison in this library?

sarxos, 2015-01-03:
@therapia, the ```WebcamException``` is a runtime exception and thus it cannot be thrown in the compilation time. You must be getting this exception when you run your code, not when you compile it.

Why won't you simply close the other application that is showing preview? If the Swing preview you are talking about is running in the same JVM, then you can simply get the webcam and close it, e.g.:

```java
Webcam.getDefault().close();
```

But if the Swing preview, and you app are running in two different JVMs, then there is no way to close one from the other.

The simplest way to solve your issue is to close previous application which uses your webcam.

closed 2015-01-12 by sarxos

shahnamita, 2015-04-28:
I am just trying to open webcam through webcam-capture-0.3.10-RC6-dist.zip file by copy into my project lib folder in eclipse and try to run on liferay framework but then after webcam light will be on but webcam does not display its area and shows error like as below:
com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:255)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:66)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.lang.Thread.run(Unknown Source)

and 

Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:57)
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:122)
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:38)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:189)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:149)
        at org.apache.jsp.view_jsp._jspService(view_jsp.java:156)
        at org.apache.jasper.runtime.HttpJspBase.service(HttpJspBase.java:70)
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:728)
        at org.apache.jasper.servlet.JspServletWrapper.service(JspServletWrapper.java:432)

Above two type of error shows so pls suggest some regarding solution

My purpose is to open webcam then capture live photo and then set that photo as password so pls suggest me some useful solution..

Reply me as soon as possible

sarxos, 2015-04-28:
Hi @shahnamita,

Have you tried the newest binaries?

https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture/0.3.11-SNAPSHOT/

Please update BridJ to 0.7 as well (bridj-0.7.0.jar).

http://search.maven.org/#artifactdetails|com.nativelibs4java|bridj|0.7.0|bundle

If this won't help you will have to enable debug logs to see more details on what is happening beneath the surface.

https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs

virtualmethod1, 2021-02-27:
I had the same problem the the original poster. Doing some debugging I found that in the TakePictureExample the call to  Webcam.getDefault() was returning a non-functioning webcam object that was experiencing this "Cannot start native grabber" error.  I wrote some code to iterate through all of the detected cameras and print the names of them. I found that the first camera was always called "Webcam 0" and the second camera was my logitech called "Logitech Webcam 500 1".  By setting the camera to use the last camera detected I was able to get the example to work and to take a picture (see below):

`public class TakePictureExample {

	public static void main(String[] args) throws IOException {

		// get default webcam and open it
                List<Webcam> cams = Webcam.getWebcams();
               
                
                System.out.println("counting cams...");
                Webcam webcam = null;
		for(Webcam cam:cams)
                {
                    System.out.println("cam: ");
                   System.out.println(cam.getName());
                   webcam = cam;
                }
                
                //Webcam webcam = Webcam.getWebcamByName("Webcam Logitech Webcam 500 1");
                webcam.setViewSize(new Dimension(640,480));
                
		webcam.open();

		// get image
		BufferedImage image = webcam.getImage();

		// save image to PNG file
		ImageIO.write(image, "PNG", new File("test.png"));
	}
}`