sarxos/webcam-capture #289: EXCEPTION_ACCESS_VIOLATION closing Webcam

cerberous, 2015-01-06:
I am facing this error, when a do fe following steps, on my project see Github's example here: <a href='https://github.com/cerberous/webcam-capture-jafafx' target='_blanck'>webcam-capture-jafafx</a>

1 - Star application (Main.java);
2 - Open Webcam preview by clicking on 'Dialog' Button;
3 - Select WebCam, Stop Webcam, Dispose Webcam and close the dialog;
When doing again, beginning from step 2, the application crashes when clicking on 'Dispose'.
See 'hs_err_pid2144.log' file for log details.

sarxos, 2015-01-11:
Hi @cerberous, I'm sorry to say, but I'm not very fluent in JavaFX and thus it's hard for me to understand what is happening inside this code. However, this kind of exception may be threw in some cases when more then one thread is accessing the same native reference. This should not happen if you are using ```Webcam``` class because it takes care of this.

Also, please do not dispose the webcam, it should be closed instead.

Can you share application [debug logs](https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs)?

cerberous, 2015-01-12:
Hi <a href='https://github.com/sarxos'>@sarxos</a>, I solved it just not calling 'Webcam.shutdown()', for some reason calling it, after closing webcam, it breaks on reopening the webcam dialog! Thanks for support!!

sarxos, 2015-01-12:
@cerberous, indeed, the ```Webcam.shutdown()``` may be safely invoked only when you are completely exiting JVM. Actually, you don't need to call it at all because JVM will dispose allocated ```Webcam``` instances for you so this method is redundant. Just FYI, this method is useful only in case of Applets (due to their weird lifecycle).

closed 2015-01-12 by cerberous

sarxos, 2015-01-12:
I updated examples for JavaFX so people are not confused.