sarxos/webcam-capture #292: [Help]:Working on BoofCV which use Webcam-Capture Library then I get this Error?

pacozaa, 2015-01-11:
## Here is a Code example I'm trying to make it works.
http://boofcv.org/index.php?title=Example_Webcam_Capture
https://github.com/lessthanoptimal/BoofCV/blob/master/integration/WebcamCapture/examples/boofcv/examples/ExampleTrackingKlt.java

The Error message is
===============================================================================
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
OpenJDK 64-Bit Server VM warning: You have loaded library /tmp/BridJExtractedLibraries7328880225100073781/libbridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Exception in thread "main" java.lang.NoSuchMethodError: com.github.sarxos.webcam.Webcam.open()V
	at boofcv.io.webcamcapture.UtilWebcamCapture.openDefault(UtilWebcamCapture.java:38)
	at exampletrackingklt.ExampleTrackingKlt.main(ExampleTrackingKlt.java:31)
Java Result: 1

===============================================================================

Note: My "Line 31" is as Following
30:  // Open a webcam at a resolution close to 640x480
31:  Webcam webcam = UtilWebcamCapture.openDefault(640,480);|

#### More Information:
1. I use Netbeans IDE
2. I Add jar file of all boofcv folder so that have to warry about I miss BoofCV Library.
3. I Add all 3 jar file of webcam-capture following these lists  
   3.1 bridj-0.7-20130703.103049-42.jar  
   3.2 slf4j-api-1.7.2.jar  
   3.3 webcam-capture-0.3.10-RC4.jar  

### *So How Should I fix this problem and make my project run without error Please Help,sir?*
:+1: 
[image]

[image]

sarxos, 2015-01-11:
Hi,

I suppose this may be caused by broken JAR or by the incorrectly weaved binary code (in case you are using such frameworks like AspectJ, Javassist or ASM). Otherwise, if the JAR is not linked at all, you would get ```NoClassDefFound```.

Please download the newest one from Maven Central:

http://search.maven.org/#artifactdetails|com.github.sarxos|webcam-capture|0.3.10|bundle

Did your code work well before?

pacozaa, 2015-01-11:
1. No, it didn't work before it is my first time to use Boofcv.
2. I haven't  used any framework.
3. I create normal Java Application project so I don't have Maven dependency in my project . I just Add JAR file to my project. In additional I never worked with Maven Central before.
Do I miss some thing?
Well I will try your Webcam-Capture without BoofCV but please consider the solution?
Thank you and appriciate :+1:

sarxos, 2015-01-11:
Can you send me the webcam capture JAR you are using? E.g. via dropbox?

sarxos, 2015-01-12:
Hi @pacozaa,

I just tested with [BoofCV v0.18](http://search.maven.org/#artifactdetails|org.boofcv|WebcamCapture|0.18|jar) and it's working perfectly well.

My code:

```java
import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;
import javax.imageio.ImageIO;
import boofcv.io.webcamcapture.UtilWebcamCapture;
import com.github.sarxos.webcam.Webcam;

public class Test {
	public static void main(String[] args) throws IOException {
		Webcam webcam = UtilWebcamCapture.openDefault(640, 480);
		webcam.open();
		BufferedImage image = webcam.getImage();
		ImageIO.write(image, "JPG", new File("test.jpg"));
		webcam.close();
	}
}
```

POM:

```xml
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
	<modelVersion>4.0.0</modelVersion>
	<groupId>bubu</groupId>
	<artifactId>bubu-test</artifactId>
	<version>0.0.1-SNAPSHOT</version>
	<name>Bubu</name>
	<dependencies>
		<dependency>
			<groupId>org.boofcv</groupId>
			<artifactId>WebcamCapture</artifactId>
			<version>0.18</version>
		</dependency>
	</dependencies>
</project>
```

I suppose that JAR you are using is somehow broken. Just download newest and replace the one you were using till now.

pacozaa, 2015-01-12:
Hi @sarxos 
Sorry for late reply and Thank you for your help  :)
I try to figure it out as your above suggestions
1.I try to add the newest JAR from your link to my project and It have the same error.
2.BoofCV I also download the newest from the official website.
3.I try to run you code above and It still got and error.
So here is the question
4. Do I have to add
bridj-0.7-20130703.103049-42.jar
and
slf4j-api-1.7.2.jar
to my project?
because when I add and remove both two JAR I got different error.
when I add both two JAR either your code or my code from example got the same error as my first post
but when I remove both JAR files both two code from you or example gave me the same error
which is

Exception in thread "main" java.lang.NoClassDefFoundError: org/slf4j/LoggerFactory
	at com.github.sarxos.webcam.Webcam.<clinit>(Webcam.java:97)
	at boofcv.io.webcamcapture.UtilWebcamCapture.openDefault(UtilWebcamCapture.java:36)
	at webcamcapturetest.WebcamCaptureTest.main(WebcamCaptureTest.java:17)
Caused by: java.lang.ClassNotFoundException: org.slf4j.LoggerFactory
	at java.net.URLClassLoader$1.run(URLClassLoader.java:366)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
	... 3 more
Java Result: 1

5.Conclusion 
5.1  I'm sure that I use the newest JAR of BoofCV and webcam-capture 
5.2 Do I have the both two JAR file from 4.  ?(I've trid add or non-add then I got different error )
5.3 Please help me to solve the problem. Any question to me for more detail or Any suggestion or Any solution will  appriciate

Thank you,sir.

sarxos, 2015-01-12:
@pacozaa, 

> Do I have to add bridj-0.7-20130703.103049-42.jar and slf4j-api-1.7.2.jar to my project?

Yes, Webcam Capture will not run without these.

> NoClassDefFoundError: org/slf4j/LoggerFactory

This is because you did not add slf4j-api-1.7.2.jar.

> 5.3 Please help me to solve the problem. Any question to me for more detail or Any suggestion or Any solution will appriciate

The exception says that method ```com.github.sarxos.webcam.Webcam.open()V``` does not exist. The ```V``` at the end means that it's ```void``` method, but when you take a look at the Webcam Capture source code you can clearly see it should be ```boolean```, so the character at the end would be ```Z```. The ```open()``` method was a ```void``` once in the past. That was changed in March 2013 and it became ```boolean``` in 0.3.10-RC1, so the last version which used ```void``` was 0.3.9. 
Sooo, maybe you cannot use anything newer than 0.3.9 with that version of BoofCV? In your screen I saw you are using 0.3.10-RC4, which already uses ```boolean``` signature.

pacozaa, 2015-01-13:
Do you have a link for download 0.3.9 ?

hepin1989, 2015-01-13:
Hi guy,you need mvnrepository.com,and you need learn maven or sbt or gradle etc.
thanks @sarxos guide me about maven

sarxos, 2015-01-13:
Thanks @hepin1989 :) Indeed, all you need is to find JAR in the Maven Central Repository or use Maven in your project.

There are several sites where you can search:

* [mvnrepository.com](http://mvnrepository.com) (imho, the most user friendly),
* [search.maven.org](http://search.maven.org/),
* etc.

Instead of downloading JRAs manually you can create "Maven Project" in Netbeans (or Eclipse, or IntelliJ) and add dependency description in POM file (some IDE has some fancy GUI for that, but I prefer XML), and Maven will download all necessary dependencies and build your project. See my previous post where I posted my POM. I didn't have to download anything - neither BuffCV JARs nor Webcam Capture JARs - Maven simply runs and do whole the hard job for you. I strongly encourage everyone to learn using Maven (or Gradle, or Ant+Ivy, or sbt - all these uses Maven artifacts, just the config files are different). If you have more then a few JARs in your classpath then you have to start using it, or your work will change into nightmare when you need to upgrade some dependencies to newer one (I manage projects with thousands JARs and Maven really saves my ass).

For the manual download just type in the search box "webcam-capture", and it will find it. Repository search instruction:

[image]

[image]

[image]

[image]

pacozaa, 2015-01-13:
### Hello @sarxos @hepin1989  I think version 0.3.9 is work for me.
1.I try to run @sarxos 's code above it take a picture from webcam. but I got red message occur which I'm not sure about it.

=============================================================================
run:
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
OpenJDK 64-Bit Server VM warning: You have loaded library /tmp/BridJExtractedLibraries2590020022099403485/libbridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Jan 13, 2015 7:16:27 PM org.bridj.BridJ log
INFO: Not enough args for null OpenIMAJGrabber.startSession(int, int, double)
BUILD SUCCESSFUL (total time: 4 seconds)

==============================================================================
[image]

2.I try to run this code( https://github.com/lessthanoptimal/BoofCV/blob/master/integration/WebcamCapture/examples/boofcv/examples/ExampleTrackingKlt.java ) and everything work fine. but I got the red message and I'm not sure about it.

==========================================================================
run:
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
OpenJDK 64-Bit Server VM warning: You have loaded library /tmp/BridJExtractedLibraries8200512403949337514/libbridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Jan 13, 2015 7:25:10 PM org.bridj.BridJ log
INFO: Not enough args for null OpenIMAJGrabber.startSession(int, int, double)
BUILD STOPPED (total time: 1 minute 11 seconds)

=============================================================================
[image]

3.Please explain above two red message if you can.
Thank you and appriciate for your help until my project works !

sarxos, 2015-01-14:
Hi @pacozaa, it's great to hear everything works well. In regards to the red warning messages, please take a look at #55 where these are described in details. To cut the story short - you can ignore these - both are harmless.

closed 2015-01-14 by sarxos