sarxos/webcam-capture #293: Why still false and can't run??

SPRL08, 2015-01-11:
import java.awt.Dimension;
import java.awt.FlowLayout;
import java.awt.image.BufferedImage;
import java.util.concurrent.Executor;
import java.util.concurrent.Executors;
import java.util.concurrent.ThreadFactory;

import javax.swing.JFrame;
import javax.swing.JTextArea;

import com.google.zxing.BinaryBitmap;
import com.google.zxing.LuminanceSource;
import com.google.zxing.MultiFormatReader;
import com.google.zxing.NotFoundException;
import com.google.zxing.Result;
import com.google.zxing.client.j2se.BufferedImageLuminanceSource;
import com.google.zxing.common.HybridBinarizer;

public class WebcamQRCodeExample extends JFrame implements Runnable, ThreadFactory {

	private static final long serialVersionUID = 6441489157408381878L;

	private final Executor executor = Executors.newSingleThreadExecutor(this);

	private Webcam webcam = null;      *false
	private WebcamPanel panel = null;      *false
	private JTextArea textarea = null;

	public WebcamQRCodeExample() {
		super();

		setLayout(new FlowLayout());
		setTitle("Read QR / Bar Code With Webcam");
		setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);

		Dimension size = WebcamResolution.QVGA.getSize();     *false

		webcam = Webcam.getWebcams().get(0);      *false
		webcam.setViewSize(size);

		panel = new WebcamPanel(webcam);        *false
		panel.setPreferredSize(size);

		textarea = new JTextArea();
		textarea.setEditable(false);
		textarea.setPreferredSize(size);

		add(panel);
		add(textarea);

		pack();
		setVisible(true);

		executor.execute(this);
	}

	@Override
	public void run() {

		do {
			try {
				Thread.sleep(100);
			} catch (InterruptedException e) {
				e.printStackTrace();
			}

			Result result = null;
			BufferedImage image = null;

			if (webcam.isOpen()) {

				if ((image = webcam.getImage()) == null) {
					continue;
				}

				LuminanceSource source = new BufferedImageLuminanceSource(image);
				BinaryBitmap bitmap = new BinaryBitmap(new HybridBinarizer(source));

				try {
					result = new MultiFormatReader().decode(bitmap);       *false
				} catch (NotFoundException e) {
					// fall thru, it means there is no QR code in image
				}
			}

			if (result != null) {
				textarea.setText(result.getText());
			}

		} while (true);
	}

	@Override
	public Thread newThread(Runnable r) {
		Thread t = new Thread(r, "example-runner");
		t.setDaemon(true);
		return t;
	}

	public static void main(String[] args) {
		new WebcamQRCodeExample();       *false
	}
}

SPRL08, 2015-01-11:
[image]

SPRL08, 2015-01-11:
[image]

SPRL08, 2015-01-11:
[image]

SPRL08, 2015-01-11:
Help me please,,,
I confused with this problem,,

sarxos, 2015-01-11:
You haven't add JARs to the classpath. To use any library you need to add them to the classpath before the calasses from these JARs can be used in your application. In Eclipse this can be done by right clicking on JAR and choosing "Add to build path". I do not use Netbeans so I cannot answer how to do that in this IDE. You need to figure it out on your own or simply [Google it](https://www.google.pl/search?client=ubuntu&channel=fs&q=netbeans+add+jar+to+classpath&ie=utf-8&oe=utf-8&gfe_rd=cr&ei=TQizVKT-AY-E5AaomIHwAg) - there are dozens of answers on the Internet.

SPRL08, 2015-01-12:
Thanks sarxos,,I'll trying with eclipse

sarxos, 2015-01-12:
Great. Please also note that this is problem with your IDE settings, not with the Webcam Capture library. I'm closing this ticket as invalid.

closed 2015-01-12 by sarxos

SPRL08, 2015-01-12:
I try with eclipse and add the jar to the classpath,,,but still can't run,,, :(

SPRL08, 2015-01-12:
public static void main(String[] args) {
        
            new ManyCamerasAtOnceExample();     *new instace ignored
        }

    }

sarxos, 2015-01-12:
@SPRL08 

> but still can't run

Can you be more specific? What is the error text saying? Any exceptions?

> new instace ignored

Java does not ignore new operator. I would rather suppose that your code was not compiled at all.

SPRL08, 2015-01-12:
java.lang.VerifyError: Cannot inherit from final class
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:631)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:615)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)
Could not find the main class: WebcamQRCodeExample.  Program will exit.
Exception in thread "main" Java Result: 1

sarxos, 2015-01-12:
There is your problem:

```plain
Cannot inherit from final class
```

Can you share you source code? Some class extends other one which is marked as final. This causes your code to fail.

SPRL08, 2015-01-12:
import java.awt.Dimension;
import java.awt.FlowLayout;
import java.awt.image.BufferedImage;
import java.util.concurrent.Executor;
import java.util.concurrent.Executors;
import java.util.concurrent.ThreadFactory;

import javax.swing.JFrame;
import javax.swing.JTextArea;

import com.google.zxing.BinaryBitmap;
import com.google.zxing.LuminanceSource;
import com.google.zxing.MultiFormatReader;
import com.google.zxing.NotFoundException;
import com.google.zxing.Result;
import com.google.zxing.client.j2se.BufferedImageLuminanceSource;
import com.google.zxing.common.HybridBinarizer;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;

public class WebcamQRCodeExample extends JFrame implements Runnable, ThreadFactory {

	private static final long serialVersionUID = 6441489157408381878L;

	private final Executor executor = Executors.newSingleThreadExecutor(this);

	private Webcam webcam = null;
	private WebcamPanel panel = null;
	private JTextArea textarea = null;

	public WebcamQRCodeExample() {
		super();

		setLayout(new FlowLayout());
		setTitle("Read QR / Bar Code With Webcam");
		setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);

		Dimension size = WebcamResolution.QVGA.getSize();

		webcam = Webcam.getWebcams().get(0);
		webcam.setViewSize(size);

		panel = new WebcamPanel(webcam);
		panel.setPreferredSize(size);

		textarea = new JTextArea();
		textarea.setEditable(false);
		textarea.setPreferredSize(size);

		add(panel);
		add(textarea);

		pack();
		setVisible(true);

		executor.execute(this);
	}

	@Override
	public void run() {

		do {
			try {
				Thread.sleep(100);
			} catch (InterruptedException e) {
				e.printStackTrace();
			}

			Result result = null;
			BufferedImage image = null;

			if (webcam.isOpen()) {

				if ((image = webcam.getImage()) == null) {
					continue;
				}

				LuminanceSource source = new BufferedImageLuminanceSource(image);
				BinaryBitmap bitmap = new BinaryBitmap(new HybridBinarizer(source));

				try {
					result = new MultiFormatReader().decode(bitmap);
				} catch (NotFoundException e) {
					// fall thru, it means there is no QR code in image
				}
			}

			if (result != null) {
				textarea.setText(result.getText());
			}

		} while (true);
	}

	@Override
	public Thread newThread(Runnable r) {
		Thread t = new Thread(r, "example-runner");
		t.setDaemon(true);
		return t;
	}

	public static void main(String[] args) {
        new WebcamQRCodeExample();
	}
}

sarxos, 2015-01-12:
I'm running your code and it works well. What are your dependencies (what JARs)? The proper ones for this example to run are:

* webcam-capture-0.3.10.jar (Webcam Capture API)
* bridj-0.6.2.jar (BridJ)
* slf4j-api-1.7.2.jar (SLF4J)
* javase-2.1.jar (zxing)
* core-2.1.jar (zxing)

SPRL08, 2015-01-12:
where i can download 
bridj-0.6.2.jar (BridJ)
slf4j-api-1.7.2.jar (SLF4J)

sarxos, 2015-01-12:
From the brief look at your screen shoots I see that you are using some archaic zxing JARs (1.7 instead of 2.1).

All these JARs can be found in [Maven Central](http://search.maven.org/):

* http://search.maven.org/#artifactdetails|com.github.sarxos|webcam-capture|0.3.10|bundle
* http://search.maven.org/#artifactdetails|com.nativelibs4java|bridj|0.6.2|bundle
* http://search.maven.org/#artifactdetails|org.slf4j|slf4j-api|1.7.10|jar
* http://search.maven.org/#artifactdetails|com.google.zxing|javase|2.1|jar
* http://search.maven.org/#artifactdetails|com.google.zxing|core|2.1|jar

You should learn more about the Maven so you can avoid such problems in the future because it has been designed to manage JAR dependencies for you (people make mistakes, properly used Maven does not). There are [dozens](https://www.google.pl/search?client=ubuntu&channel=fs&q=maven+dependency+management+tutorial&ie=utf-8&oe=utf-8&gfe_rd=cr&ei=BNWzVJC7LJDCVL_TguAJ#channel=fs&q=maven+tutorial) of tutorials on the internet.

SPRL08, 2015-01-12:
Thanks sarxos,,,its Succesed   yeaahhhh

SPRL08, 2015-12-22:
Hai sarxos,,,i have question again for you
How i build if i want result like this
[image]

SPRL08, 2015-12-22:
Hai sarxos,,how to build if i want the result like this
[image]

SPRL08, 2015-12-22:
thanks before

sarxos, 2015-12-23:
@SPRL08, please see my comments in #424

meetravi1334, 2016-02-11:
Hi sarxos,

Thanks for the detailed information. I am having a different problem, while running this application. When I run this in the standalone mode, I get through easily and I am able to run the webcam and capture QRCode text. But the moment, I integrate this in spring based application, I get the error "Cannot Inherit from Final Class". I am using the same Jar files as highlighted by you above. Any pointers on this would be highly appreciated. 

I am attaching my source code alongwith the jar files :-
[QRCode.zip](https://github.com/sarxos/webcam-capture/files/125944/QRCode.zip)

sarxos, 2016-02-11:
Hi @meetravi1334,

The code you provided is useless because it does not compile (BasketSearchWidgetRendererImpl).

On which class from Webcam Capture API you have this error?

meetravi1334, 2016-02-12:
Greetings sarxos,

I got the problem. I was trying to integrate the code with a ZK based application and was continuously getting cannot inherit from Final class error. The problem was with the incompatible jars present in the source code. I am able to move ahead on this issue by synching the same jars across the applicaiton. Thanks a lot for the useful article and for replying timely. Thanks.

sarxos, 2016-02-12:
@meetravi1334, are you maybe using Maven to manage JAR dependencies? In all these huge applications I was working on (more than 100+ JARs) Maven always was a lifesaver.

meetravi1334, 2016-02-12:
I wish to use Maven but unfortunately not. That's why struggled for 2 days but finally I am through. Thanks to you for the help :-)

sarxos, 2016-02-12:
Ok, take care!