sarxos/webcam-capture #294: NullPointerException 	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImage(IpCamDevice.java:290)

10sw33, 2015-01-16:
here is my class where i want to connect with my mobile and i have set mobile ip address and port in moveable.xml. 
public class DetectMotionFromIpCamerasExample {
	static {
		Webcam.setDriver(new IpCamDriver(new IpCamStorage("src/moveable.xml")));
	}
	public static void main(String[] args) throws MalformedURLException {
		JFrame f = new JFrame("Detect Motion Multiple Cameras");
		f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		f.setLayout(new GridLayout(0, 3, 1, 1));
		WebcamMotionListener listener = new WebcamMotionListener() {
			@Override
			public void motionDetected(WebcamMotionEvent wme) {
				synchronized (this) {
					System.out.println("motion detected ---------------- ");
					System.out.println("webcam: " + ((WebcamMotionDetector) wme.getSource()).getWebcam());
					System.out.println("diff area: " + wme.getArea() + "%");
					System.out.println("cog: [" + wme.getCog().getX() + "," + wme.getCog().getY() + "]");
				}
			}
		};
		final List<Webcam> webcams = new ArrayList<Webcam>();
		final List<WebcamPanel> panels = new ArrayList<WebcamPanel>();
		final List<WebcamMotionDetector> detectors = new ArrayList<WebcamMotionDetector>();
		for (Webcam webcam : Webcam.getWebcams()) {
			webcams.add(webcam);
			WebcamPanel panel = new WebcamPanel(webcam, new Dimension(256, 144), false);
			panel.setFitArea(true);
			panel.setFPSLimited(true);
			panel.setFPSLimit(0.5); // 0.5 FPS = 1 frame per 2 seconds
			panel.setBorder(BorderFactory.createEmptyBorder());
			f.add(panel);
			panels.add(panel);
			WebcamMotionDetector detector = new WebcamMotionDetector(webcam);
			detector.addMotionListener(listener);
			detector.setInterval(100); // one motion check per 2 seconds
			detectors.add(detector);
		}
		f.pack();
		f.setVisible(true);
		for (int i = 0; i < webcams.size(); i++) {
			final int x = i;
			Thread t = new Thread() {
				@Override
				public void run() {
					webcams.get(x).open(); // open in asynchronous mode
					panels.get(x).start();
					detectors.get(x).start();
				}
			};
			t.setDaemon(true);
			t.start();
		}
	}
}

moveable.xml is here 
<?xml version="1.0" encoding="UTF-8" ?>
<storage>

<ipcam name="my" url="http://192.168.2.33:8090" mode="push" >
</ipcam>
</storage>

P.S : i have installed an app on my mobile called IP Cam which allows me to use it as webcam through its ip:port, thats same ip port i m giving in this xml. is this right way or i m messing something, actually i want to connect any camera through ip whether it is mobile or SLRs..

sarxos, 2015-01-16:
Hi, can you post the stack trace?

10sw33, 2015-01-16:
https://cloud.githubusercontent.com/assets/10362378/5772140/3d430c72-9d6e-11e4-9b85-4716d416f3ac.png this is same as i showed you on #216

sarxos, 2015-01-16:
@10sw33 Ok, but this is not the problem you described in this ticket. In the title you wrote it is about ```NullPointerException``` in ```IpCamDevice.java:290```. The #216 is about using invalid IP camera authorization credentials. And from the screen shoot you posted in the other ticket is seems like your URL is wrong.

10sw33, 2015-01-19:
yes you were right, thanks i changed the URL and it worked :100:  you made my day :)

sarxos, 2015-01-19:
Ok, I'm closing this ticket. Take care!

closed 2015-01-19 by sarxos