sarxos/webcam-capture #296: Can't add Webcam panel to existing jPanel.

Qbaloz, 2015-01-17:
Hello,

I'm using NetBeans and I made basic layout with jPanel in the middle of it. Now I would like to add Webcam in this jPanel.

    Webcam webcam = Webcam.getDefault();
    webcam.setViewSize(WebcamResolution.VGA.getSize());

    WebcamPanel panel = new WebcamPanel(webcam);
    panel.setFPSDisplayed(true);
    panel.setDisplayDebugInfo(true);
    panel.setImageSizeDisplayed(true);
    panel.setMirrored(true);

    jPanel5.add(panel);
    jPanel5.setVisible(true);
    jPanel5.revalidate();
    jPanel5.repaint();

This code is in my class constructor. I would like to run this at startup. Just after other netbeans componenst are loaded. When I add example to my program it's working but my main layout is loaded and webcam screen in another window. I would like to get it in the same window.

I tried to do it with other componenst like jInternalFrame, even jLabel and I just have no idea how to add webcam to existing element, not to new one.

Thanks for any help, Qbaloz

sarxos, 2015-01-17:
Hi @Qbaloz ,

Can you please prepare small example to demonstrate the problem? I'm using this code and everything works well:

```java
import javax.swing.JFrame;
import javax.swing.JPanel;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;

public class TestQbaloz {

	public static void main(String[] args) throws InterruptedException {

		Webcam webcam = Webcam.getDefault();
		webcam.setViewSize(WebcamResolution.VGA.getSize());

		WebcamPanel panel = new WebcamPanel(webcam);
		panel.setFPSDisplayed(true);
		panel.setDisplayDebugInfo(true);
		panel.setImageSizeDisplayed(true);
		panel.setMirrored(true);

		JPanel jPanel5 = new JPanel();
		jPanel5.add(panel);
		jPanel5.setVisible(true);
		jPanel5.revalidate();
		jPanel5.repaint();

		JFrame window = new JFrame("Test webcam panel");
		window.add(jPanel5);
		window.setResizable(true);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}
```

The result:

[image]

Qbaloz, 2015-01-17:
Hi @sarxos , thanks for fast response. I will describe my problem more precisely.

Here's my main window:
[image]

In the middle of my layout is JTabbedPane with two tabs - camera and map. Camera Tab is my JPanel5. I would like to add camera screen to this tab. Your example is working but still it's making new window.

I removed this:

JPanel jPanel5 = new JPanel();

Because my panel already exist. I should also remove creating new JFrame, because my Frame already exist too, but then I don't know how to add this to my existing jPanel in Tab.

As your code is working I think problem is with my lack of knowledge about NetBeans but maybe you will be able to help me.

sarxos, 2015-01-18:
@Qbaloz I guess what you were missing is:

```java
jPanelWebcam.getParent().revalidate();
```

Here is the code. I created ```JTabbedPane``` with two empty tabs, each with a single ```JPanel``` inside. The button will cause ```WebcamPanel```ro be created and added into one of the ```JPanel``` instances.

```java
import java.awt.Dimension;
import java.awt.FlowLayout;
import java.awt.event.ActionEvent;

import javax.swing.AbstractAction;
import javax.swing.JButton;
import javax.swing.JFrame;
import javax.swing.JPanel;
import javax.swing.JTabbedPane;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;

public class TestQbaloz {

	public static void main(String[] args) throws InterruptedException {

		final JPanel jPanelWebcam = new JPanel();
		final JPanel jPanelMap = new JPanel();

		JTabbedPane tabbedPane = new JTabbedPane();
		tabbedPane.addTab("Webcam", jPanelWebcam);
		tabbedPane.addTab("Map", jPanelMap);

		final JButton button = new JButton(new AbstractAction("dodaj") {

			private static final long serialVersionUID = 1L;

			@Override
			public void actionPerformed(ActionEvent e) {

				Webcam webcam = Webcam.getDefault();
				webcam.setViewSize(WebcamResolution.VGA.getSize());

				WebcamPanel webcamPanel = new WebcamPanel(webcam);
				webcamPanel.setFPSDisplayed(true);
				webcamPanel.setDisplayDebugInfo(true);
				webcamPanel.setImageSizeDisplayed(true);
				webcamPanel.setMirrored(true);

				jPanelWebcam.add(webcamPanel);
				jPanelWebcam.getParent().revalidate();

			}
		});

		JFrame window = new JFrame("Test webcam panel");
		window.setLayout(new FlowLayout());
		window.setPreferredSize(new Dimension(800, 600));
		window.add(tabbedPane);
		window.add(button);
		window.setResizable(true);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}
```

The effects (please ignore the weird layout on the first image - I had no time to layout these components properly and used ```FlowLayout``` to do the job, but as you can see - it has some disadvantages):

[image]

[image]

I hope this helps.

sarxos, 2015-01-18:
Ok, I just tested without ```getParent()``` and everything still works well :( This is surprising. To be honest, I don't know what may be wrong in your code.

Qbaloz, 2015-01-18:
Hi @sarxos 
I have a solution. Your code gave me good idea. Instead of making Tabs using NetBeans window designer I made it with code.

I made empty JTabbedPane in Netbeans and then add this to code:

        final JPanel jPanelCamera = new JPanel();
        final JPanel jPanelMap = new JPanel();
        
        jTabbedPane1.addTab("Camera", jPanelCamera);
        jTabbedPane1.addTab("Map", jPanelMap);
        
        Webcam webcam = Webcam.getDefault();
        webcam.setViewSize(WebcamResolution.VGA.getSize());

        WebcamPanel webcamPanel = new WebcamPanel(webcam);
        webcamPanel.setFPSDisplayed(true);
        webcamPanel.setDisplayDebugInfo(true);
        webcamPanel.setImageSizeDisplayed(true);
        webcamPanel.setMirrored(true);

        jPanelCamera.add(webcamPanel);
        jPanelCamera.getParent().revalidate();
        
        System.out.println("Camera OK");

I don't know why there is a difference between adding tabs in NetBeans and adding them via code, but thats good it's working. Thank you for your help :)

sarxos, 2015-01-18:
Great, I'm glad it's working :)

closed 2015-01-18 by sarxos

marcohp100, 2016-02-03:
Hi @sarxos ! How are you ?

I got to add WebcamPanel to JPanel using your WebcamViewerExample.java example, but it dosen't work if I try to add a JTextField and WebcamPanel in the same JPanel, like the code snippet shown below:

```
Webcam.addDiscoveryListener(this);

setTitle("Java Webcam Capture POC");
setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
setLayout(new BorderLayout());

picker = new WebcamPicker();
picker.addItemListener(this);

webcam = picker.getSelectedWebcam();

if (webcam == null) {
	System.out.println("No webcams found...");
        System.exit(1);
}

webcam.setViewSize(WebcamResolution.VGA.getSize());

panel = new WebcamPanel(webcam, false);
panel.setFPSDisplayed(true);
		
JPanel p = new JPanel();
p.add(picker);

add(p, BorderLayout.NORTH);

add(panel, BorderLayout.CENTER);

**add(new JTextField(), BorderLayout.SOUTH);**

pack();
setVisible(true);

Thread t = new Thread() {
	@Override
	public void run() {
		panel.start();
	}
};
t.setName("example-starter");
t.setDaemon(true);
t.start();

```

If I add a JButton in the JPanel, instead JTextField, it works successfully. The problem is JTextField only. Do you know why this?

The printed error is 
"com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice updateFrameBuffer
GRAVE: Timeout when requesting image!"

Thank you !

sarxos, 2016-02-05:
Hi @marcohp100,

I think it's because your ```JTextField``` is empty, so even if it's added into parent it's rendered as component with zero size. Swing is tricky sometimes :)

idontwantosleep, 2016-06-25:
sirr im using java netbeans and i drag and drop the components i copy your idea above and your qr scanner , the purpose of my program  is to read qr code through camera, for some reason the camera will not open if i put the codes for reading the qr code and if i didnt it will run smoothly, i dont really know what to do, your my last hope this is my code  :

//--------put the camera in jpanel
           Webcam webcam = Webcam.getDefault();
           webcam.setViewSize(WebcamResolution.VGA.getSize());
           WebcamPanel panel = new WebcamPanel(webcam);
         
           jPanel1.add(panel);
           jPanel1.getParent().revalidate();
           
           //------READ qr/barcode---------
                do {
			try {
				Thread.sleep(100);
			} catch (InterruptedException e) {
				e.printStackTrace();
			}

			Result result = null;
			BufferedImage image = null;

			if (webcam.isOpen()) {

				if ((image = webcam.getImage()) == null) {
					continue;
				}

				LuminanceSource source = new BufferedImageLuminanceSource(image);
				BinaryBitmap bitmap = new BinaryBitmap(new HybridBinarizer(source));

				try {
					result = new MultiFormatReader().decode(bitmap);
				} catch (NotFoundException e) {
					// fall thru, it means there is no QR code in image
				}
			}

			if (result != null) {
				jTextArea1.setText(result.getText());
			}

		} while (true);

riddyndoma, 2018-07-11:
Hi @sarxos  i have problem when i need to add webcamPanel to a JPanel created via Netbeans in drag and drop.

        Webcam webcam = Webcam.getDefault();
        WebcamPanel webcamPanel;
        webcam.setViewSize(WebcamResolution.VGA.getSize());
        webcamPanel = new WebcamPanel(webcam);
        webcamPanel.setFPSDisplayed(true);
        webcamPanel.setDisplayDebugInfo(true);
        webcamPanel.setImageSizeDisplayed(true);
        webcamPanel.setMirrored(true);
        
        panneau.add(webcamPanel);
        panneau.getParent().revalidate();

Panneau is a panel created via netbeans in drag&drop.

lilblazekonceptz, 2019-01-30:
Hi @sarxos is it possible to display webcamPanel on JPanel designed with SWING Builder on Netbeans?
because am having problem displaying it. Thanks

sarxos, 2019-01-30:
Hi @lilblazekonceptz, 

There should be no issue when adding `WebcamPanel` to other `JPanel`. If it's not working in your case it means that the `JPanel` added from Netbeans builder have some fancy _something_ set which differs it from a normal `JPanel` (I guess, e.g. special layaout manager, preferred size, bounds, or something else).

I tested this idea of embedding `WebcamPanel` inside another `JPanel` and it works well:

[image]