sarxos/webcam-capture #298: Example not available for IP cameras

sidneyhaddad, 2015-01-22:
Hi there!! Congratulations about your work. I download it and tested, it is really cool and works like a chain! 

Problem is that I couldn't find the IP cameras example on your code and also on your website it is redirecting to 404 page.

Can you please check that.

txs.
Sidney Haddad
Myvillage CEO - Brazil

sarxos, 2015-01-22:
Hi @sidneyhaddad, 

Thank you for the report.

Which link is redirecting to 404? The examples for IP camera are [here](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam/src/examples/java)

sidneyhaddad, 2015-01-22:
Txs for  your response! I will test it and let you know if everything is ok!

In your website: http://webcam-capture.sarxos.pl/ <http://webcam-capture.sarxos.pl/>

Click on More Examples and click on IP example.

it is redirecting to: 

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/webcam-capture-driver-ipcam/src/examples/java/JpegDasdingStudioExample.java <https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/webcam-capture-driver-ipcam/src/examples/java/JpegDasdingStudioExample.java>

which does not exist.

We are building a javafx system to be used in Brazil, it is going to be used in a monitored central place to view cams on building. I hope I can count on you and on your tool. If I manage to make it work and I get to sell it I will donate you cause of the good work and tool!

txs for your support!

sarxos, 2015-01-22:
Ok, thanks @sidneyhaddad. I will fix these links soon :)

commit 33468845 (2015-01-23): Update incorrect links, fixes #298 [ci skip]
closed 2015-01-23 by sarxos