sarxos/webcam-capture #299: Failed to initialize BridJ

amroamer, 2015-01-28:
Dear Sarxos 

I am trying to use the webcam-capture Applet Example , i am following your example : 

first i am writing this applet class like the following : 

    public class WebcamCapture extends JApplet implements ActionListener {
    
    

    private Webcam webcam = null;
    private WebcamPanel webcamPanel = null;
    private Dimension size = WebcamResolution.VGA.getSize();
    private JButton takePictureButton = null;

    private Dimension[] nonStandardResolutions = new Dimension[] {
        WebcamResolution.PAL.getSize(), WebcamResolution.HD720.getSize(), new Dimension(2000, 1000),
        new Dimension(1000, 500),
    };

    public WebcamCapture() {
        super();
    }

    @Override
    public void start() {
        //        WebcamSecurityManager securityManager = new WebcamSecurityManager();
        //        System.setSecurityManager(securityManager);
        super.start();
        System.out.println("WebcamApplet ======> start()");

        setSize(WebcamResolution.HD720.getSize());
        webcam =  Webcam.getDefault();

  

     

        webcam.setCustomViewSizes(nonStandardResolutions);
        webcam.setViewSize(WebcamResolution.HD720.getSize());

        webcamPanel = new WebcamPanel(webcam, false);
        webcamPanel.setFPSDisplayed(true);
        webcamPanel.setSize(WebcamResolution.HD720.getSize());

        takePictureButton = new JButton("Take Picture");
        takePictureButton.addActionListener(this);
        webcamPanel.add(takePictureButton);

        add(webcamPanel);

        if (webcam.isOpen()) {
            webcam.close();
        }

        int i = 0;
        do {
            if (webcam.getLock().isLocked()) {
                System.out.println("Waiting for lock to be released " + i);
                try {
                    Thread.sleep(2000);
                } catch (InterruptedException e1) {
                    return;
                }
            } else {
                break;
            }
        } while (i++ < 3);

        webcam.open();
        webcamPanel.start();

    }

    @Override
    public void destroy() {

        System.out.println("WebcamApplet ======> destroy() ");
        webcam.close();
        System.out.println("WebcamApplet ======> destroyed ");
    }

    @Override
    public void init() {
     System.out.println("WebcamApplet ======> init()");
    }

    @Override
    public void stop() {

        System.out.println("WebcamApplet ======> stop()");
        webcam.close();
        System.out.println("WebcamApplet ======> Stopped");
    }

    @Override
    public void actionPerformed(ActionEvent e) {
        if (e.getSource() == takePictureButton) {
            System.out.println("TakePicture Button actionPerformend");
        }
    }
    } 

and i signed my applet using the jarsigner .
and as you said in the tutorial grant the permissions required so in my java.policy i added this : 

permission java.security.AllPermission;

and i placed my applet tag inside my JSF page like the following : 

 	<?xml version='1.0' encoding='UTF-8'?>
	<!DOCTYPE html>
	<f:view xmlns:f="http://java.sun.com/jsf/core" xmlns:af="http://xmlns.oracle.com/adf/faces/rich">
	<af:document title="testPage.jsf" id="d1">
	<af:form id="f1">
	<APPLET code="clientapplet.WebcamCapture.class" id="cameraApp"
				    name="webcameraApplet" height="100" width="100"
				    archive="/applet/webcamapplet.jar,/applet/webcam-capture-0.3.11-20141227.105512-4.jar,/applet/bridj-0.6.2.jar,/applet/slf4j-api-1.7.10.jar,/applet/slf4j-simple-1.7.10.jar">
	</APPLET>

	</af:form>
	</af:document>
	</f:view>

and then tried to run my applet and i got the following stacktrace : 

	java.lang.RuntimeException: Failed to initialize BridJ
		at org.bridj.Platform.initLibrary(Platform.java:370)
		at org.bridj.Platform.<clinit>(Platform.java:175)
		at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
		at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
		at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
		at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
		at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
		at java.lang.Thread.run(Unknown Source)
	Caused by: java.lang.UnsatisfiedLinkError: C:\Users\user\AppData\Local\Temp\BridJExtractedLibraries7384257206371869727\bridj.dll: %1 is not a valid Win32 application
		at java.lang.ClassLoader$NativeLibrary.load(Native Method)
		at java.lang.ClassLoader.loadLibrary0(Unknown Source)
		at java.lang.ClassLoader.loadLibrary(Unknown Source)
		at java.lang.Runtime.load0(Unknown Source)
		at java.lang.System.load(Unknown Source)
		at org.bridj.Platform.initLibrary(Platform.java:347)
		... 7 more
	com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
		at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
		at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:826)
		at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:889)
		at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:866)
		at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:844)
		at clientapplet.WebcamCapture.start(WebcamCapture.java:54)
		at com.sun.deploy.uitoolkit.impl.awt.AWTAppletAdapter.start(Unknown Source)
		at sun.plugin2.applet.Plugin2Manager$AppletExecutionRunnable.run(Unknown Source)
		at java.lang.Thread.run(Unknown Source)
	Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
		at java.util.concurrent.FutureTask.report(Unknown Source)
		at java.util.concurrent.FutureTask.get(Unknown Source)
		at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
		... 8 more
	Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
		at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
		at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
		at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
		at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:46)
		at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:118)
		at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
		at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
		at java.util.concurrent.FutureTask.run(Unknown Source)
		at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
		at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
		... 1 more
	Caused by: java.lang.UnsatisfiedLinkError: org.bridj.Platform.sizeOf_ptrdiff_t()I
		at org.bridj.Platform.sizeOf_ptrdiff_t(Native Method)
		at org.bridj.Platform.<clinit>(Platform.java:179)
		at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
		at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
		at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
		... 3 more
	Ignored exception: com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task

i dont know whats wrong , or what should i do , i tried every possible solution and tried to go all over the issues that are already solved , but nothing works for me .
i have e

sarxos, 2015-01-28:
Hi @amroamer,

Is your Java 7 32- or 64-bit?

Can you please tell me what JARs are you using? The versions are very important.

amroamer, 2015-01-28:
Hello @sarxos  

as i told you my jre is jre1.8.0_31 64bit   : 
JAVA_VERSION="1.8.0_31"
OS_NAME="Windows"
OS_VERSION="5.1"
OS_ARCH="i586"

in my java control panel , java tab . i am choosing jre1.8.0_31 . 

JDK  64bit   that is associated with the Jdeveloper is  :
JAVA_VERSION="1.7.0"
OS_NAME="Windows"
OS_VERSION="5.2"
OS_ARCH="amd64"

the jar files that i am using are the following : 
webcam-capture-0.3.11-20141227.105512-4 
bridj-0.6.2
slf4j-api-1.7.10
slf4j-simple-1.7.10

and i have also tried with the following jars as well : 
bridj-0.7-20141231.010908-73
slf4j-nop-1.7.10

Thank you

sarxos, 2015-01-28:
Is this happening when you write normal application, not the applet? Use the following sample to test:

```java
import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;
import javax.imageio.ImageIO;
import com.github.sarxos.webcam.Webcam;

public class TakePictureExample {
  public static void main(String[] args) throws IOException {
    Webcam webcam = Webcam.getDefault();
    webcam.open();
    ImageIO.write(webcam.getImage(), "PNG", new File("test.png"));
  }
}
```

Please enable debug logging by following the steps described here (all 6): https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs

Don't forget to remove slf4j-nop and slf4j-simple after you add logback because you may get no logs at all.

When you have debug logs ready, please attach them here. We can then verify what bridj DLL is loaded and/or clarify what may be wrong.

amroamer, 2015-01-28:
Hello @sarxos  

as i told you i am developing an applet , so please direct me how to enable debugging in the applet

where should i place the logback.xml ? inside the web application directory ? inside  the jar file of the applet ? in the same directory of the applet ? 

Regards

sarxos, 2015-01-28:
@amroamer please test the above example as Java application, not the applet. If example works well it means you have wrong applet configuration. Before you start debugging applet you need to know what to debug in the first place (class loader, applet, permissions, bridj, webcam-capuite), so let me ask you again - is the example I posted above works well?

amroamer, 2015-01-28:
@sarxos  sorry for not answering that earlier .
yes its working as a standalone application , and its working as standalone applet when i try to run my applet in the applet viewer not from inside a webpage .

sarxos, 2015-01-28:
@amroamer, ok then, it means that something wrong is happening in applet runtime/initialization. Let me check - did you sign all the JARs using ```jarsigner``` tool? In the working [applet example](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-applet) this is done by [maven-jarsigner-plugin](http://maven.apache.org/plugins/maven-jarsigner-plugin/), but if you do not use Maven you need to sign all JARs manually.

amroamer, 2015-01-28:
@sarxos i have self signed my applet "webcamapplet.jar" using the jarsigner tool that comes with the JDK , the JDK version is 1.7.0 .

i didnt sign the other jar files (webcam-capture-0.3.11-20141227.105512-4 ,bridj-0.6.2,slf4j-api-1.7.10,slf4j-simple-1.7.10 ) .

and as i told you before , i added the following in java.policy file : 
      permission java.security.AllPermission;

so that i could have all permission granted to my applet for my testing purposes .

sarxos, 2015-01-28:
@amroamer, you need to sign all the JARs, not only applet. The classpath permissions are derived from JARs and thus, if application tries to load DLL, or write to file, access hardware, etc, the applet loader first check if class has permission to do that. This class comes from the JAR which must be signed because settings all permissions grant is not enough. In your case you must sign BridJ JAR to load DLL, and Webcam Capture JAR to access camera hardware. 

Personally I used to create one JAR from all classes and sign only one file. You can take a look at the [applet example](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-applet) where only one shaded JAR is created and signed. The [maven-shade-plugin](http://maven.apache.org/plugins/maven-shade-plugin/) is used to create one JAR.

Moreover, please be aware that in one of the Java 7 updates, Oracle disabled possibility to load self-signed applets. This must be manually changed in Java security settings on every computer that will be running given applet.

amroamer, 2015-01-28:
i have signed all the jar files , just like you said , but using the jarsigner tool , same way as i signed the applet jar . 

now i am getting a new Exception like the following :  
java.lang.SecurityException: untrusted class package "org.slf4j.impl" in class path
	at com.sun.deploy.security.CPCallbackHandler$ParentElement.checkResource(Unknown Source)
	at com.sun.deploy.security.DeployURLClassPath$JarLoader.checkResource(Unknown Source)
	at com.sun.deploy.security.DeployURLClassPath$JarLoader.getResource(Unknown Source)
	at com.sun.deploy.security.DeployURLClassPath.getResource(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader$2.run(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader$2.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.plugin2.applet.Plugin2ClassLoader.findClassHelper(Unknown Source)
	at sun.plugin2.applet.Applet2ClassLoader.findClass(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass0(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass0(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at org.slf4j.LoggerFactory.bind(LoggerFactory.java:142)
	at org.slf4j.LoggerFactory.performInitialization(LoggerFactory.java:121)
	at org.slf4j.LoggerFactory.getILoggerFactory(LoggerFactory.java:332)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:284)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:305)
	at com.github.sarxos.webcam.Webcam.<clinit>(Webcam.java:97)
	at clientapplet.WebcamCapture.start(WebcamCapture.java:54)
	at com.sun.deploy.uitoolkit.impl.awt.AWTAppletAdapter.start(Unknown Source)
	at sun.plugin2.applet.Plugin2Manager$AppletExecutionRunnable.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
java.lang.ExceptionInInitializerError
	at clientapplet.WebcamCapture.start(WebcamCapture.java:54)
	at com.sun.deploy.uitoolkit.impl.awt.AWTAppletAdapter.start(Unknown Source)
	at sun.plugin2.applet.Plugin2Manager$AppletExecutionRunnable.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.IllegalStateException: Unexpected initialization failure
	at org.slf4j.LoggerFactory.bind(LoggerFactory.java:168)
	at org.slf4j.LoggerFactory.performInitialization(LoggerFactory.java:121)
	at org.slf4j.LoggerFactory.getILoggerFactory(LoggerFactory.java:332)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:284)
	at org.slf4j.LoggerFactory.getLogger(LoggerFactory.java:305)
	at com.github.sarxos.webcam.Webcam.<clinit>(Webcam.java:97)
	... 4 more
Caused by: java.lang.SecurityException: untrusted class package "org.slf4j.impl" in class path
	at com.sun.deploy.security.CPCallbackHandler$ParentElement.checkResource(Unknown Source)
	at com.sun.deploy.security.DeployURLClassPath$JarLoader.checkResource(Unknown Source)
	at com.sun.deploy.security.DeployURLClassPath$JarLoader.getResource(Unknown Source)
	at com.sun.deploy.security.DeployURLClassPath.getResource(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader$2.run(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader$2.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.plugin2.applet.Plugin2ClassLoader.findClassHelper(Unknown Source)
	at sun.plugin2.applet.Applet2ClassLoader.findClass(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass0(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass0(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass(Unknown Source)
	at sun.plugin2.applet.Plugin2ClassLoader.loadClass(Unknown Source)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at org.slf4j.LoggerFactory.bind(LoggerFactory.java:142)
	... 9 more

sarxos, 2015-01-28:
The [google-foo](https://www.google.pl/search?client=ubuntu&channel=fs&q=untrusted+class+package+%22org.slf4j.impl%22+in+class+path&ie=utf-8&oe=utf-8&gfe_rd=cr&ei=9PnIVKOkMpHn7Ab154CYBw#channel=fs&q=untrusted+class+package+in+class+path) says that you are mixing signed and unsigned content.

http://download.java.net/jdk7u8/docs/technotes/guides/jweb/mixed_code.html

> The following two SecurityExceptions are thrown when untrusted components have been previously accessed, an attempt to load a trusted component was previously detected, and a decision was made to allow mixed components to coexist, and a jar containing trusted components is opened and a component name collision is detected between trusted and untrusted components.

```plain
"untrusted resource \"" + name + "\" in class path"
"untrusted class package \"" + packageName + "\" in class path"
```

The above means that you have two JARs with the same package, but one is trusted, and other is not trusted (at least this is how I understood it).

Are you sure you've signed **all** the JARs?

amroamer, 2015-01-29:
@sarxos i tripled  checked all the jars , and they are all signed  with the same self-signed certificate , and i signed them again to make sure again , and then went back and checked them and they are all signed .

and i went back to java  control panel , and from the Advanced tab ( Mixed code (sandboxed vs trusted) security verification ) and i disabled the verification which is not requried , but its ok for me now as i am in development and test phase .

sarxos, 2015-01-29:
Maybe you should consider shading JAR as I did in the example? Can you verify if the JAR generated from example works in your case (you need to build it with maven)?

amroamer, 2015-02-02:
@sarxos sorry for late reply , but its really weird whats happening . 
in the first place i am developing using weblogic 12c application server , and running a Fusion Web application , and have been struggling for the last 10 days to make this applet work .
at the end i tried to develop the applet on apache tomcat 8.0.17 , and without signing , it worked !!! 
so maybe the issue is in weblogic 12c security configuration , i am not sure yet , but i will be working on this , and see how it goes , do you have any idea on this subject ?

sarxos, 2015-02-02:
Hi @amroamer, what you said is really strange. From what I know, the applet itself, does not have any dependency on the web server it is running on. Imho, it should work in the same way on both WebLogic and Tomcat.

amroamer, 2015-02-02:
@sarxos i know , but if you want   i can provide you with two test projects , one is a Fusion application running on top of Weblogic server , and the other is a regular  Dynamic web project that is running on top of Apache Tomcat .
both are using same libraries , same jar files , also same webpage that is containing the applet . 
and you are right , for me as well my understanding of applets it runs everywhere  the same way . 
Thats why i am telling you its really weird .

sarxos, 2015-02-02:
I'm afraid I cannot afford to investigate this issue (WebLogic vs Tomcat problem). I would love to help with this but I'm pretty busy with my own work, so I hope you understand.

amroamer, 2015-02-02:
@sarxos  sure i understand , i said if you want i can provide .

i will work on this issue now , and i will update you with the results .

amroamer, 2015-02-15:
Hello @sarxos  
after alot of digging into this problem , finally i came to some solution .
what i just did is moving the page that contain the library from one directory to another , and it worked like a charm .

sarxos, 2015-02-15:
Omg, I can't believe the solution was this simple :)

amroamer, 2015-02-15:
yes , i am closing this issue , thank you @sarxos ,

closed 2015-02-15 by amroamer

sarxos, 2015-02-15:
OK, thank you for the feedback!