sarxos/webcam-capture #300: IP camera view while camera disconnected

milindasaranga, 2015-01-28:
I could view IP cameras in javafx application using below code. But if I disconnect the ip camera then also webcam object return the last image before the disconnection. webcam object doesn't return error if the network is disconnected.

Here is the code: 

protected void startWebCamStream() {

              stopCamera = false;

              Task<Void> task = new Task<Void>() {

                     @Override
                     protected Void call() throws Exception {

                           while (!stopCamera) {
                                  try {
                                         try {
                                                Thread.sleep(100);
                                         } catch (InterruptedException e) {
                                                // TODO Auto-generated catch block
                                                e.printStackTrace();
                                         }
                                         System.out.println("view-1");
                                         if ((grabbedImage = webCam.getImage()) != null) {
                                                System.out.println("view-2");
                                                Platform.runLater(new Runnable() {
                                                       
                                                       @Override
                                                       public void run() {
                                                              Image mainiamge = SwingFXUtils.toFXImage(grabbedImage, null);
                                                              imageProperty.set(mainiamge);
                                                              System.out.println("view-3");
                                                              
                                                       }
                                                });

                                                grabbedImage.flush();
                                         }
                                  } catch (Exception e) {
                                         e.printStackTrace();
                                  }
                           }

                           return null;
                     }
              };

              Thread th = new Thread(task);
              th.setDaemon(true);
              th.start();
              imgWebCamCapturedImage.imageProperty().bind(imageProperty);

       }

I'm using yawcam for stream. Hope that you can help me with this .

sarxos, 2015-01-28:
Hi @milindasaranga,

I will try to implement this as an enhancement together with the #271, but I'm currently pretty busy at work and cannot implement those enhancements right now. 

If you want to contribute, please feel free to change the code to fit your needs. The IP camera device source code is available here:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/main/java/com/github/sarxos/webcam/ds/ipcam/IpCamDevice.java

What should be changed I think is ```PushImageReader``` nested class, because in the ```run()``` method it simply loops over the frames and silently break when anything is wrong.

milindasaranga, 2015-01-28:
ok, I'll try to find a solution for this issue...

milindasaranga, 2015-01-28:
Hi @sarxos 

I hope to change the image If the disconnectedTime value is more than 25. If I checked previously captured image and newly captured image, will it be correct?. I have changed the code. 

Here is the code:

private void startWebCamStream() throws Exception {

		isStoped = false;

		Task<Void> task = new Task<Void>() {

			@Override
			protected Void call() throws Exception {

				while (!isStoped) {
					try {

						Thread.sleep(200);
						if (isStoped) {
							return null;
						}

						grabbedImage = getWebCam().getImage();

						if (bufferedImagePrevious != null) {
							if (bufferedImagePrevious == grabbedImage) {
								System.out.println("the same image captured");
								disconnectedTime++;								
							} else {
								disconnectedTime = 0;			
								System.out.println("the different image captured");
							}
						}
						bufferedImagePrevious = grabbedImage;

						if (grabbedImage != null) {

							Platform.runLater(new Runnable() {

								public void run() {
									try {
										Image mainImage = SwingFXUtils
												.toFXImage(grabbedImage, null);
										imageProperty.set(mainImage);
										allCameras.get(getCameraIndex())
												.setConnectFailed(false);										
										
									} catch (WebcamException e) {
										allCameras.get(getCameraIndex())
												.setConnectFailed(true);
										try {
											updateCameraIconColor(getCameraIndex());
										} catch (Exception ex) {
											LOG.error(e.getMessage(), ex);								
										}
									} catch (Exception e) {
										LOG.error(e.getMessage(), e);										
									}
								}
							});

							grabbedImage.flush();
						}
					} catch (Exception e) {
						LOG.error(e.getMessage(), e);
					}
				}				
				return null;
			}
		};

		Thread th = new Thread(task);
		th.setDaemon(true);
		th.start();
		imvCamera.imageProperty().bind(imageProperty);

	}

sarxos, 2015-01-28:
@milindasaranga, yes, it should be ok.

milindasaranga, 2015-01-28:
ok sarxos, many thanx...

closed 2015-01-29 by milindasaranga

sarxos, 2015-01-29:
Ok, thanks @milindasaranga :) Take care!