sarxos/webcam-capture #301: Can I use Webcam Capture with DCS-2120 IP cam

sarxos, 2015-01-29:
Issue by @hwathanh:

---

I used sarxos lib to capture USB webcam successfully but not to my wireless IP cam. I own DCS-2120 IP cam. Could i use sarxos lib to connect to my IP cam via VPN and how to implement. Waiting for your reply. Thanks!

sarxos, 2015-01-29:
Hi @hwathanh, 

You can use Webcam Capture API to capture images from IP camera if and only if your DCS-2120 supports MJPEG streaming. It's enough to get MJPEG image URI and use it in the API. The working (more or less) IP camera examples can be found here:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam/src/examples/java

hwathanh, 2015-01-29:
On site http://www.ispyconnect.com/man.aspx?n=D-Link, connection type of DCS-2120 is JPEG. It means my IP cam could not be implemented with sarxos API. May be i have to find another API. Thanks for your rapid reply, sarxos!

sarxos, 2015-01-29:
Hi @hwathanh, please note that it's **Webcam Capture API**, and yes, you can use it to access your camera (in case of ```JPEG``` you need to use ```PULL``` mode). Link to examples can be found in my previous post.

hwathanh, 2015-01-30:
Thanks sarxos. I'll try and let you know result later.

hwathanh, 2015-01-30:
I did it. Thank you very much sarxos!

closed 2015-01-30 by sarxos