sarxos/webcam-capture #302: Reopen Camera

adrianlubitz, 2015-01-30:
Hi,
I want to reopen a camera, for the case of loss of connection(wireless webcam).
But when I try to open the camera again I get the following Error: 

Exception in thread "Thread-2" com.github.sarxos.webcam.WebcamLockException: Webcam USB2.0 Camera 0 has already been locked

Its the only Thread dealing with the camera.

I use the newest jar (webcam-capture-0.3.11-20141227.105512-4.jar) and use the jdk1.8.0

for testing reasons I did just this:

webcam.close();
System.out.println(webcam.isOpen());
webcam.open();
System.out.println(webcam.isOpen());

which makes the Error occur.
Thank you for your help!

sarxos, 2015-01-30:
1. Is it IP camera or UVC webcam (I assume this is UVC connected via USB)?
2. Is the above code working well when you do not cause connection loss?

adrianlubitz, 2015-02-06:
You are right its UVC connected via USB. And Code is working, when there is no connection loss. The only case it does not work is, when I remove and replug the dongle.

sarxos, 2015-02-06:
@adrianlubitz, this is unusual situation and I will take a look into this, but I would rather consider solution for this as an enhancement instead of the bug fix. I hope you are ok with that. For a temporary w/a you can monitor connection and disconnection events with ```WebcamDiscoveryListener``` and force webcam unlock when it has been unplugged.

bablunani, 2015-02-28:
plz..slove this fast bro

sarxos, 2015-02-28:
Hi @bablunani,

This issue has much lower priority than a Raspberry Pi and ARM support I'm working on lately.

If this is really urgent for you, please feel free to investigate and change the code to fit your needs. I will be happy to merge any pull request that does not break the code and introduce additional functionalities :)

bablunani, 2015-03-03:
Thank u Bro.....I used unlock() method ...it worked

sarxos, 2015-03-03:
Ok, but please note this is a simple w/a and it may fool you when you are on Windows. On Linux it works pretty well so when you try to open camera twice it should throw ```WebcamException``` and print warning in the logs, but on Windows it may actually do nothing, cause JVM crashes or capture graph instability which requires OS restart to get it back working.

AlbosHajdari, 2020-06-12:
Hello everyone, I'm sorry to post on such an old thread but I have the same problem. @bablunani said that the unlock() method worked, but it doesn't work for me when I try to run two or more applications to use the same camera.
I really need to solve this problem asap :'( .

derpnooner, 2020-11-23:
How do you force webcam.unlock() ?  I am not able to find that information.

ksx501, 2021-03-25:
the unlock method isn't working for me either. the camera is still locked.

ksx501, 2021-03-25:
and after six hours the program was working fine. without any reason