sarxos/webcam-capture #304: random crap in console using this

beacha0, 2015-02-01:
I used this to take a picture when someone enters the wrong code, and I get this in the console:

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
OpenJDK 64-Bit Server VM warning: You have loaded library /tmp/BridJExtractedLibraries8853146268352694915/libbridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Not enough args for null OpenIMAJGrabber.startSession(int, int, double)

sarxos, 2015-02-01:
These are warnings from SLF4J and BridJ. First one is caused by the fact that you did not add any logger implementation and second one is related to BridJ dyncall implementation. Both are harmless.

closed 2015-02-01 by sarxos

beacha0, 2015-02-01:
How do I fix this? I'm new to libraries and I wanna make my program seamless.

sarxos, 2015-02-01:
There is nothing to fix since these are not errors, but just an ordinary message.
* To get rid of SLF4J: see https://github.com/sarxos/webcam-capture/issues/278 or google solution - there are tons of pages about how to clean this. 
* To get rid of ```Not enough args for null OpenIMAJGrabber.startSession``` you need to use to BridJ 0.6.3 or 0.7-SNAPSHOT
* The ```OpenJDK 64-Bit Server VM warning``` is also BridJ related. Maybe the above will clean this message as well.