sarxos/webcam-capture #308: Cannot Access Webcam Image

betrand, 2015-02-11:
I tried out the example code below after obtaining the dependency via maven central

Webcam webcam = Webcam.getDefault();
webcam.open();
  if (webcam.isImageNew()) {
   LOG.info("webcam has image");
  }
BufferedImage bufferedImage = webcam.getImage();
LOG.info("BufferedImage", bufferedImage);

webcam has image
BufferedImage: null

I get null printed what can I be doing wrong?

sarxos, 2015-02-11:
@betrand,

Can you please try:

```java
Webcam webcam = Webcam.getDefault();
webcam.open();

for (int i = 0; i < 5; i++) {
  if (webcam.isImageNew()) {
    LOG.info("webcam has image");
  }
  BufferedImage bufferedImage = webcam.getImage();
  LOG.info("BufferedImage " + bufferedImage);
}
```

Are you getting ```null``` in all 5 loops, or only in first ones?

betrand, 2015-02-11:
>  @betrand,Can you please try:Webcam webcam = Webcam.getDefault();

Hi,
I can't access webcam on the first but can access from 1 - 4. I tried to run the code again it say Severe:   com.github.sarxos.webcam.WebcamLockException: Webcam Chicony USB 2.0 Camera 0 has already been locked
Am trying to see if theres a way to bypass this error occuring @ webcam.open();
Will send code shortly.
Kind regards  Database Development EngineerKTP AssociateUniversity Of HertfordshireE-mail:b.ugorji2@herts.ac.ukBetrandU@acoustical.co.ukSkype: betrandugorji

sarxos, 2015-02-11:
This code should work just as it is. Do you observe any errors except the ```null``` and the lock exception? Just FYI, the lock exception means that there is another process running Webcam Capture API which uses the same camera. It can also occur when Java process has been suddenly interrupted. In this case the lock is not released and will hold for several seconds (around ~4s). If it isn't released after this time, it most likely means that there is indeed another process which uses your webcam.

Please do not mess with the lock because it was designed to keep Windows users safe and prevent the API from crash if one decide to use the same webcam instance from multiple processes (which is forbidden).

You can have better insight into the runtime if you enable debug logs by following [this procedure](https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Dubug-Logs).

sarxos, 2015-02-11:
Just a note - you can take a look in your application work directory if there are some logs from Java crash.

Also, here is the newest Webcam Capture API development version (from yesterday):

https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture/0.3.11-SNAPSHOT/webcam-capture-0.3.11-20150210.090726-5.jar

Can you please try with the above instead of the one from Maven Central?

sarxos, 2015-02-11:
If you can't access it on the first loop, but later it's working well, then it may be linked to #285 which has been fixed in 1a7198a569aee82d630f2435f7b87d579e6fe6c3.

amroamer, 2015-02-15:
@sarxos  you said "the lock exception means that there is another process running Webcam Capture API which uses the same camera" .
but i think the LockException will occur  even if the webcam is being used from another application .

sarxos, 2015-02-15:
Hi @amroamer, when you take a look into the code ([link](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/WebcamLock.java)) you will find that webcam lock is nothing more and nothing less than a file in temporary directory which contains timestamp stored inside. This timestamp is periodically updated while webcam is used and when another thread/process (Java using Webcam Capture API) tries to open the webcam, the exception is being thrown.

There is no system-wide lock which will guard against using the same webcam in separate applications (e.g. in Java and Skype). The one implemented in Webcam Capture API works only across the library.

I designed webcam lock to avoid using webcam from two distinct Java applications using Webcam Capture API. Without the lock, depending on the operating system, when you try to open the same camera twice, you may expect:
* On Linux - silent error, the webcam cannot be open (some kind of grabber exception and warning in console),
* On Mac - _probably_ the same as above,
* On Windows - the webcam will be open, you can access is, but when you close webcam, the other application will most likely crash. Side effect is a broken capture graph subsystem so operating system needs to be restarted to open camera again.

amroamer, 2015-02-15:
@sarxos but while working on the library , i was testing the webcam that i have "microsoft lifecam" through its application , and trying to access the same webcam through the library . 

if the LifeCam is open and its application its working , and i run my application using the library , an exception will be thrown m like the following : 

com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:271)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:232)
	at com.bd.jmf.test.WebcamAppletExample.start(WebcamAppletExample.java:117)
	at sun.applet.AppletPanel.run(AppletPanel.java:475)
	at java.lang.Thread.run(Thread.java:745)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:335)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615)

sarxos, 2015-02-15:
@amroamer, this is happening in your case, of course, but the wide spectrum of possible reactions on Windows is not limited to this one only. I was testing this behaviour on my old Windows XP SP3 on the project early phase and verified that I was able to work with webcam from both Java and some other programs (not all of them, thou). I suppose there is some mechanism deeper in the operating system internals that allows or disallows sharing capture graph across the processes depending on some criteria (I don't know, maybe resolution, maybe something else).

shyakalee, 2018-07-13:
hey guys tell me how i ca use this libraly to get images from the usb webcam
i waiting from you

ivan888x, 2023-07-27:
@sarxos can you please let us know the location of the temporary directory where the locking files are handled? Thanks a lot in advance.