sarxos/webcam-capture #309: how can i take picture from a digital camera attached to my computer through USB cable.

10sw33, 2015-02-12:
HI, how are you.
First of all i like to mention that is it is not an issue , i want to know that is there any way to capture picture from a digital camera attached to my computer throught USB cable. I have sony Model # DSC-S750 Cyber Shot.
I have used this api for webcam and ipcam and it worked like charm. 
But now I need this, is it possible through webcam-Capture, if not then could you suggest me some workarounds, but it would be great  if it is possible through webcam-Capture.
Thanks

sarxos, 2015-02-12:
Hi @10sw33,

If your camera is a [UVC device](http://en.wikipedia.org/wiki/USB_video_device_class) compliant, then you should be able to use it as any other webcam. Otherwise this is not possible without writing dedicated driver (moreover, such driver can be created if and only if camera expose some kind of API which can be used to access the image).

Please note that there may be some software or hardware which can act like a bridge between your camera and UVC spec. I googled a little bit and found [this youtube video](https://www.youtube.com/watch?v=VCKo1VXMFm4) demonstrating individual using HDMI grabber to convert camera HDMI output to UVC compliant video stream, but I haven't found HDMI output in your camera specification online. Please also note that I never tested if such configuration (with grabber) can be possible with Webcam Capture API.

closed 2015-02-12 by sarxos

sarxos, 2015-02-12:
@10sw33, I google also this one, but seems like it works with Canon & Nikon only: http://sparkosoft.com/sparkocam

10sw33, 2015-02-13:
and what about other companies and models?? is it not possible todo with Webcam Capture API???

sarxos, 2015-02-13:
As I stated above, you can use any camera model until it's UVC compliant.