sarxos/webcam-capture #310: Cannot get initial image from IP camera device

viv816, 2015-02-13:
I am trying out your api to create a sample app(WebcamPanel in JFrame) which connects to IP camera when clicked on start button and captures the screen shot and saves locally when clicked on capture button.

Using xml file for ip camera setup [dlink - dcs 5220]

I get following exception when i click start button in WebcamPanel.

Exception in thread "Thread-3" com.github.sarxos.webcam.WebcamException: Cannot get initial image from IP camera device DlinkIP Camera
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getResolutions(IpCamDevice.java:282)
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getResolution(IpCamDevice.java:295)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:32)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:40)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:272)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:231)
	at com.github.sarxos.webcam.WebcamPanel.start(WebcamPanel.java:798)
	at CapCam2$3.run(CapCam2.java:116) 

Am I missing something? Kindly help.

=============== Java File Start============================
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamStorage;
import java.awt.Dimension;
import java.awt.FlowLayout;
import java.io.File;
import javax.imageio.ImageIO;
import javax.swing.JOptionPane;
/**
 *
 * @author T08
 */
public class CapCam2 extends javax.swing.JFrame {

    static {
		Webcam.setDriver(new IpCamDriver(new IpCamStorage("src/movable.xml")));
	}

    private Dimension ds    = new Dimension(320,240);
//    private Dimension cs    = WebcamResolution.VGA.getSize();
    private Webcam webcam   = null;

    private WebcamPanel wpanel = null;
    /**
     * Creates new form CamCap
     */
    public CapCam2() {
        initComponents();

        for (Webcam cam : Webcam.getWebcams()) {
            webcam = cam;
            webcam.setCustomViewSizes(new Dimension[]{ds});
            wpanel= new WebcamPanel(webcam,ds,false);
        }
        wpanel.setFillArea(true);
        panelCam.setLayout(new FlowLayout());
        panelCam.add(wpanel);

    }

    /**
     * This method is called from within the constructor to initialize the form.
     * WARNING: Do NOT modify this code. The content of this method is always
     * regenerated by the Form Editor.
     */
    @SuppressWarnings("unchecked")
    // <editor-fold defaultstate="collapsed" desc="Generated Code">
    private void initComponents() {

        btStart = new javax.swing.JButton();
        btCapture = new javax.swing.JButton();
        panelCam = new javax.swing.JPanel();

        setDefaultCloseOperation(javax.swing.WindowConstants.EXIT_ON_CLOSE);

        btStart.setText("Start");
        btStart.addActionListener(new java.awt.event.ActionListener() {
            public void actionPerformed(java.awt.event.ActionEvent evt) {
                btStartActionPerformed(evt);
            }
        });

        btCapture.setText("Capture");
        btCapture.addActionListener(new java.awt.event.ActionListener() {
            public void actionPerformed(java.awt.event.ActionEvent evt) {
                btCaptureActionPerformed(evt);
            }
        });

        panelCam.setBackground(new java.awt.Color(0, 0, 0));

        javax.swing.GroupLayout panelCamLayout = new javax.swing.GroupLayout(panelCam);
        panelCam.setLayout(panelCamLayout);
        panelCamLayout.setHorizontalGroup(
            panelCamLayout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING)
            .addGap(0, 450, Short.MAX_VALUE)
        );
        panelCamLayout.setVerticalGroup(
            panelCamLayout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING)
            .addGap(0, 360, Short.MAX_VALUE)
        );

        javax.swing.GroupLayout layout = new javax.swing.GroupLayout(getContentPane());
        getContentPane().setLayout(layout);
        layout.setHorizontalGroup(
            layout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING)
            .addGroup(layout.createSequentialGroup()
                .addGroup(layout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING, false)
                    .addComponent(btStart, javax.swing.GroupLayout.DEFAULT_SIZE, javax.swing.GroupLayout.DEFAULT_SIZE, Short.MAX_VALUE)
                    .addComponent(btCapture, javax.swing.GroupLayout.DEFAULT_SIZE, 87, Short.MAX_VALUE))
                .addPreferredGap(javax.swing.LayoutStyle.ComponentPlacement.UNRELATED)
                .addComponent(panelCam, javax.swing.GroupLayout.PREFERRED_SIZE, javax.swing.GroupLayout.DEFAULT_SIZE, javax.swing.GroupLayout.PREFERRED_SIZE)
                .addGap(0, 0, Short.MAX_VALUE))
        );
        layout.setVerticalGroup(
            layout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING)
            .addGroup(layout.createSequentialGroup()
                .addGroup(layout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING)
                    .addGroup(layout.createSequentialGroup()
                        .addComponent(btStart)
                        .addPreferredGap(javax.swing.LayoutStyle.ComponentPlacement.UNRELATED)
                        .addComponent(btCapture))
                    .addComponent(panelCam, javax.swing.GroupLayout.PREFERRED_SIZE, javax.swing.GroupLayout.DEFAULT_SIZE, javax.swing.GroupLayout.PREFERRED_SIZE))
                .addGap(0, 167, Short.MAX_VALUE))
        );

        pack();
    }// </editor-fold>

    private void btStartActionPerformed(java.awt.event.ActionEvent evt) {
        Thread t = new Thread(){
            public void run(){
              wpanel.start();
            }
        };
        t.setDaemon(true);
        t.start();
    }

    private void btCaptureActionPerformed(java.awt.event.ActionEvent evt) {
        try{

            File file;
            file = new File(String.format("capture-%d.jpg",System.currentTimeMillis()));
            ImageIO.write(webcam.getImage(), "JPG", file);
            JOptionPane.showMessageDialog(this, "Msg :\n"+file.getAbsolutePath(), "CamCap", 1);

        }catch(Exception e){
            JOptionPane.showMessageDialog(this, "Image loading failed\n"+e.getMessage(),"CamCap",0);
        }
    }

    /**
     * @param args the command line arguments
     */
    public static void main(String args[]) {
        /* Set the Nimbus look and feel */
        //<editor-fold defaultstate="collapsed" desc=" Look and feel setting code (optional) ">
        /* If Nimbus (introduced in Java SE 6) is not available, stay with the default look and feel.
         * For details see http://download.oracle.com/javase/tutorial/uiswing/lookandfeel/plaf.html
         */
        try {
            for (javax.swing.UIManager.LookAndFeelInfo info : javax.swing.UIManager.getInstalledLookAndFeels()) {
                if ("Nimbus".equals(info.getName())) {
                    javax.swing.UIManager.setLookAndFeel(info.getClassName());
                    break;
                }
            }
        } catch (ClassNotFoundException ex) {
            java.util.logging.Logger.getLogger(CapCam2.class.getName()).log(java.util.logging.Level.SEVERE, null, ex);
        } catch (InstantiationException ex) {
            java.util.logging.Logger.getLogger(CapCam2.class.getName()).log(java.util.logging.Level.SEVERE, null, ex);
        } catch (IllegalAccessException ex) {
            java.util.logging.Logger.getLogger(CapCam2.class.getName()).log(java.util.logging.Level.SEVERE, null, ex);
        } catch (javax.swing.UnsupportedLookAndFeelException ex) {
            java.util.logging.Logger.getLogger(CapCam2.class.getName()).log(java.util.logging.Level.SEVERE, null, ex);
        }
        //</editor-fold>

        /* Create and display the form */
        java.awt.EventQueue.invokeLater(new Runnable() {
            public void run() {
                new CapCam2().setVisible(true);
            }
        });
    }

    // Variables declaration - do not modify
    private javax.swing.JButton btCapture;
    private javax.swing.JButton btStart;
    private javax.swing.JPanel panelCam;
    // End of variables declaration
}
=========================Java File End================

========================XML Start===============
<?xml version="1.0" encoding="UTF-8" ?>
<storage>
	<ipcam name="Dlink IP Camera" url="http://192.168.200.215" mode="pull">
		<auth user="admin" password="admin" />
	</ipcam>
</storage>
========================XML End===============

viv816, 2015-02-13:
<?xml version="1.0" encoding="UTF-8" ?>

	<ipcam name="Dlink IP Camera" url="http://192.168.200.215" mode="pull">
		<auth user="admin" password="ns1788" />
	</ipcam>

sarxos, 2015-02-14:
@viv816, the URL is incorrect - it should point **the image URL** and not the camera panel. For example, in case of DCS-2121 this would be:

```plain
http://[server.ip]/cgi-bin/viewer/video.jpg?streamid=0:
```

You have to find out correct image URL in your camera model (DSC-5220).

Just FYI, I see you are using JPEG and ```PULL```, but MJPEG and ```PUSH``` mode gives faster FPS (please note that image URLs for MJPEG and JPEG are different).

closed 2015-02-14 by sarxos

viv816, 2015-02-15:
Hello,
Thank you very much for the quick response. I was able to make it work after following your feedback.
However, I would like to know if there is any way the API can be integrated in a web application (not using java swing approach)   ?

   

 Closed #310.—

sarxos, 2015-02-15:
You can use native HTML5 to access webcam (you do not need Webcam Capture API for that). For more details on how to access webcam from JavaScript please see https://github.com/jhuckaby/webcamjs. I found it working well on both my Windows and Linux computer.

You could also use applet, but I strongly suggest to **avoid** this approach because applets are dying technology and Oracle do nothing to change it. Moreover, dealing with permissions in applets is a pain in the butt. You can read issues other people had with applets when they tried to access webcam and you understand. Moreover, with applet approach you will have to pay for the code signing certificate from some certificate authority (like Verisign etc). This is circa $150-200.

sarxos, 2015-02-15:
@viv816, in regards to IP camera, you could setup everything on server (wire cameras with XML) and expose them as MJPEG with ```WebcamStreamer```, or even better, send new images via WebSocket to your web page (someone from Webcam Capture API community reported such usage in his project). I think WebSocket communication is better than exposing everything via ```WebcamStreamer```.

sarxos, 2015-02-15:
@viv816, here you can find very basic example demonstrating of how to transport images feed from server to fronted via WebSockets. 

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-websockets

viv816, 2015-02-23:
Thanks a lot. 
I tried the websockets example along with front end. 
I will have to modify it to suit my objective.  
Thanks again.

viv816, 2015-03-19:
Hi. The websockets example works fine with dlink-5220 model and also the Dasding cameras you mentioned. However, when I test it for fos cam IP cameras (e.g. model F19821W v2) with the xml configuration given below, it gives the same error "Cannot get initial image from IP camera device"

I tried the jpeg and mjpeg configurations separately. I followed the image url for fos cam ip camera from this link http://www.ispyconnect.com/man.aspx?n=Foscam. Am I missing something? 

<?xml version="1.0" encoding="UTF-8" ?> 
<storage>
	<ipcam name="fos cam jpeg" url="http://x.y.com:82/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr=[USERNAME]&pwd=[PASSWORD]" mode="pull"/>
	<ipcam name="fos cam mjpeg" url="http://x.y.com:82/videostream.cgi?usr=[USERNAME]&pwd=[PASSWORD]" mode="push"/>
</storage>

sarxos, 2015-03-19:
Hi @viv816,

My first guess is an incorrect XML. The ampersand sign ```&``` in XML is reserved for [entities](http://en.wikipedia.org/wiki/List_of_XML_and_HTML_character_entity_references) markup, so when you _want_ to have it in the XML attribute, you should write it as entity, that is ```&amp;``` instead of just a normal ```&```. Therefore, due to above, your XML should look like this one:

```xml
<?xml version="1.0" encoding="UTF-8" ?> 
<storage>
    <ipcam name="fos cam jpeg" url="http://x.y.com:82/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&amp;usr=[USERNAME]&amp;pwd=[PASSWORD]" mode="pull"/>
    <ipcam name="fos cam mjpeg" url="http://x.y.com:82/videostream.cgi?usr=[USERNAME]&amp;pwd=[PASSWORD]" mode="push"/>
</storage>
```

sarxos, 2015-03-19:
Just a note. I verified the above, and changing ```&``` to ```&amp;``` is fixing ```PULL``` mode, however I'm getting 404 on ```videostream.cgi``` in ```PUSH```. Is it _really_ available on FI9821W, or maybe I'm testing wrong device (I used live demo from Foscam)?

This is for ```PULL```:

[image]

And I'm getting this 404 for MJPEG URL for ```PUSH``` mode, in both:

* http://50.197.211.181:9822/videostream.cgi?usr=user&pwd=foscam and
* http://50.197.211.181:9822/cgi-bin/videostream.cgi?usr=user&pwd=foscam

[image]

Please verify if the above URL is correct.