sarxos/webcam-capture #314: Cannot identify '▒y▒Xp ▒▒ ▒(▒▒': 2, No such file or directory

aaav19983008, 2015-02-17:
Hi, I'm trying to capture an image from Raspberry PI, using this code in java

import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;
import javax.imageio.ImageIO;
import com.github.sarxos.webcam.Webcam;
import java.awt.Dimension;
/**
 * Example of how to take single picture.
 *
 * @author Bartosz Firyn (SarXos)
 */
public class FinchTest {

        public static void main(String[] args) throws IOException {

                // get default webcam and open it
                Webcam webcam = Webcam.getDefault();
                webcam.setViewSize(new Dimension(640,480));

                webcam.open(false);
                //webcam.open();

                // get image
                BufferedImage image = webcam.getImage();

                // save image to PNG file
                ImageIO.write(image, "PNG", new File("/home/pi/dev/FinchTest/PH.png"));
        }

}

I'm using these libraries
slf4j-api-1.7.10.jar:xuggle-xuggler-5.4.jar:log4j-1.2.17.jar:logback-classic-1.1.2.jar:logback-core-1.1.2.jar:slf4j-api-1.7.2.jar:core-image-1.3.jar:webcam-capture-0.3.11-20141227.105512-4.jar:webcam-capture-driver-ipcam-0.3.10-RC7.jar:bridj-0.6.3-20130316.190111-13.jar

the operating system I'm using
Linux drswifty 3.18.6+ #1 PREEMPT Sat Feb 7 18:11:00 GMT 2015 armv6l GNU/Linux

this is the error log
2015-02-10 12:18:58 INFO  [main] com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
2015-02-10 12:18:58 DEBUG [webcam-discovery-service] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver - Searching devices
2015-02-10 12:19:02 DEBUG [webcam-discovery-service] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver - Found device Camera Board OV5647 /dev/video0
2015-02-10 12:19:02 DEBUG [main] com.github.sarxos.webcam.WebcamLock - Lock Webcam Camera Board OV5647 /dev/video0
2015-02-10 12:19:02 INFO  [atomic-processor-1] com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam Camera Board OV5647 /dev/video0
2015-02-10 12:19:02 DEBUG [atomic-processor-1] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Opening webcam device Camera Board OV5647 /dev/video0
2015-02-10 12:19:02 DEBUG [atomic-processor-1] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Webcam device /dev/video0 starting session,
 size java.awt.Dimension[width=176,height=144]
Cannot identify '▒y▒Xp ▒▒ ▒(▒▒': 2, No such file or directory
2015-02-10 12:19:02 DEBUG [main] com.github.sarxos.webcam.WebcamLock - Unlock Webcam Camera Board OV5647 /dev/video0
2015-02-10 12:19:02 DEBUG [webcam-lock-[Camera Board OV5647 /dev/video0]] com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
2015-02-10 12:19:02 DEBUG [main] com.github.sarxos.webcam.Webcam - Webcam exception when opening
com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72) ~[webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140) ~[webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46) ~[webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20) ~[webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:271) [webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:232) [webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at FinchTest.main(FinchTest.java:18) [FinchTest/:na]
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:335) ~[webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38) ~[webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81) ~[webcam-capture-0.3.11-20141227.105512-4.jar:na]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1110) ~[na:1.8.0-ea]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:603) ~[na:1.8.0-ea]
        at java.lang.Thread.run(Thread.java:722) ~[na:1.8.0-ea]
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:271)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:232)
        at FinchTest.main(FinchTest.java:18)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:335)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1110)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:603)
        at java.lang.Thread.run(Thread.java:722)

I'm accessing raspberry pi over shh(putty) using wifi, I have solved a lot of error to come at this stage, out of clue now, no idea how to solve it, if you look at the error log ( Cannot identify '▒y▒Xp ▒▒ ▒(▒▒':  ), I have no idea what this part is telling. Can you please help me, this is part of my university project, any help would be highly appreciated.

amroamer, 2015-02-17:
@aaav19983008  when you try to run the above code, is your webcam running  using another application ?

sarxos, 2015-02-17:
Hi @aaav19983008,

I suppose this error:

```plain
Cannot identify '▒y▒Xp ▒▒ ▒(▒▒': 2, No such file or directory
```

Comes from:

http://sourceforge.net/p/openimaj/code/HEAD/tree/trunk/hardware/core-video-capture/src-native/linux/capture.cpp line 577.

However, the root cause, i.e. why is device name ```▒y▒Xp ▒▒ ▒(▒▒``` instead of ```/dev/video0``` is an unknown to me. It's too low level for me to find out what may be wrong.

I'm trying to find out if gstreamer or vlc can be used instead.

sarxos, 2015-02-17:
@aaav19983008,

I managed to run the following stack on my RasPi. It works on top of the VLC instead of OpenIMAJ. I hope you can use this configuration to w/a your issue, but please note that FPS rate of this solution is not very impressive.

```plain
pi@raspberrypi ~ $ sudo apt-get install vlc
pi@raspberrypi ~ $ vlc --version
VLC media player 2.0.3 Twoflower (revision 2.0.2-93-g77aa89e)
VLC version 2.0.3 Twoflower (2.0.2-93-g77aa89e)
Compiled by buildd on build01.raspbian.lan (Jul  7 2014 21:46:39)
Compiler: gcc version 4.6.3 (Debian 4.6.3-14+rpi1) 
```

Artifacts:

* [webcam-capture-0.3.10.jar](http://search.maven.org/remotecontent?filepath=com/github/sarxos/webcam-capture/0.3.10/webcam-capture-0.3.10.jar)
* [webcam-capture-driver-vlcj-0.3.10.jar](http://search.maven.org/remotecontent?filepath=com/github/sarxos/webcam-capture-driver-vlcj/0.3.10/webcam-capture-driver-vlcj-0.3.10.jar)
* [vlcj-2.4.1.jar](http://search.maven.org/remotecontent?filepath=uk/co/caprica/vlcj/2.4.1/vlcj-2.4.1.jar)
* [jna-4.1.0.jar](http://search.maven.org/remotecontent?filepath=net/java/dev/jna/jna/4.1.0/jna-4.1.0.jar)

As I stated, the FPS is not impressive, and it could be potentially improved by magnitude of tenth just by using Webcam Capture API v0.3.11 (snapshot), but I checked it on RasPi and all the images in such configuration are black and application does not want to stop.

aaav19983008, 2015-02-17:
Thank's a lot @amroamer and @sarxos for the responses, :)

@amroamer I'm using the camera comes with Raspberry PI or created for Raspberry PI only, as a result I don't think other program using the camera. (That's what I understand)

@sarxos Since I don't have that much deep knowledge about all these stuff, so I have no idea why the device name is ▒y▒Xp ▒▒ ▒(▒▒ instead of /dev/video0, if you please can tell me how to change the name I will change that. 

And just to give you a little more overview, my project is with Finch Robot, at this stage I'm trying to take a picture successfully, because I have to do image processing. I'm trying everything you have suggested, will be back. :)

sarxos, 2015-02-17:
@aaav19983008,

You wrote that you are using "the camera comes with Raspberry PI or created for Raspberry PI only". Are you referring to Raspberry Pi camera module, like this for example?

http://www.raspberrypi.org/products/camera-module/

Or you are referring to some other USB camera (or rather UVC device) that you bought together with Pi?

sarxos, 2015-02-17:
I'm asking because Webcam Capture API never supported Raspberry Pi camera module. This is because camera module works on top of MMAL which is not UVC compliant. For the camera module you can use ```raspistill``` tool to get the image.

aaav19983008, 2015-02-17:
@sarxos 
I'm referring to Raspberry Pi camera module, and what you said that means I'm trying in a wrong way to make things work. :( 
if you please tell me little more about raspistill tool, or just provide a link i need to have a look that how to implement that one, if that will take long time to work with I will manage USB camera, because I'm already trying to work with Webcam capture.

sarxos, 2015-02-17:
@aaav19983008, 

It seems like there was some development effort to make RasPi Cam usable via V4L ([link](http://www.linux-projects.org/modules/sections/index.php?op=viewarticle&artid=14)) but I never tested if this solution may work or not since I do not have my own camera module.

http://www.linux-projects.org/modules/sections/index.php?op=viewarticle&artid=16

In regards to the ```raspistill``` and ```raspivid```, the first one is to capture image from camera module, and second one to record a movie.

Here is some good resource. Both tools are described with options they expose. If this is not enough then you

http://elinux.org/Rpi_Camera_Module

If you would like to stay with Java in your project, then you can simple use ```ProcessBuilder``` to run ```raspistill``` whenever you need new image. You can save the output to file or pre-created named pipe. By using named pipe you will be able to stream frames in the same way as you would do with the webcam, so from the Java perspective nothing would change actually. Instead of ```ProcessBuilder``` you can also use _signal_ mode (```-s```) with infinity time ```-t 0``` to capture image in respond to ```SIGUSR1``` (e.g. ```kill -SIGUSR1 <pid>```).

There is no plain Java API that will act like a nice proxy to ```raspistill``` tool, but wrapping it in Java is IMHO a hundred times better option than writing own interface from scratch using MMAL. Everything is doable, the resources are available, so you will not have to reinvent the wheel, but everything depends on your needs (e.g. if you need high or low FPS, etc).

aaav19983008, 2015-02-19:
@sarxos 
Thank you very much for the information you have provided, :) . I will try that.

Before that I have managed a webcam today (HP Webcam HD 2300), I have tried with this code 

import java.awt.Dimension;
import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;
import javax.imageio.ImageIO;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.ds.vlcj.VlcjDriver;
import com.github.sarxos.webcam.WebcamResolution;
import com.github.sarxos.webcam.WebcamPanel;
import javax.swing.JFrame;

public class FinchTest {

        static {
        Webcam.setDriver(new VlcjDriver());
        }

        public static void main(String[] args) throws InterruptedException {

        Webcam webcam = Webcam.getWebcams().get(0);
        webcam.setViewSize(WebcamResolution.VGA.getSize());

        WebcamPanel panel = new WebcamPanel(webcam);
        panel.setFPSDisplayed(true);
        panel.setImageSizeDisplayed(true);

        JFrame window = new JFrame("Webcam Panel");
        window.add(panel);
        window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
        window.pack();
        window.setVisible(true);
	}
}

and I have got this error

03:25:11.394 [main] DEBUG com.github.sarxos.webcam.Webcam - Setting new capture driver VlcjDriver
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
        at java.util.ArrayList.rangeCheck(ArrayList.java:603)
        at java.util.ArrayList.get(ArrayList.java:381)
        at java.util.Collections$UnmodifiableList.get(Collections.java:1225)
        at FinchTest.main(FinchTest.java:20)

I believe this line (Webcam webcam = Webcam.getWebcams().get(0);) is responsible.

----------------------------------------------------------------------------------------------------

then I have modified code little bit

import java.awt.Dimension;
import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;
import javax.imageio.ImageIO;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamResolution;
import com.github.sarxos.webcam.WebcamPanel;
import javax.swing.JFrame;

public class FinchTest {

        
        public static void main(String[] args) throws InterruptedException {

        Webcam webcam = Webcam.getDefault();
[... 123 lines cut ...]
libv4l2: error dequeuing buf: Input/output error
libv4l2: error dequeuing buf: Input/output error
libv4l2: error dequeuing buf: Input/output error
libv4l2: error dequeuing buf: Input/output error
libv4l2: error dequeuing buf: Input/output error
libv4l2: error dequeuing buf: Input/output error

-------------

02:22:46.250 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
02:22:46.512 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
warning: error opening device
02:22:51.553 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device HP Webcam HD 2300 /dev/video0
02:22:51.624 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam HP Webcam HD 2300 /dev/video0
02:22:52.351 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam HP Webcam HD 2300 /dev/video0
02:22:52.354 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Opening webcam device HP Webcam HD 2300 /dev/video0
02:22:52.372 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device /dev/video0 starting session, size java.awt.Dimension[width=176,height=144]
warning: error opening device
warning: error opening device
02:22:55.685 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device session started
02:23:01.977 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
02:24:29.077 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Initialize buffer
Killed

Can you please give me some idea, I'm desperate to make it work.

sarxos, 2015-02-19:
@aaav19983008,

Today after I'm back at home I will commit changes in ```webcam-capture-driver-v4l4j``` which I verified to be the most stable on the RasPi, however, from what I see in the logs, the default driver should work in your case because it calls ```libv4l2``` as expected.

Why are you releasing the lock?

```java
 webcam.getLock().disable();
```

If you experience problems with the lock it means that the other Java process is using this webcam. If this is the case, then you will not be able to open it (this may cause ```warning: error opening device```) again. You first need to kill other process (grep your Java process and kill, the ```<pid>``` is a process ID).

```plain
$ ps -ef | grep -i java
$ kill -9 <pid>
```

This is strange:

```plain
libv4l2: error dequeuing buf: Input/output error
```

I read people having trouble with some USB cameras on RasPi, however, according to [this](http://elinux.org/RPi_USB_Webcams) wiki page your camera should work fine.

Can you please install ```fswebcam``` and verify if you can capture the image? In case the ```fswebcam``` works well, you can switch to ```webcam-capture-driver-fswebcam``` to start, and then debug the problem deeper to make use either default or V4L4J driver.

aaav19983008, 2015-02-19:
@sarxos 

I don't really remember why I was releasing the lock! But still, even without releasing the lock the problem is same, 

07:52:28.590 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
07:52:28.783 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
warning: error opening device
07:52:33.805 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device HP Webcam HD 2300 /dev/video0
07:52:33.884 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam HP Webcam HD 2300 /dev/video0
07:52:33.950 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam HP Webcam HD 2300 /dev/video0
07:52:33.953 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Opening webcam device HP Webcam HD 2300 /dev/video0
07:52:33.958 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device /dev/video0 starting session, size java.awt.Dimension[width=176,height=144]
warning: error opening device
warning: error opening device
07:52:44.941 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
07:52:50.959 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device session started
07:53:03.064 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Initialize buffer
Killed

by using this command "ps -ef | grep -i java" what I found is
pi@drswifty ~/dev/FinchTest $ ps -ef | grep -i java
pi        2724  2635  0 07:44 pts/0    00:00:00 grep --color=auto -i java

and if I execute "kill -9 2635", it shuts down Putty immediately.

I have Installed fswebcam, this link 
http://www.raspberrypi.org/documentation/usage/webcams/
has the instruction how to install. I think I don't know how to switch to webcam-capture-driver-fswebcam. The way I'm using library is by adding them to .sh file and I run .sh file to run program. since there is no jar file for this I have nothing to add and not sure what to do, can you please tell me little more specifically.

And do I need to remove installed driver (don't know how to remove), like I have installed vlc before.

I have tried with process builder today, it's simple and works with PI camera module.

sarxos, 2015-02-19:
@aaav19983008, the ```ps``` didn't show any java process running. Instead you killed your terminal :)

Ok, when you already have ```fswebcam``` please test if you can capture the image. In the terminal execute:

```plain
$ fswebcam test.jpg
```

Then, copy ```test.jpg``` on your PC and verify if everything is ok with this snapshoot. When everything is good, please follow to the ```webcam-capture-fswebcam-driver``` page where you can read more about how to use it:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-fswebcam

Unfortunately the download link on the above page points to the old version (the ```0.3.10-RC7``` instead of ```0.3.10```), but the newest stable one can be found in Maven Central Repository:

http://search.maven.org/#artifactdetails|com.github.sarxos|webcam-capture-driver-fswebcam|0.3.10|jar

aaav19983008, 2015-02-19:
:)
I have tried already, the image quality was not good with fswebcam xxx.jpg command,

then with this fswebcam -r 1280x720 xxx.jpg command was little better

the with this fswebcam -r 1280x720 --no-banner xxx.jpg command was better than before

ok, I following the link, will be back. :)

aaav19983008, 2015-02-19:
@sarxos

I have tried fswebcam, with the provided code (I have tried with webcam-capture-0.3.10.jar and webcam-capture-0.3.10-RC7.jar, didn't make any difference), not surprised that I got error, here is the log

20:46:47.826 [main] DEBUG com.github.sarxos.webcam.Webcam - Setting new capture driver FsWebcamDriver
20:46:48.055 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video1
20:46:48.299 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video0
20:46:48.446 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam /dev/video1
20:46:48.769 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam /dev/video1
20:46:48.774 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - Creating pipe: mkfifo /tmp/fswebcam-pipe-video1.mjpeg
20:46:48.835 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - New stream reader
20:46:48.858 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - New stream reader
20:46:48.882 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open /dev/video1
20:46:48.924 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - Invoking command: /usr/bin/fswebcam --no-banner --no-shadow --no-title --no-subtitle --no-timestamp --no-info --no-underlay --no-overlay -d /dev/video1 -r 176x144 /tmp/fswebcam-pipe-video1.mjpeg -v
20:46:48.936 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - New stream reader
20:46:48.940 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - New stream reader
20:46:49.048 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR --- Opening /dev/video1...
20:46:49.058 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR Trying source module v4l2...
20:46:49.067 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR Error opening device: /dev/video1
20:46:49.072 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR open: Operation not permitted
20:46:49.076 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR Trying source module v4l1...
20:46:49.081 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR Error opening device: /dev/video1
20:46:49.084 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR open: Operation not permitted
20:46:49.092 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR Unable to find a source module that can read /dev/video1.
20:46:49.097 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR
20:46:58.496 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video1
20:46:58.498 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video0
20:47:08.529 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video1
20:47:08.531 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video0
20:47:18.557 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video1
20:47:18.563 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video0

sarxos, 2015-02-19:
Seems like fswebcam can only read from ```/dev/video0```. Isn't ```/dev/video1``` your RasPi camera module? This would explain why it cannot be read neither ```v4l``` nor ```v4l2```.

aaav19983008, 2015-02-20:
@sarxos 

only the USB camera is attached to the Pi not the RasPi camera, I have disconnected that, and from the previous error I can see that /dev/video0 is the RasPi camera module not the /dev/video1, I'm saying that because this is the first time this /dev/video1 one is appearing. 

I have unplugged the camera and plugged it again (I suppose you meant that by saying "Can you disconnect it?"), but didn't make any difference. 

As a result what can I do to make v4l module read form /dev/video1 device?

sarxos, 2015-02-20:
@aaav19983008,

In regards to your question:

> what can I do to make v4l module read form /dev/video1 device?

You can do nothing I guess. This device is not compatible with V4L. Some magic tricks with [UV4L](http://www.linux-projects.org/modules/news/) could possibly help, but I'm not familiar with this framework and cannot advice here.

One of these cameras is USB and when you ran ```fswebcam test.jpg``` it choose correct one (I suppose it was ```/dev/video0``` because we verified that you cannot read from ```/dev/video1```. I have no idea why your system is showing two cameras instead of just one, but I guess this is due to some kind of permanent configuration for RsPi camera module.

On, back to the topic, you have two cameras in your system, one can be read, and second cannot. You can try this to find out which one is ok:

```java
System.out.println(Webcam.getWebcams().size()); // this should give you 2
```

Lets try first one:

```java
Webcam w = Webcam.getWebcams().get(0);
w.open();
ImageIO.write(w.getImage(), "JPG", new File("0.jpg"));
w.close();
```

And second one:

```java
Webcam w = Webcam.getWebcams().get(1);
w.open();
ImageIO.write(w.getImage(), "JPG", new File("1.jpg"));
w.close();
```

By doing this you should be able to determine which one is readable.

sarxos, 2015-02-20:
PS. Just bought RasPi NoIR camera module. I will be experimenting with it in the next week.

aaav19983008, 2015-02-23:
@sarxos 

I was little busy with something else for few days, sorry about that

By executing this line of code ( System.out.println(Webcam.getWebcams().size()); ) I found this 
21:22:52.388 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
21:22:52.548 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
warning: error opening device
21:22:56.738 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device HP Webcam HD 2300 /dev/video0
1

By executing these line of code 
             Webcam w = Webcam.getWebcams().get(0);
             w.open();
             ImageIO.write(w.getImage(), "JPG", new File("0.jpg"));
             w.close(); 
 I found this 
21:45:48.070 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
21:45:48.253 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
warning: error opening device
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
        at java.util.ArrayList.rangeCheck(ArrayList.java:603)
        at java.util.ArrayList.get(ArrayList.java:381)
        at java.util.Collections$UnmodifiableList.get(Collections.java:1225)
        at FinchTest.main(FinchTest.java:22)

By executing these line of code 
             Webcam w = Webcam.getWebcams().get(1);
             w.open();
             ImageIO.write(w.getImage(), "JPG", new File("1.jpg"));
             w.close();
 I found this
21:47:44.098 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
21:47:44.258 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
warning: error opening device
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 1, Size: 0
        at java.util.ArrayList.rangeCheck(ArrayList.java:603)
        at java.util.ArrayList.get(ArrayList.java:381)
        at java.util.Collections$UnmodifiableList.get(Collections.java:1225)
        at FinchTest.main(FinchTest.java:22)

sarxos, 2015-02-23:
Hi @aaav19983008,

To be honest I have no idea what is happening on your RasPi...

1. On Friday there was 2 webcams (```/dev/video0``` and ```dev/video1```).
2. On the first run today you had 1 webcam (```/dev/video0```).
3. On the second run today you was not able to open device...
4. On the third run today you was not able to open device again...

It looks like the camera is open already when you are trying to open it in 2. and 3.

Are you using fswebcam-driver or none?

sarxos, 2015-02-23:
Above you said you are using some shell script to run application, can you post it here?

aaav19983008, 2015-02-25:
@sarxos

to run program I run 2 .sh file, 
here is the first one build.sh (this version is just for finch {to make finch move} )

#!/bin/sh
LIBS=/home/pi/dev/BBTechSoftwareForPi/FinchJava
javac -classpath .:$LIBS/finch.jar FinchTest.java

-----------this one I used for camera, you will see name of the libraries

build.sh

#!/bin/sh
LIBS=/home/pi/dev/BBTechSoftwareForPi/FinchJava
javac -classpath .:$LIBS/ slf4j-api-1.7.10.jar:xuggle-xuggler-5.4.jar:log4j-1.2.17.jar:logback-classic-1.1.2.jar:logback-core-1.1.2.jar:
slf4j-api-1.7.2.jar:core-image-1.3.jar:webcam-capture-0.3.11-20141227.105512-4.jar:webcam-capture-driver-ipcam-0.3.10-RC7.jar:
bridj-0.6.3-20130316.190111-13.jar	FinchTest.java

other one run.sh
#!/bin/sh
LIBS=/home/pi/dev/BBTechSoftwareForPi/FinchJava
java -classpath .:$LIBS/finch.jar FinchTest

I have run those code one more time with both camera plugged in
here are the results (I was feeling so excited when I saw that both of the camera is working) :) :)

for code System.out.println(Webcam.getWebcams().size())

here is the log 
00:35:08.052 [main] DEBUG com.github.sarxos.webcam.Webcam - Setting new capture driver FsWebcamDriver
00:35:08.306 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video1
00:35:08.469 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video0
2

for code
Webcam w = Webcam.getWebcams().get(0);
             w.open();
             ImageIO.write(w.getImage(), "JPG", new File("0.jpg"));
             w.close();

here is log
00:36:04.989 [main] DEBUG com.github.sarxos.webcam.Webcam - Setting new capture driver FsWebcamDriver
00:36:05.240 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video1
00:36:05.402 [webcam-discovery-service] INFO  c.g.s.w.ds.fswebcam.FsWebcamDriver - Found video file /dev/video0
00:36:05.536 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam /dev/video1
00:36:05.856 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam /dev/video1
00:36:05.860 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - Creating pipe: mkfifo /tmp/fswebcam-pipe-video1.mjpeg
00:36:05.917 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - New stream reader
00:36:05.938 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - New stream reader
00:36:05.962 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open /dev/video1
00:36:06.004 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - Invoking command: /usr/bin/fswebcam --no-banner --no-shadow --no-title --no-subtitle --no-timestamp --no-info --no-underlay --no-overlay -d /dev/video1 -r 176x144 /tmp/fswebcam-pipe-video1.mjpeg -v
00:36:06.012 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - New stream reader
00:36:06.069 [atomic-processor-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - New stream reader
00:36:06.093 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR --- Opening /dev/video1...
00:36:06.100 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR Trying source module v4l2...
00:36:06.343 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR /dev/video1 opened.
00:36:06.355 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_get_capability,87: /dev/video1 information:
00:36:06.369 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_get_capability,88: cap.driver: "raspicam"
00:36:06.377 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_get_capability,89: cap.card: "Camera Board OV5647"
00:36:06.382 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_get_capability,90: cap.bus_info: "CSI"
00:36:06.386 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_get_capability,91: cap.capabilities=0x05000001
00:36:06.390 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_get_capability,92: - VIDEO_CAPTURE
00:36:06.394 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_get_capability,101: - READWRITE
00:36:06.397 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_get_capability,103: - STREAMING
00:36:06.401 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR No input was specified, using the first.
00:36:06.403 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_set_input,181: /dev/video1: Input 0 information:
00:36:06.408 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_set_input,182: name = "default"
00:36:06.412 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_set_input,183: type = 00000002
00:36:06.420 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_set_input,185: - CAMERA
[... 195 lines cut ...]
00:19:11.184 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR Using palette JPEG
00:19:11.658 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_set_mmap,693: mmap information:
00:19:11.661 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR src_v4l2_set_mmap,694: frames=4
[... cut ...]
00:19:12.319 [process-reader-1] DEBUG c.g.s.w.ds.fswebcam.FsWebcamDevice - FsWebcam: ERROR
00:19:13.670 [main] DEBUG com.github.sarxos.webcam.Webcam - Closing webcam /dev/video1
00:19:13.682 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamCloseTask - Closing /dev/video1
00:19:13.687 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock Webcam /dev/video1
00:19:13.691 [webcam-lock-[/dev/video1]] DEBUG com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
00:19:13.704 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam /dev/video1 has been closed

sarxos, 2015-02-25:
That is great! In this case I'm closing this ticket.

closed 2015-02-25 by sarxos

aaav19983008, 2015-02-26:
@sarxos 
Thank you so much for your support, guideline and time you have spent, really appreciate that. One additional question Since I have to do image processing after that, I'm struggling to find a helpful resource, I have less than a week to do that, if you know anything about it can you please just provide link or something you think appropriate.

sarxos, 2015-02-26:
@aaav19983008,

You're welcome :)

In regards to your question - it **really** depends on what processing you need to to, but if you want to stick to the Java I would suggest to start with OpenIMAJ (which yesterday migrated from Sourceforge to Github):

http://www.openimaj.org/
https://github.com/openimaj/openimaj

This project consist of many dependencies so at least basic knowledge of Maven may be required in your case, but it's imho still a better option than using OpenCV, which may be much harder to learn. But do not use ```VideoCapture``` class since it most likely won't run on your RasPi (Webcam Capture API default driver use the same classes and we saw you get SIGSEGV).

If you think OpenIMAJ does not fit into your needs you can try BoofCV (I think it's a little bit simpler when working with images, but I have no idea about performance):

http://boofcv.org
https://github.com/lessthanoptimal/BoofCV

Or use JavaCV wrapper for OpenCV (but I'm not sure if it will run on RasPi):

https://github.com/bytedeco/javacv