sarxos/webcam-capture #315: SIGSEGV when running Webcam Capture API on Raspberry Pi model B

sarxos, 2015-02-17:
```plain
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam USB2.0 Camera /dev/video0
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xa8838a64, pid=4577, tid=2827678832
#
# JRE version: Java(TM) SE Runtime Environment (7.0_40-b43) (build 1.7.0_40-b43)
# Java VM: Java HotSpot(TM) Client VM (24.0-b56 mixed mode linux-arm )
# Problematic frame:
# C  [OpenIMAJGrabber.so+0x1a64]  Device::getIdentifier()+0x14
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/Desktop/webcam-capture-parent/webcam-capture-examples/webcam-capture-test/hs_err_pid4577.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
#
```

Dranithix, 2015-02-24:
Hello sarxos,

Thanks a lot for the wonderful wrapper for webcam capturing, though do you have any estimated time of date for when this issue notably with the Raspberry Pi B+ will be fixed?

I am currently using the Pi for a regional science fair project competition with your library, and would want to display a demo of the program on a Raspberry Pi. Though, I have obtained this exact error as well.

sarxos, 2015-02-24:
Hi @Dranithix,

This issue can be observed when using default capture driver (or [webcam-capture-vlcj-driver](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-vlcj), but the SIGSEGV stack is different in this case). I doubt it can be fixed without changes in the [BridJ issue 525](https://github.com/ochafik/nativelibs4java/issues/525) so I would not expect this to be fixed soon.

The good news is that I managed to run Webcam Capture API on my RasPi model B using [webcam-capture-driver-v4l4j](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-v4l4j). Therefore, everything should work well in your case when you set it instead of default one.

```java
// set Webcam Capture V4L4J driver to be used instead of default one
static {
    Webcam.setDriver(new V4l4jDriver());
}

// do the other things you would like to do
public static void main(String[] args) throws Exception {
    Webcam webcam = Webcam.getDefault();
    webcam.open();
    ImageIO.write(webcam.getImage(), "PNG", new File("hello-world.png"));
    webcam.close();
}
```

But, as I stated above, it was tested on model B and I'm not sure how B+ will react with the so library compiled on B.

Just FYI, to make use of webcam-capture-driver-v4l4j you should add these JARs to your classpath:

* [webcam-capture-driver-v4l4j-0.3.11-SNAPSHOT.jar](https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture-driver-v4l4j/0.3.11-SNAPSHOT/webcam-capture-driver-v4l4j-0.3.11-20150219.205806-6.jar)
* [v4l4j-0.9.1-r507.jar](http://search.maven.org/remotecontent?filepath=com/github/sarxos/v4l4j/0.9.1-r507/v4l4j-0.9.1-r507.jar)