sarxos/webcam-capture #316: Would you like a view of architect - project web (running jetty) + Webcam-capture-ipcam

henriquem, 2015-02-25:
I would like to listen the opinion from Sarxos about this Architecture:
please at look if It is recommended ?

I would like to create an project web, in HTML and jetty-websocket I can transmit images in HTML 
The following solution is ugly?

My Thread for each Camera registered:

    class ThreadIpCamera  extends Thread implements WebcamMotionListener{
				    WebcamMotionDetector detectorMoviment;
					private String name; // Related IpCamMpjeg name=#1... for this thread
					ThreadIpCamera(WebCam web,String name){
					    this.name=name;
					    detectorMoviment = new WebcamMotionDetector(camera);
					    detectorMoviment.setInterval(500); // one check per 500 ms
					    detectorMoviment.addMotionListener(this);
					    detectorMoviment.start();
				    }
    }

In web.xml (jetty container)

    
    <servlet>
        <description>Servlet initialize all IPCameras</description>
        <display-name>ThreadsCamerasIP</display-name>
        <servlet-name>ThreadsCamerasIP</servlet-name>
        <servlet-class>com.myproject.LoadCamerasIPServlet</servlet-class>
        <load-on-startup>1</load-on-startup> 
    </servlet>

When Start Jetty load All IP Cameras and start an Thread for these:

     class LoadCamerasIPServlet extends HttpServlet {
            public void init(ServletConfig config) throws ServletException {
                Webcam.setDriver(new IpCamDriver());
                IpCamDeviceRegistry.register(new IPCamMjpeg("#1", "http://ip1", IpCamMode.PUSH, "admin", "123456"));
                IpCamDeviceRegistry.register(new IPCamMjpeg("#2", "http://ip2", IpCamMode.PUSH, "admin", "123456"));
                IpCamDeviceRegistry.register(new IPCamMjpeg("#3", "http://ip3", IpCamMode.PUSH, "admin", "123456"));
			
                List<Webcam> webcams = Webcam.getWebcams();
                    for (Webcam webcam : webcams) {
				    webcam.getLock().disable();
				    /* start a thread for each camera treatment */
				    new ThreadIpCamera(webcam, webcam.getName()).start(); 
			    }
		    }
     }

Conclusion:

Another Alternative I send to Trash all code above, and Create 2 projects: 1 Project Web, and Java Desktop;
So When  starting Jettty container, start also the java desktop, where Java Desktop contains only WebCam framework and ThreadIpCamera class and each motion from camera, send request for http://localhost ? 
Ex. new URL("localhost/NotifyWebSocket?cam=1&img_encoded=AERe324$3ds....... 

Please could you help me, what the options is better model arquitecture ?

Thks

sarxos, 2015-02-26:
If you consider these two options, either:

* Have a server stack and transmit images to frontend via ```ws://```, or
* Have a server stack and desktop application to transmit images via MJPEG stream,

I think the first option is better. The work overhead is smaller in first case and the web traffic generated by the MJPEG and WebSocket is pretty the same. Moreover, if you have motion listener you can send new images just in case when motion has been detected, which decreases traffic needs. This is not true in case of MJPEG.

Just a note. Please do not mess with the webcam lock. It was designed to prevent opening camera twice. This may not has such big impact in case of IP camera, but you for sure do not want to waste the bandwidth to transmit images which will never be used. Thus, you should get rid of this:

```java
webcam.getLock().disable();
```

henriquem, 2015-02-26:
Thks so much!!!
Good explanation clear, 
I'll follow your recommendation, take out from code: 

    webcam.getLock().disable();

After that,Using the same ideia and Jetty, my next challenges is implements in this idea stream in DVR  (by internet) using differents media formats.

I have seen application in android that works fine it... but I don't know yet what the secret, but follow the same ideia of driver "webcam-ipcam": host:post user:pwd.  therefore DVR has Setting for configuration to view by internet.
But the difference is instead of mjpeg the formats can be: RTSP or H.264 ...

If you have some suggestions, will be welcome ...

sarxos, 2015-02-26:
The h.264 implementations in Java are pretty rough and, honestly, I cannot recommend any of the ones I was trying in the past, but you try them in hope anything has changed or any of them fit into your needs:
* [JCodec](https://github.com/jcodec/jcodec) - pure Java implementation of h.264 encoder/decoder, project owners claims it is working also on Android,
* [Xuggler](https://github.com/artclarke/xuggle-xuggler) - wide set of coder and decoders which can be used in Java, but the project is dead and it's not working on Android.
* [gstreamer-java](https://code.google.com/p/gstreamer-java/) - same as above, but work well only with GStreamer 0.10.x (which is actually a carbon age dated nowadays).

Unfortunately I cannot give you any details on RTMP/RTSP since I'm completely unfamiliar with this protocol (never tried to do it myself).

closed 2015-02-27 by sarxos