sarxos/webcam-capture #321: Linux initialization error

PredragBoksic, 2015-03-18:
I am developing this project:
http://perceptron.sourceforge.net/

It works fine in Windows, but recently it stopped working in Linux for unknown reasons. (e.g. after an update?)

0:15:39.272 [Perceptron : main window thread] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
20:15:39.280 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
Java HotSpot(TM) 64-Bit Server VM warning: You have loaded library /tmp/BridJExtractedLibraries1408520079867548750/libbridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Not enough args for null OpenIMAJGrabber.startSession(int, int, double)
java.lang.reflect.InvocationTargetException
        at java.awt.EventQueue.invokeAndWait(EventQueue.java:1312)
        at java.awt.EventQueue.invokeAndWait(EventQueue.java:1287)
        at javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1348)
        at perceptron.Main.<init>(Main.java:126)
        at perceptron.Main.main(Main.java:51)
Caused by: java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
        at java.util.ArrayList.rangeCheck(ArrayList.java:653)
        at java.util.ArrayList.get(ArrayList.java:429)
        at java.util.Collections$UnmodifiableList.get(Collections.java:1309)
        at perceptron.Perceptron.webcam(Perceptron.java:2736)
        at perceptron.Perceptron.<init>(Perceptron.java:394)
        at perceptron.Main.run(Main.java:138)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:301)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:749)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:702)
        at java.awt.EventQueue$3.run(EventQueue.java:696)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:719)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Perceptron could not be made a reality.
20:15:42.525 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
20:15:45.526 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices

....goodbye from Perceptron....
Exception in thread "Perceptron : main window thread" java.lang.NullPointerException
        at perceptron.Main.exit(Main.java:148)
....
....
....

PredragBoksic, 2015-03-18:
The error appears at startup:

    /**
     * Webcam setup.
     */
    void webcam() {
        try {
            Webcam.setDriver(new WebcamCompositeDriver(
                    //new WebcamDefaultDriver(),
                    //new OpenImajDriver(),
                    new IpCamDriver(new IpCamStorage("resource/cameras.xml"))
            ));

            /*
            try {
                IpCamDeviceRegistry.register(new IpCamDevice("Lignano", "http://88.37.116.138/mjpg/video.mjpg", IpCamMode.PUSH));
            } catch (MalformedURLException e) {
                e.printStackTrace();
            }
            */
            webcams = Webcam.getWebcams(8, TimeUnit.SECONDS);
            if (webcams != null) {
                webcam = webcams.get(0);
                if (webcam != null) {
                    for (int i = 0; i < webcams.size(); i++) {
                        System.out.println("Webcam " + (i + 1) + " : " + webcams.get(i).getName());
                    }
                    // size of webcam raw image and the buffered image that will be filled with that data
                    webcam.setViewSize(WebcamResolution.VGA.getSize()); // Size 640x480 // TODO choose webcam resolution
                    webcam_capture_formatted = new BufferedImage(screen_width, screen_height, BufferedImage.TYPE_INT_RGB);
                    //Webcam.getDiscoveryServiceRef().stop();
                    cam_works = true;
                } else {
                    System.out.println("No webcam detected");
                    cam_works = false;
                }
            }
        } catch (WebcamException | TimeoutException e) {
            cam_works = false;
            System.err.println("Webcam was not initialized.");
            e.printStackTrace();
        }
    }

PredragBoksic, 2015-03-18:
I tried switching drivers, but nothing happened.

PredragBoksic, 2015-03-18:
I am getting closer to the truth!

The linux version cannot start given that the linux does not have any Webcam. The sarxos driver causes the program to stuck.

When the physical webcam is connected to the latest Mint Linux, the program loads, except that it cannot read the appropriate preset file for some inexplicable reason.

The physical camera cannot be accessed, however. The program is stuck again.
00:21:29.967 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Initialize buffer

The IpCam can be accessed and used in linux.

    /**
     * Webcam setup.
     */
    void webcam() {
        try {
            Webcam.setDriver(new WebcamCompositeDriver(
                    new WebcamDefaultDriver(),
                    new OpenImajDriver(),
                    new IpCamDriver(new IpCamStorage("resource/cameras.xml"))
            ));

            //*
            // PULL = static JPEG (client pull image from the server)
            // PUSH = MJPEG (Motion JPEG, server push image to client)
            try {
                String name = "Test #255";
                String url = "http://ce3014.myfoscam.org:20054/videostream.cgi";
                IpCamMode mode = IpCamMode.PUSH;
                IpCamAuth auth = new IpCamAuth("user", "user");
                IpCamDeviceRegistry.register(name, url, mode, auth);
            } catch (MalformedURLException e) {
                e.printStackTrace();
            }
            //*/
            webcams = Webcam.getWebcams(8, TimeUnit.SECONDS);
            if (webcams != null) {
                webcam = webcams.get(0);
                if (webcam != null) {
                    for (int i = 0; i < webcams.size(); i++) {
                        System.out.println("Webcam " + (i + 1) + " : " + webcams.get(i).getName());
                    }
                    // size of webcam raw image and the buffered image that will be filled with that data
                    webcam.setViewSize(WebcamResolution.VGA.getSize()); // Size 640x480 // TODO choose webcam resolution
                    webcam_capture_formatted = new BufferedImage(screen_width, screen_height, BufferedImage.TYPE_INT_RGB);
                    //Webcam.getDiscoveryServiceRef().stop();
                    cam_works = true;
                } else {
                    System.out.println("No webcam detected");
                    cam_works = false;
                }
            }
        } catch (WebcamException | TimeoutException e) {
            cam_works = false;
            System.err.println("Webcam was not initialized.");
            e.printStackTrace();
        }
    }

sarxos, 2015-03-19:
Hi @PredragBoksic,

These drivers:

```java
new WebcamDefaultDriver(),
new OpenImajDriver(),
```

Are to support UVC webcams. You should not mix two UVC drivers in one composite because every webcam may be reported twice in ```getWebcams()``` output. These drivers uses native library to access webcam hardware. Some system misconfiguration may be causing these drivers to fail (e.g. when user is not permitted to access webcam, when V4L is missing, etc).

This one however:

```java
new IpCamDriver(new IpCamStorage("resource/cameras.xml"))
```

Is for IP cameras only. The only possibilities which may cause it to fail are:

* User is not permitted to read ```resource/cameras.xml``` file,
* The is no Internet connection,
* Some web proxies or firewalls are blocking network traffic,
* IP camera username/password are incorrect.

Which driver your application (Perceptron) is using finally? UVC or IP camera? These are two different worlds and to find out where the problem is located we need to identify which driver is causing it.

sarxos, 2015-03-19:
You may also try newest BridJ instead of 0.6.2:

* [0.6.3-SNAPSHOT](https://oss.sonatype.org/content/repositories/snapshots/com/nativelibs4java/bridj/0.6.3-SNAPSHOT/bridj-0.6.3-20130316.190111-13.jar)
* [0.7-SNAPSHOT](https://oss.sonatype.org/content/repositories/snapshots/com/nativelibs4java/bridj/0.7-SNAPSHOT/bridj-0.7-20150307.234752-76.jar)
* [0.7.1-SNAPSHOT](https://oss.sonatype.org/content/repositories/snapshots/com/nativelibs4java/bridj/0.7.1-SNAPSHOT/bridj-0.7.1-20150311.095906-3.jar)

PredragBoksic, 2015-03-19:
I re-created the project Perceptron and included minimum required libraries from Maven + BridJ 0.7.0.

It works well in Windows even without the specification of driver. 

It works well with the specification of drivers too:

                Webcam.setDriver(new WebcamCompositeDriver(
                        new WebcamDefaultDriver(),
                        new IpCamDriver(new IpCamStorage("resource/cameras.xml"))
                ));

The local webcam and the internet webcams work also good.

    <ipcam name="Dasding 01" url="http://www.dasding.de/ext/webcam/webcam770.php?cam=1"/>

This camera address does NOT work:

    <ipcam name="ce3014" url="http://ce3014.myfoscam.org:20054/videostream.cgi" mode="push">
        <auth user="user" password="user"/>
    </ipcam>

and the error is 

17:11:40.049 [Perceptron : settings window thread] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam ce3014
17:11:40.051 [Perceptron : settings window thread] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock Webcam ce3014
17:11:40.053 [webcam-lock-[ce3014]] DEBUG com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
17:11:40.063 [Perceptron : settings window thread] DEBUG com.github.sarxos.webcam.Webcam - Webcam exception when opening
com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72) ~[webcam-capture-0.3.10.jar:na]
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140) ~[webcam-capture-0.3.10.jar:na]
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46) ~[webcam-capture-0.3.10.jar:na]
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20) ~[webcam-capture-0.3.10.jar:na]
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:271) ~[webcam-capture-0.3.10.jar:na]
	at perceptron.Perceptron$Settings_Window.lambda$new$49(Perceptron.java:4930) [Perceptron/:na]
	at perceptron.Perceptron$Settings_Window$$Lambda$54/413056597.itemStateChanged(Unknown Source) [Perceptron/:na]
	at javax.swing.AbstractButton.fireItemStateChanged(AbstractButton.java:2050) [na:1.8.0_40]
	at javax.swing.AbstractButton$Handler.itemStateChanged(AbstractButton.java:2353) [na:1.8.0_40]
	at javax.swing.DefaultButtonModel.fireItemStateChanged(DefaultButtonModel.java:455) [na:1.8.0_40]
	at javax.swing.JToggleButton$ToggleButtonModel.setSelected(JToggleButton.java:272) [na:1.8.0_40]
	at javax.swing.JToggleButton$ToggleButtonModel.setPressed(JToggleButton.java:289) [na:1.8.0_40]
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252) [na:1.8.0_40]
	at java.awt.Component.processMouseEvent(Component.java:6525) [na:1.8.0_40]
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324) [na:1.8.0_40]
	at java.awt.Component.processEvent(Component.java:6290) [na:1.8.0_40]
	at java.awt.Container.processEvent(Container.java:2234) [na:1.8.0_40]
	at java.awt.Component.dispatchEventImpl(Component.java:4881) [na:1.8.0_40]
	at java.awt.Container.dispatchEventImpl(Container.java:2292) [na:1.8.0_40]
	at java.awt.Component.dispatchEvent(Component.java:4703) [na:1.8.0_40]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898) [na:1.8.0_40]
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533) [na:1.8.0_40]
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462) [na:1.8.0_40]
	at java.awt.Container.dispatchEventImpl(Container.java:2278) [na:1.8.0_40]
	at java.awt.Window.dispatchEventImpl(Window.java:2750) [na:1.8.0_40]
	at java.awt.Component.dispatchEvent(Component.java:4703) [na:1.8.0_40]
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:751) [na:1.8.0_40]
	at java.awt.EventQueue.access$500(EventQueue.java:97) [na:1.8.0_40]
	at java.awt.EventQueue$3.run(EventQueue.java:702) [na:1.8.0_40]
	at java.awt.EventQueue$3.run(EventQueue.java:696) [na:1.8.0_40]
	at java.security.AccessController.doPrivileged(Native Method) [na:1.8.0_40]
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75) [na:1.8.0_40]
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86) [na:1.8.0_40]
	at java.awt.EventQueue$4.run(EventQueue.java:724) [na:1.8.0_40]
	at java.awt.EventQueue$4.run(EventQueue.java:722) [na:1.8.0_40]
	at java.security.AccessController.doPrivileged(Native Method) [na:1.8.0_40]
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75) [na:1.8.0_40]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:721) [na:1.8.0_40]
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201) [na:1.8.0_40]
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116) [na:1.8.0_40]
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105) [na:1.8.0_40]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) [na:1.8.0_40]
[... 128 lines cut ...]
                webcam = webcams.get(0);
                if (webcam != null) {
                    for (int i = 0; i < webcams.size(); i++) {
                        System.out.println("Webcam " + (i + 1) + " : " + webcams.get(i).getName());
                    }
                    // size of webcam raw image and the buffered image that will be filled with that data
                    webcam.setViewSize(WebcamResolution.VGA.getSize()); // Size 640x480 // TODO choose webcam resolution
[... cut ...]
                    //Webcam.getDiscoveryServiceRef().stop(); // not used with composite driver
                    cam_works = true;
                } else {
                    System.out.println("No webcam detected");
                    cam_works = false;
                }
            }
        } catch (WebcamException | TimeoutException e) {
            cam_works = false;
            System.err.println("Webcam was not initialized.");
            e.printStackTrace();
        }
    }

And the method of use is:

 webcam.open(true);

sarxos, 2015-03-31:
Hi @PredragBoksic,

Sorry for the late response. I was pretty busy with my private business. In regards the the issue in question, it seems like the NPE is causing whole this troubles, not the fact that you are working on Linux. If I remember correctly I was investigating very similar one in the past and now will have to take another look into the ```IpCamDevice``` code.

```plain
Caused by: java.lang.NullPointerException
    at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImage(IpCamDevice.java:302)
    at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getResolutions(IpCamDevice.java:262)
    at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getResolution(IpCamDevice.java:285)
    at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:32)
    at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
    at java.lang.Thread.run(Thread.java:745)
```

I'm marking this as bug to be resolved since there should be no NPE being threw.

sarxos, 2015-03-31:
Just FYI, I'm changing the title to reflect the problem.

PredragBoksic, 2015-03-31:
The IpCam driver makes a NPE when the xml file is empty also, but that's fine. Thanks for looking into this.