sarxos/webcam-capture #322: Strange error message ...

Kurohasan, 2015-03-22:
Hello,

I work with the intel edison module (on a arduino board) and try to get access to my webcam, which is connected via the usb type A port. Yocto haven't any problems with detecting the cam (i execute lsusb -t, and the device appear in the list), but when I execute my application I get strange error messages. I found an similar Issue here, but it seems that my english isn't that good to understand the instructions ... 

Here is the error: 

<code>SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/home/root/programms/Client/lib/logback-classic-1.1.2.jar!/org/sl                                                                                                                                          f4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/home/root/programms/Client/lib/slf4j-simple-1.7.10.jar!/org/slf4                                                                                                                                          j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [ch.qos.logback.classic.util.ContextSelectorStaticBinder]
10:02:19.352 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will                                                                                                                                           be used
10:02:19.503 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
Java HotSpot(TM) Client VM warning: You have loaded library /tmp/BridJExtractedLibraries23719068806                                                                                                                                          49318115/libbridj.so which might have disabled stack guard. The VM will try to fix the stack guard                                                                                                                                           now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z                                                                                                                                           noexecstack'.
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.                                                                                                                                          so : libv4l2.so.0: cannot open shared object file: No such file or directory
java.lang.RuntimeException: Library 'OpenIMAJGrabber' was not loaded successfully from file '/tmp/B                                                                                                                                          ridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.so'
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:922)
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:901)
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:513)
        at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:627)
        at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:828)
        at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:755)
        at org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:209)
        at org.bridj.BridJ.initialize(BridJ.java:970)
        at org.bridj.NativeObject.<init>(NativeObject.java:23)
        at org.bridj.StructObject.<init>(StructObject.java:14)
        at org.bridj.cpp.CPPObject.<init>(CPPObject.java:27)
        at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:                                                                                                                                          64)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(Webc                                                                                                                                          amDefaultDriver.java:56)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:745)
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.Execution                                                                                                                                          Exception: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:1                                                                                                                                          24)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:789)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:768)
        at robotclient.RobotClient.main(RobotClient.java:24)
[... cut ...]                     ile '/tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.so'
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:922)
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:901)
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:513)
        at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:627)
        ... 13 more
</code>
 
Is there anyone who can help me? Please excuse my bad english - I hope that I described the problem in an understandable way.

sarxos, 2015-03-29:
Hi @Kurohasan,

It seems like you are missing ```libv4l2``` installed on the Yoko:

```plain
# BridJ: dlopen error when loading 
/tmp/BridJExtractedLibraries2371906880649318115/OpenIMAJGrabber.so : 
libv4l2.so.0: cannot open shared object file: No such file or directory
```

Please note that community confirmed OpenIMAJGrabber to have problems on ARM architectures and therefore it may not work properly even when you install ```libv4l2``` (but still it's worth to try). If you confirm it not working on Edison board we can try to build custom driver on top of [driver-v4l4j](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-v4l4j) which runs much better on ARM.

Kurohasan, 2015-03-29:
Thanks for your reply sarxos,

"It seems like you are missing libv4l2 installed on the Yoko:"

Yes - a few looks later I was able to figure it out. I flashed Ubilinux to install the libv4l-dev libraries (I was not able to get it on Yocto) - perhaps I will also install the JRE on the edison tomorrow to test my application. 

Sorry for my bad english again.