sarxos/webcam-capture #324: GStreamer error in 64bit windows

anuraagbs, 2015-03-25:
Hello,
First of all thanks for the webcam capture driver, I am using GStreamer and it is working like a charm on 32-bit OS,  but when I am using on 64bit, its not working. I do not see any GStreamer installation file  for  64 bit . Can you please help me in resolving this issue.

Thanks,
Anuraag

sarxos, 2015-03-29:
Hi @anuraagbs,

Unfortunately I found only 32-bit GStreamer installers, no 64-bit ones, however you should be able to w/a this problem by using both 32-bit Java and 32-bit GStreamer installation, thought I never tried such solution.

anuraagbs, 2015-03-29:
Thank you, will surely try that, appreciate your time

sarxos, 2015-03-31:
@anuraagbs, please let us know about the outcome :) Take care!

anuraagbs, 2015-03-31:
Did not work :(

sarxos, 2015-03-31:
Damn :( In this case I see no possibility to have GStremaer driver running on 64-bit.

closed 2015-04-16 by sarxos