sarxos/webcam-capture #325: Debug logs

robolivable, 2015-03-30:
Should I be seeing debug logs like these in the current distribution?

    14:56:07.562 [AWT-EventQueue-0] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
    14:56:07.569 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
    14:56:08.541 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device FaceTime HD Camera (Built-in) 0x1a11000005ac8509

I might be overlooking a detail regarding disabling the logs, but every JAR I try importing gives me the same results. These logs are getting mixed in with the debug logs I'm outputting and it's starting to get a bit too cluttered.

sarxos, 2015-03-31:
Hi @robolivable,

Logs are configurable and disabled by default, but in case root logger is set to debug, SLF4J will of course enable debug logs in Webcam Capture API as well.

What logging framework are you using? Can you post the configuration?

If you are using Logback, adding:

```xml
<logger name="com.github.sarxos.webcam" level="warn" />
```

Into your ```logback.xml``` should be enough to cut down logs amount from Webcam Capture API.

robolivable, 2015-03-31:
Hey @sarxos, thanks for the quick reply!

I'm not using any logging framework aside from my own imbedded logging interface. I played around with including a logback configuration file in my working directory and got some output from the logger that your API is using. Should there be any particular configuration settings I should be including in the logback.xml file?

Thanks again.

robolivable, 2015-03-31:
Hey @sarxos,

Just to run it by you, I am deploying this API in a Java project that will be running on a Hummingboard embedded system. I'll be happy to keep you posted with any compatibility trouble I may run into. The board is currently running a light version of a Debian distribution.

sarxos, 2015-04-01:
Hi @robolivable,

In regards to Logback, you can of course control logging level by adding the following logger element:

```xml
<logger name="com.github.sarxos.webcam" level="warn" />
```

Same with other SLF4J-compatible frameworks - all you need to do is to set level on the ```com.github.sarxos.webcam```, e.g. in JUL with SLF4J bridge this would be:

```plain
com.github.sarxos.webcam=SEVERE
```

Please keep me informed on the progress with Webcam Capture API on Hummingboard integration. I'm really curious if it will work well.

Just FYI, from the debug logs I see you are using default driver, which may cause you some troubles on ARM processors (dues to issues in BridJ library). For more details on how to w/a this please take a look at the wiki page [How-To-Configure-Raspberry-Pi](https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi).

closed 2015-04-16 by sarxos