sarxos/webcam-capture #327: " Not able to write lock file "

FrLukeDuke, 2015-04-07:
Trying to execute a jar file compiled with the "How to take picture and save to file" example on a raspbberry pi and am encountering the following error:

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Not able to write lock file
        at com.github.sarxos.webcam.WebcamLock.write(WebcamLock.java:215)
        at com.github.sarxos.webcam.WebcamLock.update(WebcamLock.java:272)
        at com.github.sarxos.webcam.WebcamLock.lock(WebcamLock.java:294)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:296)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:251)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:236)

Any suggestions of a fix?

Kind Regards

FrLukeDuke, 2015-04-07:
UPDATE

I enabled the camera in raspi-config and ran it again and got the following result. 

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.

# A fatal error has been detected by the Java Runtime Environment:
#  SIGSEGV (0xb) at pc=0xada09a64, pid=2333, tid=2907325552
# JRE version: Java(TM) SE Runtime Environment (8.0-b132) (build 1.8.0-b132)
# Java VM: Java HotSpot(TM) Client VM (25.0-b70 mixed mode linux-arm )
# Problematic frame:
# C  [OpenIMAJGrabber.so+0x1a64]  Device::getIdentifier()+0x14
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
# An error report file with more information is saved as:
# /home/pi/Camera/dist/hs_err_pid2333.log
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp

So I ran the suggested command 

pi@raspberrypi ~/Camera/dist $ ulimit -c unlimited

And then I got this result

pi@raspberrypi ~/Camera/dist $ sudo java -jar Camera.jar
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xad3d7a64, pid=2367, tid=2907165808
#
# JRE version: Java(TM) SE Runtime Environment (8.0-b132) (build 1.8.0-b132)
# Java VM: Java HotSpot(TM) Client VM (25.0-b70 mixed mode linux-arm )
# Problematic frame:
# C  [OpenIMAJGrabber.so+0x1a64]  Device::getIdentifier()+0x14
#
# Core dump written. Default location: /home/pi/Camera/dist/core or core.2367
#
# An error report file with more information is saved as:
# /home/pi/Camera/dist/hs_err_pid2367.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.sun.com/bugreport/crash.jsp
#

FrLukeDuke, 2015-04-07:
Actually just figured it out.... Apologies for posting but thought I'd through my solution up anyway. 

Added fswebcam as a driver, downloaded it here 

http://mvnrepository.com/artifact/com.github.sarxos/webcam-capture-driver-fswebcam/0.3.10

and then added the jar to my library

then added this code above the main method

  static {
    Webcam.setDriver(new FsWebcamDriver());
  }

sarxos, 2015-04-08:
Hi @FrLukeDuke,

The ```FsWebcamDriver``` will work but it's pretty slow (1 FPS, maybe). For the best performance you should use ```V4l4jDriver``` (up to 25 FPS with RasPi Camera module and up to 5 FPS with UVC devices). More details (with all necessary links) on how to configure and use camera on RasPi can be found here:

https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi

FrLukeDuke, 2015-04-08:
I had trouble when trying to use the V4l4jDriver. Does this need to be installed on the pi aswell for it to work?

sarxos, 2015-04-08:
No, you just download the JAR in the same way as with ```FsWebcamDriver``` and include it in your applications classpath. That's all.

Some dependencies must be installed (usually they already are), if I am correct this should do it:

```plain
sudo apt-get install libjpeg8-dev libv4l-dev
```

_(note: here I'm using devel, but you can install production)_

FrLukeDuke, 2015-04-14:
Tried that and tried to run with the V4l4jDriver. I'm now getting this error in the terminal 

pi@raspberrypi ~/MotionDetect/dist $ java -jar MotionDetect.jar
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further detail                               s.
Exception in thread "main" java.lang.NoClassDefFoundError: com/github/sarxos/v4l                               4j/V4L4J
        at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.<clinit>(V4l4jDriver.ja                               va:33)
        at motiondetect.MotionDetect.<clinit>(MotionDetect.java:26)
Caused by: java.lang.ClassNotFoundException: com.github.sarxos.v4l4j.V4L4J
        at java.net.URLClassLoader$1.run(URLClassLoader.java:372)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(URLClassLoader.java:360)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        ... 2 more

sarxos, 2015-04-14:
Seems like you have not added JAR with ```V4l4jDriver``` to the classpath. Is ```MotionDetect.jar``` shaded? If so can you please tell me which JARs it consists?

The message from SLF4J is not an error. It's a warning and it tells you that you have not added any logger implementation (e.g. Logback, SLF4J Simple, etc) into the classpath.

FrLukeDuke, 2015-04-14:
I'm afraid I don't understand what you mean by 'shaded'. Here's a picture of all the JAR files associated with the program

[image]

sarxos, 2015-04-15:
Hi @FrLukeDuke,

The "shaded" JAR is a name for a JAR with _all_ necessary classes included inside. E.g. in classpath you have many, sometimes even hundred JARs, which can be repackaged into the single JAR which is later described as "shaded". Instead of deploying these many JARs you can deploy single, shaded JAR.

Ok, now back to your problem. I should describe this in more details on the wiki page (this is already fixed) instead of the driver page, but you need to have one more JAR (for V4L4J JAR itself). It can be downloaded from Maven Central:

http://search.maven.org/#artifactdetails|com.github.sarxos|v4l4j|0.9.1-r507|jar

Just FYI, please take a look at [Maven](https://maven.apache.org/) project, I'm using it for many years now and I do not have to bother about JARs and dependencies any more. It's enough to add small XML snippet into ```pom.xml``` and all JARs are automatically downloaded from Maven Central. All modern IDEs (Eclipse, Netbeans, InteliJ) has very good integration with this tool.

FrLukeDuke, 2015-04-15:
Much obliged for all your help. I certainly will look into the Maven project. 

I added the additional JAR and now I'm getting this error. I suspect it may be my webcam?

``SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further detail                                                                s.
[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:302)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:251)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:236)
        at motiondetect.MotionDetect.<init>(MotionDetect.java:35)
        at motiondetect.MotionDetect.main(MotionDetect.java:60)
Caused by: com.github.sarxos.webcam.WebcamException: No valid resolution detected for /dev/video0
        at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.getResolution(V4l4jDevice.java:111)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:32)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:744)
``

sarxos, 2015-04-16:
Hi @FrLukeDuke,

Before telling if this is caused by the webcam or not, lets try to debug things a little bit. Device impl scans all possible formats and prints them to the console in trace level. To make use of this feature you will have to enable logging mechanism. To do that please follow the steps below:

1.
Download [slf4j-simple-1.7.2.jar](http://search.maven.org/#artifactdetails|org.slf4j|slf4j-simple|1.7.2|jar) and include it in classpath,

2.
Add below code in your app

 ```java
static {
	System.setProperty("org.slf4j.simpleLogger.defaultLogLevel", "info");
	System.setProperty("org.slf4j.simpleLogger.log.com.github.sarxos.webcam.ds.v4l4j", "trace");
	Webcam.setDriver(new V4l4jDriver());
}
```

Example:

```java
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamViewer;
import com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver;

public class V4l4jViewerExample {

	static {
		System.setProperty("org.slf4j.simpleLogger.defaultLogLevel", "info");
		System.setProperty("org.slf4j.simpleLogger.log.com.github.sarxos.webcam.ds.v4l4j", "trace");
		Webcam.setDriver(new V4l4jDriver());
	}

	public static void main(String[] args) {
		new WebcamViewer();
	}
}
```

After you run the app, you should see some messages in the console, e.g. in my case this is something like:

```plain
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Get video device info
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing MJPEG
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Testing YUYV
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Ignore RGB24, resolution type not supported
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Ignore BGR24, resolution type not supported
[webcam-discovery-service] TRACE com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Ignore YUV420, resolution type not supported
[webcam-discovery-service] DEBUG com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice - Best image format match MJPEG - 48
```

In my case it does not support neither RGB24, nor BGR24, nor YUV420. I need to figure out why.

sarxos, 2015-04-16:
@FrLukeDuke,

I did small enhancement in the code to w/a this strange scenario and now, even when format resolution is reported as ```unsupported```, the code artificially sets some predefined resolutions which I know to be supported by 99.9% of devices worldwide (QQVGA, QVGA and VGA).

Can you please try with this fix? You have to download and replace JAR for V4L4J Webcam Capture driver:

[webcam-capture-driver-v4l4j-0.3.11-20150416.101824-7.jar](https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture-driver-v4l4j/0.3.11-SNAPSHOT/webcam-capture-driver-v4l4j-0.3.11-20150416.101824-7.jar)

FrLukeDuke, 2015-04-16:
It works! Brilliant! It worked straight away with the addition of the updated JAR. Cant thank enough you for all your help.

sarxos, 2015-04-16:
Great :) I'm glad it works! Take care.

closed 2015-04-16 by sarxos

trbLeeciN, 2016-07-22:
Hello, sarxos, I am trying to run webcam in Odroid c2, I need v4l4j jar for it, the link you gave crashed, can you give alternative jar link please. ([https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture-driver-v4l4j/0.3.11-SNAPSHOT/webcam-capture-driver-v4l4j-0.3.11-20150416.101824-7.jar])
Btw many thanks for this awesome webcam code.