sarxos/webcam-capture #328: Error when set custom resolutions in multiple webcams simultaneously connected

SandraGuzman, 2015-04-08:
Hi I am trying  to execute  TakeSnapshotFromAllWebcamsExample and  set custom resolutions for all, only initialize two first and  webcam has image BufferedImage: null
I tested this example with 4 webcams simultaneously connected and I'm using HD720p in custom resolution

Any suggestions of a fix?

Kind Regards

sarxos, 2015-04-09:
Hi Sandra,

You may be hitting problem with USB capacity. Depending on the image format (which also depends on the operating system, camera hardware, and drivers), various bandwidth may be reserved on USB controller.

For example (assume you are on Windows), in case of:

HD720p with RGB format at 30 FPS, the bandwidth required is:

```
1280 px * 720 px * 3 bytes * 30 FPS = 82 MB /s
```

HD720p with YUV format at 30 FPS, the bandwidth required is:

```
1280 px * 720 px * 2 bytes * 30 FPS = 55 MB /s
```

In my HP EliteBook 8460p, the HD webcam may work in HD720p at 10 FPS in RGB format. This gives me:

```
1280 px * 720 px * 3 bytes * 10 FPS = 28 MB /s
```

Therefore, half of my USB 2.0 capacity is reserved (because USB 2.0 protocol is able to process circa 60 MB/s) and thus giving me limit of max 2 HD cameras connected at once to the same USB controller.

Note that the above limits are valid on USB 3.0 as well. This is because even when you are connecting webcam to USB 3.0, the protocol supported by webcam is still 2.0, which switches controller to 2.0 protocol mode.

If you are on Linux, you are more lucky because you may use [webcam-captrure-driver-v4l4j](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-v4l4j) which select the less bandwidth consuming format available from the camera. In case of most of the medium and high quality models this will be MJPEG (low quality and cheap cameras support only RGB in most cases), which, at least in theory, should be very low bandwidth consuming, but please bear in mind that I never tested if this is indeed true.

To summarize - to w/a this problem you will have to either:
* lower resolutions,
* use two distinct USB controllers (please note that one of the options is that all USB ports may be connected to single USB controller on your motherboard - in this case it sucks),
* use Linux with webcam-captrure-driver-v4l4j to optimize format,
* open only two cameras at once, take picture, close them, and open another two ones, etc.

SandraGuzman, 2015-04-09:
Hi @sarxos ,
Thank you very much for the fast response. I was able to make it work after following your feedback : 
open only two cameras at once, take picture, close them, in bucle for all webcams

Thank you for your help :D

sarxos, 2015-04-10:
The pleasure is mine :) Take care!

closed 2015-04-10 by sarxos