sarxos/webcam-capture #329: Web Cam Cannot be found

noellaneiv, 2015-04-09:
I developed a simple web app using the WebcamMotionListener, and have run into an issue that googleing has not assisted me with. The application works fine on my development machine, using the same usb webcam, but once I deployed it on my raspbery pi, it can no longer access/find the webcam. it gives me null when i use webcam.getdefault(). I can access and use the webcam manually with mplayer, on the pi, so the drivers are installed and the web cam appears to be working correctly. here's the stack trace, though i doubt it will be of use, since it just tells me that getdefault() returned null basically.

Apr 09, 2015 2:30:27 PM org.apache.catalina.core.StandardWrapperValve invoke
SEVERE: Servlet.service() for servlet [Servlets.Navigator] in context with path [] threw exception
java.lang.NullPointerException
        at WebCamMotion.DetectMotion.<init>(DetectMotion.java:27)
        at domain.Initializer.<init>(Initializer.java:91)
        at Servlets.Navigator.processRequest(Navigator.java:47)
        at Servlets.Navigator.doGet(Navigator.java:97)
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:621)
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:722)
        at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:305)
        at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:210)
        at org.apache.catalina.core.StandardWrapperValve.invoke(StandardWrapperValve.java:225)
        at org.apache.catalina.core.StandardContextValve.invoke(StandardContextValve.java:123)
        at org.apache.catalina.authenticator.AuthenticatorBase.invoke(AuthenticatorBase.java:472)
        at org.apache.catalina.core.StandardHostValve.invoke(StandardHostValve.java:168)
        at org.apache.catalina.valves.ErrorReportValve.invoke(ErrorReportValve.java:98)
        at org.apache.catalina.valves.AccessLogValve.invoke(AccessLogValve.java:927)
        at org.apache.catalina.core.StandardEngineValve.invoke(StandardEngineValve.java:118)
        at org.apache.catalina.connector.CoyoteAdapter.service(CoyoteAdapter.java:407)
        at org.apache.coyote.http11.AbstractHttp11Processor.process(AbstractHttp11Processor.java:1002)
        at org.apache.coyote.AbstractProtocol$AbstractConnectionHandler.process(AbstractProtocol.java:579)
        at org.apache.tomcat.util.net.JIoEndpoint$SocketProcessor.run(JIoEndpoint.java:310)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:744)

I have double checked and installed all of the proper raspberry pi necesities.

Thank you,
Noel Lane

sarxos, 2015-04-10:
Hi @noellaneiv,

On your Raspberry Pi, are you using RasPi Camera Module or UVC device (USB webcam)?

Have you maybe took a look at the info on this wiki page?

https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi

sarxos, 2015-04-16:
@noellaneiv, any news on this?