sarxos/webcam-capture #331: Gstreamer driver example working fine in eclipse but showing error after exporting to runnable jar

rakeshbhatt10, 2015-04-16:
Hi Sarxos,

I Tried to run gstreamer example in windows 7. 
I downloaded http://code.google.com/p/ossbuild/downloads/detail?name=GStreamer-WinBuilds-GPL-x86.msi installed it add it to environment variable. 
When i run it from eclipse everything works fine but as i export it to Runnable jar and try to run it showing me below errors;

Exception in thread "main" java.util.MissingResourceException: Can't find bundle for base name com/github/sarxos/webcam/i18n/WebcamPanel, locale en_US
        at java.util.ResourceBundle.throwMissingResourceException(ResourceBundle.java:1427)
        at java.util.ResourceBundle.getBundleImpl(ResourceBundle.java:1250)
        at java.util.ResourceBundle.getBundle(ResourceBundle.java:705)
        at com.github.sarxos.webcam.WebcamUtils.loadRB(WebcamUtils.java:78)
        at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:735)
        at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:711)
        at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:700)
        at WebcamPanelExample.main(WebcamPanelExample.java:24)
14:57:48.107 [gstreamer-shutdown-hook] DEBUG c.g.s.w.ds.gstreamer.GStreamerDriver - GStreamer deinitialization

Thanks,
Rakesh Bhatt

sarxos, 2015-04-16:
Hi @rakeshbhatt10,

This issue is not caused by the GStreamer. It's due to missing ```WebcamPanel``` translation resources (i18n), but these should be available inside the JAR (```com/github/sarxos/webcam/i18n``` directory). Can you please post some code sample where you reproduce this issue? Together with the command line to run the app? There is indeed no file for ```en_US```, but there is ```en``` which should be used as a fallback.

rakeshbhatt10, 2015-04-16:
static {
		Webcam.setDriver(new GStreamerDriver());
	}

	public static void main(String[] args) {

		Webcam webcam = Webcam.getDefault();
                Dimension[] prefD = {new Dimension(1920,1080)};
                
		webcam.setViewSize(prefD[0]);
                webcam.setCustomViewSizes(prefD);
		WebcamPanel panel = new WebcamPanel(webcam);
		panel.setDisplayDebugInfo(true);
		panel.setFPSDisplayed(true);
		panel.setFillArea(true);

		JFrame window = new JFrame("Test webcam panel");
		window.add(panel);
		window.setResizable(true);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}

After exporting it into Runnable Jar i am just running command

java -jar sample.jar in command prompt

rakeshbhatt10, 2015-04-20:
Hi,

I am getting this error:

Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: java.lang.NullPointerException
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:789)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:768)
        at com.piquor.camutility.WebCamPanelJavaCv.initWebCamCamera(WebCamPanelJavaCv.java:53)
        at com.piquor.component.NativeComponentLoader.initilizeCamera(NativeComponentLoader.java:71)
        at com.piquor.app.PiquorApp.main(PiquorApp.java:35)
Caused by: java.util.concurrent.ExecutionException: java.lang.NullPointerException
        at java.util.concurrent.FutureTask$Sync.innerGet(FutureTask.java:222)
        at java.util.concurrent.FutureTask.get(FutureTask.java:83)
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
        ... 6 more
Caused by: java.lang.NullPointerException
        at com.github.sarxos.webcam.ds.gstreamer.GStreamerDriver.getDevices(GStreamerDriver.java:116)
        at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
        at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:1)
        at java.util.concurrent.FutureTask$Sync.innerRun(FutureTask.java:303)
        at java.util.concurrent.FutureTask.run(FutureTask.java:138)
        at java.util.concurrent.ThreadPoolExecutor$Worker.runTask(ThreadPoolExecutor.java:895)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:918)
        at java.lang.Thread.run(Thread.java:662)
11:36:02.511 [gstreamer-shutdown-hook] DEBUG c.g.s.w.ds.gstreamer.GStreamerDriver - GStreamer deinitialization

Webcam capture works really fine when i run it from eclipse but as i export it into Runnable jar. It is showing above error.

Please help i tried latest BridJ 0.7.0. Your latest build but nothing working here

frankjbrown, 2016-04-12:
Hello... 
I am having the same issue, although it is one year later.....    Works fine within eclipse but the runnable jar fails...     Was a solution found to this issue? 

[main] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam Microsoft LifeCam Studio 0
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.util.MissingResourceException: Can't find bundle for base name com/github/sarxos/webcam/i18n/WebcamPanel, locale en_US
	at java.util.ResourceBundle.throwMissingResourceException(Unknown Source)
	at java.util.ResourceBundle.getBundleImpl(Unknown Source)
	at java.util.ResourceBundle.getBundle(Unknown Source)
	at com.github.sarxos.webcam.WebcamUtils.loadRB(WebcamUtils.java:78)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:731)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:707)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:696)
	at com.github.sarxos.webcam.WebcamBarCode3.<init>(WebcamBarCode3.java:149)
	at com.github.sarxos.webcam.WebcamBarCode3.main(WebcamBarCode3.java:384)
	... 5 more
[shutdown-hook-1] INFO com.github.sarxos.webcam.WebcamShutdownHook - Automatic Microsoft LifeCam Studio 0 deallocation
[shutdown-hook-1] INFO com.github.sarxos.webcam.Webcam - Disposing webcam Microsoft LifeCam Studio 0

frankjbrown, 2016-04-12:
Never mind...... after stepping back for a few minutes and thinking about the error, I just copied the i18n folder from webcam_capture into my project, and the problem has resolved....