sarxos/webcam-capture #332: com.github.sarxos.webcam.WebcamDiscoveryService - Discovery will not run - driver MyCompositeDriver does not support this feature

henriquem, 2015-04-16:
When I replace webcam-capture-0.3.10-RC7.jar to 0.3.11-SNAPSHOT.jar 
I noticed that changed something:

    class MyCompositeDriver extends WebcamCompositeDriver {
    	public MyCompositeDriver() {
    		// add(new WebcamDefaultDriver()); // usb cam 
    		add(new IpCamDriver()); // ipcam 
    	}
    }

In version webcam-capture-0.3.11-SNAPSHOT.jar occur this message:

    INFO com.github.sarxos.webcam.WebcamDiscoveryService - Discovery will not run - driver MyCompositeDriver does not support this feature

I kept the version the driver web-ipcam: 0.3.10-RC7

sarxos, 2015-04-17:
Hi @henriquem,

Well, this is rather because ```IpCamDriver``` does not support discovery. Discovery Service has been designed to detect cameras that has been connected/disconnected from the USB and therefore, in case of IP cameras, which are HTTP enabled, this feature does not make much sense imho (because it may scan cameras registry which is static), however if if you thing it should be enabled to notify about new cameras found the registry (not sure if you implemented such feature), I can enable it in SNAPSHOT.

sarxos, 2015-04-17:
@henriquem, I just noticed out that you have also USB camera in composite driver. Didn't catch this earlier (coffee machine is broken, so you know, ehh), sorry. 

I will add discovery support in composite :)

kamenomagic, 2017-11-03:
I am in need of this feature--did it ever get implemented?

sarxos, 2017-11-03:
@kamenomagic, I completely forgot to add this :(

sarxos, 2017-11-03:
@kamenomagic, I added this feature but I had no time to test it. Feel free to verify it and fix if I made some mistake :)

sarxos, 2017-11-03:
I also released newest snapshot version ```0.3.12-SNAPSHOT``` which contains this change.

closed 2017-11-03 by sarxos

kamenomagic, 2017-11-03:
Awesome! I will try it out and let you know/fix it (if I can!). Will the 0.3.12-SNAPSHOT appear on the maven repository? (I was using gradle for 0.3.11, I'm not sure how to get the 0.3.12-SNAPSHOT). Thanks for the quick turn around.

sarxos, 2017-11-03:
@kamenomagic, this snapshot version is available from the Sonatype snapshot repository (as described on the project [main page](https://github.com/sarxos/webcam-capture)). To make it work you would have to add this repository into your Maven's POM file:

```xml
<repository>
	<id>Sonatype OSS Snapshot Repository</id>
	<url>http://oss.sonatype.org/content/repositories/snapshots</url>
</repository>
```

It can be added to Gradle as well, probably (since I do not use Gradle in any of my projects):

```
repositories {
  maven { url "http://oss.sonatype.org/content/repositories/snapshots" }
  mavenCentral()
}
```

_Note: I'm not very familiar with Gradle and this snipped may be incorrect. Please verify it with some reliable Gradle source of information._

kamenomagic, 2017-11-03:
That worked for the snapshot, and it looks like it's working as expected for discovery of webcams. Thanks so much!