sarxos/webcam-capture #333: capturing video with audio recording

afzaalhussain50, 2015-04-19:
hi, sarxos your webcam-capture api is outstanding, i record video using  this code

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-video-recording/src/main/java/com/github/sarxos/webcam/Encoder.java

but its record only video not audio, can you explain how i can record video+audio using your webcam-capture project??

Thanks

sarxos, 2015-04-20:
Hi @afzaalhussain50,

I haven't prepare any example working with the audio, but there is a fork crafted by @ryancheu demonstrating live video streaming together with audio.

https://github.com/ryancheu/webcam-capture

afzaalhussain50, 2015-04-22:
thanks @sarxos 
you are great man, i solve my problem :+1:

closed 2015-04-25 by sarxos

appnuri, 2015-04-25:
NAVER - http://www.naver.com/
--------------------------------------------

appnuri@appnuri.kr 님께 보내신 메일 <Re: [webcam-capture] capturing video with audio recording (#333)> 이 다음과 같은 이유로 전송 실패했습니다.

--------------------------------------------

받는 사람이 회원님의 메일을 수신차단 하였습니다. 

--------------------------------------------

Rustem81, 2016-12-26:
Hello, determine how to add the date and time in the video stream. How to organize a video archive and search frames in time intervals.
PS Sorry I use Google translator.