sarxos/webcam-capture #334: Multiple identical webcams and getting "Cannot start native grabber!"

zbrowning, 2015-04-21:
When attempting to open a second camera, I get the following exception. Any ideas?

My goal is to have multiple cameras open concurrently.

Thanks!

````java
12:13:23: New camera selected.
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Not enough args for boolean OpenIMAJGrabber.startSession(int, int, double)
12:13:24: Found camera!
12:13:24: Opening webcam 'Microsoft® LifeCam Cinema(TM) 0'
12:13:27: Camera open.
12:13:45: New camera selected.
12:13:45: Found camera!
12:13:45: Opening webcam 'Microsoft® LifeCam Cinema(TM) 1'
com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:271)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:232)
	at gui.TestGuiOCR.doGetImage(TestGuiOCR.java:432)
	at gui.TestGuiOCR.access$37(TestGuiOCR.java:398)
	at gui.TestGuiOCR$36.load(TestGuiOCR.java:1407)
	at mil.cryptomod.util.gui.LoaderVoid$1.load(LoaderVoid.java:52)
	at mil.cryptomod.util.gui.LoaderVoid$1.load(LoaderVoid.java:1)
	at mil.cryptomod.util.gui.Loader$LoadingWorker.doInBackground(Loader.java:37)
	at javax.swing.SwingWorker$1.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at javax.swing.SwingWorker.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber!
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:330)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more
12:13:55: Cannot execute task
````

zbrowning, 2015-05-06:
It appears to be happening when calling .setViewSize() on the second camera of same make/model. I can only set the custom view size on the first camera. Any other attempt gives the above error.

sarxos, 2015-05-07:
Hi @zbrowning,

The stack trace is clearly from ```open()``` not ```setViewSize()``` but to be honest, this is the first time I see this error when using only two cameras. What resolution did you set on both cameras?

zbrowning, 2015-05-07:
I'm setting it to 1280x720.
Before `open()`, I've got:
`cam.setCustomViewSizes(xxx); cam.setViewSize(xxx[0]);`
I can only ever call those on one of the cameras, or I get the above error when trying to open the second camera with the custom view size set right before opening.

Thanks!

sarxos, 2015-05-07:
What FPS your camera has on 1280x720 (max nominal value)? If it's equal or higher than 10 FPS, then you're probably hitting your USB capacity limit. If this is the case then lowering resolution should help. You can find more details on this issue at the end of this thread #139.

zbrowning, 2015-05-07:
It does appear I'm hitting the USB capacity limit. If I split the cameras between controllers, I no longer get the error. I'm going to pursue some USB3 webcams to take advantage of the bandwidth.

Thanks for pointing me in the right direction.

closed 2015-05-07 by zbrowning

sarxos, 2015-05-07:
@zbrowning, please note that I never tested USB 3.0 webcams with Webcam Capture API and I'm not 100% sure if they will work correctly with existing impl.