sarxos/webcam-capture #336: Capturing an instance of the Motion Detector

RenanFucci, 2015-04-23:
Hi Sarxos!

I'm having a problem when i try to use the method .getImage() that i used in connected cams in the IpCams. The goal of my algorithm is capture an image when motion detection occurs.
Could you give me an example ?

Thanks for the Lib by the way!

sarxos, 2015-04-24:
Hi @RenanFucci,

Here are some motion detection examples:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-motiondetector

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/DetectMotionExample.java

Please let me know if these are sufficient for your needs.

RenanFucci, 2015-04-24:
I've already seen this examples, but i need one to take shots of the IPcam.

sarxos, 2015-04-24:
Code will be exactly the same. Just use IP camera driver:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam

```java
Webcam.setDriver(new IpCamDriver());
```

RenanFucci, 2015-04-24:
That is the problem, the method .getImage() doesn't work. 

public class testeCamereIp extends JFrame{

	private static final long serialVersionUID = 1L;

	public static void main(String[] args) throws IOException,MalformedURLException {
		IpCamDeviceRegistry.register(new IpCamDevice("Lignano", "http://172.19.2.20/", IpCamMode.PUSH));
		new testeCamereIp();
	}

	public testeCamereIp() throws IOException,MalformedURLException {
		Webcam.setDriver(new IpCamDriver());
		Container c= getContentPane();
		System.out.println("1");
		//Instanciando outra camada do JPanel, responsável pelas etiquetas e caixa de textos
		JPanel cont= new JPanel();
		cont.add(new JLabel("-------------------"));
	
		System.out.println("3");
		//WebcamPanel webcm = new WebcamPanel(Webcam.getDefault());
		System.out.println("4");
		System.out.println(""+Webcam.getWebcams());
		Webcam webc= Webcam.getWebcams().get(0);
		webc.open();
		for (int i = 0; i < 3; i++) {
			try{
			System.out.println("gravando1");
			System.out.println("asasa");
			BufferedImage imagem = webc.getImage();
			System.out.println("Tam: "+imagem.getHeight());
			File file= new File("C:/ImagensPDI/Imagens/camera1/imagem.jpg");
			ImageIO.write(imagem, "jpg", file);
			}
			catch(Exception ex){
				ex.getMessage();
				}
		}
		
		//Adicionando todas as camadas acimas no painel principal C 
		c.add(cont);
	//	c.add(webcm);
		c.setLayout(new FlowLayout());

		setVisible(true);
		setLocationRelativeTo(null);
		setDefaultCloseOperation(EXIT_ON_CLOSE);  

		// TODO Auto-generated constructor stub
	}
}

sarxos, 2015-04-24:
Hi @RenanFucci, you know, "doesn't work" is a very general term. You could explain a little bit more how it does not work.

When you create such html document:

```html
<body>
<img src="http://172.19.2.20" />
</body>
```

And open it in Firefox - do you see expected image from webcam? If not, then the URL is incorrect and you should probably add something at the path (maybe it should be like ```http://172.19.2.20/video.cgi``` or something similar - this of course depends on your camera model).

RenanFucci, 2015-04-24:
I see the image of the webcam in this site: http://172.19.2.90/viewer.html

Even changing the url, when i say it doesn't work, i mean that the Jframe doesn't open and the method .getImage() that used before in connected cams, returns Null.

RenanFucci, 2015-04-24:
When i run this algorithm, nothing happens. The JFrame does not build itself. The buffer is null.
 
public class testeCamereIp {
	static {
		Webcam.setDriver(new IpCamDriver());
	}

	public static void main(String[] args) throws MalformedURLException {
		IpCamDeviceRegistry.register(new IpCamDevice("Lignano", "http://172.19.2.90/viewer.html", IpCamMode.PUSH));
		JFrame f = new JFrame("Live Views From Lignano Beach");
		f.add(new WebcamPanel(Webcam.getDefault()));
		f.pack();
		f.setVisible(true);
		f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		try{
			System.out.println("gravando1");
			System.out.println("asasa");
			BufferedImage imagem =  Webcam.getDefault().getImage();
			System.out.println("Tam: "+imagem.getHeight());
			File file= new File("C:/ImagensPDI/Imagens/camera1/imagem.jpg");
			ImageIO.write(imagem, "jpg", file);
		}
		catch(Exception ex){
			ex.getMessage();
		}
	}

}

sarxos, 2015-04-24:
Hi @RenanFucci,

I see the root cause of your problem - the URL you are putting into the ```IpCamDevice``` should be direct MJPEG image URL. You cannot put your camera panel address (http://172.19.2.90) there, or viewer address (http://172.19.2.90/viewer.html) since this will not work. It expect image URL.

Do find out the URL you need to inspect the image element on the web page with Firebug or any other inspector and find out what the image URL is (but **do not** use Internet Explorer - it sucks and its easier to eat own socks than do anything useful with this shitty browser), then put it in the ```IpCamDevice``` and everything should work fine.

I will walk you thru the Foscam demo, but since I don't know what camera you are using, you have to find the URL on your own.

1.
Go to http://50.197.211.181:8910 (user: user, password: foscam)

2.
Select Server Push Mode (this means MJPEG),

3.
Rightclick on image ans select "Inspect element" (I'm using Firefox),

[image]

4.
The above will open inspector panel where you can find out what the clicked image URL is:

[image]

5.
Add the address from the URL (http://50.197.211.181:8910/)

6.
Finally, the image URL in this example is:  

http://50.197.211.181:8910/videostream.cgi?user=user&pwd=foscam

The same procedure applies to your camera, but please note that the image **must be** the MJPEG stream since Webcam Capture API does not support h.264!

The code for the example above would be:

```java
IpCamDeviceRegistry.register(new IpCamDevice(
  "Lignano", 
  "http://50.197.211.181:8910/videostream.cgi?user=user&pwd=foscam",
  IpCamMode.PUSH));

closed 2015-04-24 by sarxos