sarxos/webcam-capture #340: What Means method setInertia from WebcamMotionDetector class ?

henriquem, 2015-05-11:
In WebcamMotionDetector class has the method called setInertia
For example:
    detectorMoviment.setInertia (1000); // keep "motion" state for 1 seconds

What means keep motion for  1 sec ? 
What means in terms of practices, when I will use motionDetected(WebcamMotionEvent event) from WebcamMotionListener ?

    @Override
    public void motionDetected(WebcamMotionEvent event){
        WebcamMotionDetector motion=(WebcamMotionDetector) event.getSource();
        BufferedImage imgCaptured= motion.getWebcam().getImage();
        // What will happen in imgCaptured ? will be refresh after 1 seconds ?
    }

For Example I would to wait the user found an position on camera for detect the face him...
So I would to ignore the first motion (when the face show at camera)

So like example: 
  - Wait he an position apropriated before I capture the good postiion from Face...
Wait  the user to position the face in a face mold, and after I capture image
I expected He will spend 1 seconds for position .

But I is my intention for use setInertia... method

sarxos, 2015-05-11:
Hi @henriquem,

When you call ```isMotion()``` in the 1000 ms interval after the motion was detection, the method will return ```true```. After this time it will return ```false```. From the code perspective - it's the time after when [motion flag](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/WebcamMotionDetector.java#L134) is reset from ```true``` to ```false```.

```java
detector.setInertia(1000); // set inertia to 1000 ms
detector.start();
detector.isMotion(); // false
.. // motion detected
detector.isMotion(); // true
Thread.sleep(250); 
detector.isMotion(); // true
Thread.sleep(250); 
detector.isMotion(); // true
Thread.sleep(250); 
detector.isMotion(); // true
Thread.sleep(250); // 1000 ms passed
detector.isMotion(); // false
```

If inertia is not set its value is calculated to be 2 * detection interval.

In the example as you proposed:

> Wait he an position apropriated before I capture the good postiion from Face... Wait the user to position the face in a face mold, and after I capture image I expected He will spend 1 seconds for position .

We can use inertia to wait for 1 second after the person stopped moving:

```java
detector.setInertia(1000); // keep motion flag for 1000 ms after motion stopped
detector.setInterval(100); // every 100 ms check if there is motion in the scene
```

When user is moving, the ```while (!detector.isMotion())``` loop will be true. It will be also true for 1000 ms after he stopped moving. Then, after 1000 ms passed, picture will be taken with ```getImage()```.

```java
while (!detector.isMotion()) {
  Thread.sleep(10); 
}
BufferedImage image = detector.getWebcam().getImage();
```

This setting has nothing with ```motionDetected(..)``` method which is invoked every time when motion is detected between two consecutive images.

I hope the above clarifies the issue with how this method works.

closed 2015-05-12 by sarxos

henriquem, 2015-05-12:
Hi @sarxos 
Thanks a lot for your explanation
where I could put this part of code ? Because did you comments: "This setting has nothing with motionDetected(..)"
Did you means, that I need to use:

     detector.getWebCam().getImage() 

instead of 
      
      ((WebcamMotionDetector) event.getSource()).getWebcam().getImage(); 

Example:

    @Override
    public void motionDetected(WebcamMotionEvent event) {
        while (!detector.isMotion()) {
            Thread.sleep(10); 
        }
        detector.getWebcam().getImage();  // get the image when stopped motion 
    }

Sorry my doubt about how implements it !

sarxos, 2015-05-12:
@henriquem,

The ```motionDetected(WebcamMotionEvent event)``` will be invoked **always** when motion in the scene is detected, no matter if inertia is set to 100, 200 or 5000, so changing inertia does not affect listener invocations.

The example you provided is incorrect. Assume that check interval is set to 100 ms, when object is in move, you will get 10 x ```motionDetected()``` invoked in 1 second and every one of these invocations will either block detector thread or will wait for stop motion and then take picture as many times as many there was events in the time when detector was running (which is maaaany).

For this particular purpose (wait for the object to stop moving) I would not use ```WebcamMotionListener``` at all because you are waiting for **no motion** instead of motion (and motion listener reports only motion, not the stop motion events). Therefore, the code to take a picture when object stopped moving after 1 second, would be something like this (pseudocode, hand crafted, not tested):

```java
detector = new WebcamMotionDetector(Webcam.getDefault());
detector.setInertia(1000); // how long motion is valid
detector.setInterval(100); // check motion every 100 ms
detector.start();
button = new JButton(new AbstractAction("take picture") {
  public void actionPerformed(ActionEvent e) {
    while (!detector.isMotion()) {
        Thread.sleep(10); 
    }
    image = detector.getWebcam().getImage();
    // write image somewhere
  }
});
```