sarxos/webcam-capture #341: loop searching devices and found devices

charlie23k, 2015-05-14:
Good job men.

this is my code and result but i don´t understand why enter in loop searching devices.

check the libs used in my project.

sorry for my bad english.

Thank you

[image]

  public static void comprobarCam(){
		Webcam webcam = Webcam.getDefault();
		System.out.println(webcam.getName());
		Dimension[] Resolution = webcam.getViewSizes(); 
		System.out.println(Arrays.toString(Resolution)); 
    }

16:46:01.642 [AWT-EventQueue-0] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
16:46:01.652 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:02.215 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
Logitech Webcam C210 0
[java.awt.Dimension[width=176,height=144], java.awt.Dimension[width=320,height=240], java.awt.Dimension[width=640,height=480]]
false
16:46:05.220 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:05.230 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:08.230 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:08.240 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:11.240 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:11.248 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:14.248 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:14.257 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:17.258 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:17.268 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:20.268 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:20.278 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:23.278 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:23.287 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:26.287 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:26.296 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:29.296 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:29.305 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:32.306 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:32.316 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:35.316 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:35.325 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:38.326 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:38.334 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:41.335 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:41.345 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:44.345 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:44.354 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:47.355 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:47.366 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0
16:46:50.366 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
16:46:50.374 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logitech Webcam C210 0

sarxos, 2015-05-14:
Hi @charlie23k,

Thanks :)

The loop is to detect connected / disconnected devices (this is also known as _webcam discovery service_). See [this example](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/WebcamDiscoveryListenerExample.java) for more details on this feature.

If you want to permanently disable it:

```java
Webcam.getDiscoveryService().setEnabled(false);
Webcam.getDiscoveryService().stop();
```

closed 2015-05-14 by sarxos

sarxos, 2015-05-14:
Just a note - you can disable this message by configuring logging framework you are using in your app.

charlie23k, 2015-05-16:
Thank you sarxos. If I have any problem ask you

mavirroco, 2016-03-03:
cool is what i was needing thanks a lot Sarxos!!