sarxos/webcam-capture #342: Unable to capute HD image.

goodrahstar, 2015-05-19:
Hi,
I am trying to capture HD image "HD720" but it shows.
Code:
 webcam.setViewSize(WebcamResolution.HD720.getSize());

Error:
Exception in thread "main" java.lang.IllegalArgumentException: Incorrect dimension [1280x720] possible ones are [176x144] [320x240] [640x480] 
	at com.github.sarxos.webcam.Webcam.setViewSize(Webcam.java:588)
	at WebcamAPI.main(WebcamAPI.java:14)

sarxos, 2015-05-19:
Hi @goodrahstar,

This example should shed some more light on your problem. Please follow the code and use ```setCustomViewSizes(..)``` method prior the ```setViewSize(..)``` is called.

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/CustomResolutionExample.java

The "standard" resolutions (in opposite to "custom") are QQVGA, QVGA and VGA, which means that all of them are supported by all manufactured UVC devices (with 99.9% probability). The HD720 is not among them - there is many devices which does not support it.

To cut the story short, to resolve your issue just do:

```java
Dimension[] nonStandardResolutions = new Dimension[] {
	WebcamResolution.HD720.getSize(),
};
Webcam webcam = Webcam.getDefault();
webcam.setCustomViewSizes(nonStandardResolutions);
webcam.setViewSize(WebcamResolution.HD720.getSize());
```

closed 2015-05-19 by sarxos

goodrahstar, 2015-05-19:
Hi @sarxos 
Thanks man for your quick response.
It works :)

sarxos, 2015-05-19:
Great :) Take care!