sarxos/webcam-capture #343: Raspberry pi camera board

charlie23k, 2015-05-20:
It´s posible to use your lib sarxos with raspberry pi camera module??

I try to work with this but sarxos lib don´t recognize the camera pi but if I put other camera in usb connector the lib recognize but work very slow with many jumps.

thank´s

sarxos, 2015-05-20:
Hi @charlie23k,

You will not achieve fantastic results with UVC (USB Video Device) due to GPIO/CPU limits of RasPi. 5 FPS with USB webcam is max what I was able to achieve in the past.

In regards to RasPi Camera Module, have you followed these instructions?

https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi

charlie23k, 2015-05-21:
Yes but throw an exception.

import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver;

public class WebcamPanelExample {
	
	static {
	     Webcam.setDriver(new V4l4jDriver()); 
	}

	public static void main(String[] args) throws InterruptedException {
		
	    System.out.println(System.getProperty("os.name").toLowerCase());
	    System.out.println(System.getProperty("os.arch").toLowerCase());
	    System.out.println(System.getProperty("os.version").toLowerCase());
		
		
		
		Webcam webcam = null;
		webcam = Webcam.getDefault();
		System.out.println(Webcam.getDefault());
		WebcamPanel panel = new WebcamPanel(webcam);
		JFrame window = new JFrame("Test webcam panel");
		window.add(panel);
			window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
			window.pack();
			window.setVisible(true);
	}
}

and the result is the next:

Exception in thread "main" java.lang.NoClassDefFoundError: com/github/sarxos/v4l4j/V4L4J
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.<clinit>(V4l4jDriver.java:33)
	at id.samplecam.WebcamPanelExample.<clinit>(WebcamPanelExample.java:16)
Caused by: java.lang.ClassNotFoundException: com.github.sarxos.v4l4j.V4L4J
	at java.net.URLClassLoader$1.run(URLClassLoader.java:366)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:423)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:356)
	... 2 more

thank´s

charlie23k, 2015-05-21:
I try to work with sarxos lib with raspberry-pi camera module. its possible??

sarxos, 2015-05-22:
@charlie23k,

Yes, it's possible to use Webcam Capture API with RasPi Camera Module.

The ```NoClassDefFoundError``` clearly states that you are missing some classes. This is because you haven't add **required** JARs into your application's classpath.

On the page I linked in the previous post:

https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi

There is a statement:

> The following JARs are **required**:
* The webcam-capture-driver-v4l4j JAR can be found **[here](https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture-driver-v4l4j/0.3.11-SNAPSHOT)**, and
* The V4L4J JAR itself which can be found **[here](http://search.maven.org/#artifactdetails|com.github.sarxos|v4l4j|0.9.1-r507|jar)** (the project and source code is available [here](https://github.com/sarxos/v4l4j)).

Which means that you **have to** download them and add to your application's classpath. Otherwise the code will not work and will rise exception same as yours. You are missing JAR from second bullet (V4L4J). It can be found here:

http://search.maven.org/#artifactdetails|com.github.sarxos|v4l4j|0.9.1-r507|jar

closed 2015-05-22 by sarxos