sarxos/webcam-capture #345: Exception in thread main

fiyaz05, 2015-05-23:
hi sarxos, you have done a great job man. I ran your project in my laptop with windows 7 with 64 bit processor ,it worked fine, i had no issues with it. I ran your project in PC with windows 8 with 64 bit processor ,it worked cool. i had no issues with it. And I ran your project in Acer tablet with windows 8 with 32 bit Operating System 64-based processor,it worked good. But when i tried your project with Chroma tablet with windows 8 with 32 bit Operating System 64-based processor. it showed this Errors:

C:\Users\htic\Desktop>java -jar fizz.jar
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further detail
s.
LIBTBD ERROR: Data is not tagged properly!
LIBTBD ERROR: Data is not tagged properly!
LIBTBD ERROR: Data is not tagged properly!
LIBTBD ERROR: Data is not tagged properly!
LIBTBD ERROR: Data is not tagged properly!
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot exec
ute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(Webc
amProcessor.java:72)
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java
:140)
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.ja
va:20)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:302)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:251)
        at com.github.sarxos.webcam.Webcam.open(Webcam.java:236)
        at web.Web.main(Web.java:14)
Caused by: com.github.sarxos.webcam.WebcamException: Cannot start native grabber
!
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDe
faultDevice.java:335)
        at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.
java:38)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamPr
ocessor.java:81)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.
java:1145)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor
.java:615)
        at java.lang.Thread.run(Thread.java:745)

CAN YOU PLEASE HELP ME OUT .

sarxos, 2015-05-23:
Hi @fiyaz05,

Thank you for the report. 

In the past people reported problems running Webcam Capture API on Windows shipped on Intel Atoms tablets, but this time the culprit may be this error:

```
LIBTBD ERROR: Data is not tagged properly!
```

It comes from the component installed in the OS, not Webcam Capture API. I found the source of the error from log:

https://github.com/01org/libxcam/blob/master/xcore/libtbd.c#L311

But I'm completely unfamiliar with this library and have really no idea why it's used on your system.

Txeni, 2015-08-22:
Hi guys,

First thanks @sarxos for the amazing work, it saved my ass with my laptop camera, using it in processing and it works great.

On my Dell tablet, 32bits windows 64bits windows 8.1 I get exactly the same errors messages as @fiyaz05, you said that is an OS problem, since it says "cannot start native grabber", do you think it is worh it trying other capture frameworks different of the native one? opencv, gstreamer, etc...

Thanks and sorry for asking, instead of trying directly, I'm running out of time for this project and I need to prioritice where I spend it...

sarxos, 2015-08-24:
Hi @Txeni,

Yes, you can try [webcam-capture-driver-javacv](https://github.com/sarxos/webcam-capture#javacv-driver) working on top of [OpenCV](http://opencv.org/) or [webcam-capture-driver-jmf](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-jmf) running on top of [Oracle JMF](http://www.oracle.com/technetwork/java/javase/download-142937.html).

From these two I would start with JMF.