sarxos/webcam-capture #347: Native MJPEG,YUV420p , YUY2 support?

jduke32, 2015-05-25:
Hello , i have logitech c910 webcam , it gives MJPEG support natively.
Java Webcam Capture API may have native MJPEG support  with webcams?

Also some webcams natively support YUV420p output.
That will be good for VP8 , H264 encoding because all require YUV420P.

there is also YUY2 encoding support natively by webcams.
That is easier to convert YUY2 => YUV420p instead of RGB.

These are the things required much for realtime webcam processing.

For example : 
Webcam webcam = Webcam.getDefault();
WebcamFormat formats[] = webcam.getNativeFormats();
WebcamFormat formatFound;
for(WebcamFormat format : formats) {
       System.out.println(format);
       if(format == WebcamFormat.YUV420P) {
               formatFound = WebcamFormat.YUV420P;
       }
}
ByteBuffer yuv420pData;
if(formatFound == WebcamFormat.YUV420P) {
       yuv420pData = webcam.getImageByteBuffer(formatFound);
      
} else if(formatFound == WebcamFormat.RGB24) {
       ByteBuffer rgbData = webcam.getImageByteBuffer(formatFound);
       yuv420pData = GPUUtils.RGB24toYUV420P(rgbData);
}

//send YUV420p data to FFMPEG
VideoFrame frame = encoder.encodeVideo(yuv420pData,picture);

Thanks :)

sarxos, 2015-05-25:
Hi @jduke32,

This would be indeed very useful feature :) The problem currently lies in the format detection - on Linux this can be achieved pretty easily (and it's already done on top of [v4l4j](https://code.google.com/p/v4l4j/), please see [V4l4jDevice](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-v4l4j/src/main/java/com/github/sarxos/webcam/ds/v4l4j/V4l4jDevice.java) to check how it finds the most optimal format), but on Windows it's paint in the arse so I'm considering to move from [BridJ](https://github.com/nativelibs4java/BridJ) to [javacpp](https://github.com/bytedeco/javacpp) to gain possibility to access low-level [videoinput](https://github.com/ofTheo/videoInput) API via [videoinput-preset](https://github.com/bytedeco/javacpp-presets/tree/master/videoinput) (on Windows) which is a  must if we ever consider adding formats on this platform. I still have to find out how to translate format detection from C++ to Java without a need of JNI.

jduke32, 2015-05-25:
Need GPUUtils with the help of AMD Aparapi.
Future is GPU its better to use AMD Aparapi for that.
JavaCPP is very powerful and most promising JNI API its far better then BridJ because it has zero overhead java and jni

jduke32, 2015-05-25:
Windows needs DirectShow for that i have some source code i can share with you

jduke32, 2015-05-25:
#define MIN(x,y) (x)<(y) ? (x) : (y);

void YUY2toRGB24(BYTE *pDst, BYTE *pSrc) {
	for(long lSrc=0,lDst=0; lSrc<320*240*2; lSrc=lSrc+4,lDst=lDst+6) {
		int y,u,v;
		y = *(pSrc+lSrc+0);
		u = *(pSrc+lSrc+1);
		v = *(pSrc+lSrc+3);
		*(pDst+lDst+2) = MIN((9535 * (y - 16) + 13074 * (v - 128)) >> 13, 255);
		*(pDst+lDst+1) = MIN((9535 * (y - 16) - 6660 * (v - 128) - 3203 * (u - 128)) >> 13, 255);
		*(pDst+lDst+0) = MIN((9535 * (y - 16) + 16531 * (u - 128)) >> 13, 255);
		y = *(pSrc+lSrc+2);
		*(pDst+lDst+5) = MIN((9535 * (y - 16) + 13074 * (v - 128)) >> 13, 255);
		*(pDst+lDst+4) = MIN((9535 * (y - 16) - 6660 * (v - 128) - 3203 * (u - 128)) >> 13, 255);
		*(pDst+lDst+3) = MIN((9535 * (y - 16) + 16531 * (u - 128)) >> 13, 255);
	}
}

This needs GPU Accellerated solution , can someone help with that for AMD Aparapi ?

jduke32, 2015-05-25:
INT InitWebcam( INT id ) {
	g_bWebcamInit	= FALSE;
	SampleGrabberEx*	sample_grabber = NULL;
	
	HRESULT hr;
	
	hr = CoCreateInstance (CLSID_FilterGraph, NULL , CLSCTX_INPROC , IID_IGraphBuilder , (void **) &g_pGraph );
	
	if (FAILED(hr)) {
		return 1;
	}
	
	
	
	// Create the capture graph builder
	hr = CoCreateInstance (CLSID_CaptureGraphBuilder2 , NULL, CLSCTX_INPROC,IID_ICaptureGraphBuilder2, (void **) &g_pCapture);
	
	if (FAILED(hr)) {
		g_pGraph->Release();
		return 2;
	}
	
	
	
	// Obtain interfaces for media control and Video Window
	hr = g_pGraph->QueryInterface(IID_IMediaControl,(LPVOID *) &g_pMC);
	if (FAILED(hr)) {
		g_pCapture->Release();
		g_pGraph->Release();
		return 3;
	}
	
	
	
	hr = g_pGraph->QueryInterface(IID_IVideoWindow, (LPVOID *) &g_pVW);
	if (FAILED(hr)) {
		g_pMC->Release();
		g_pCapture->Release();
		g_pGraph->Release();
		return 4;
	}
	
	
	
	hr = g_pGraph->QueryInterface(IID_IMediaEvent, (LPVOID *) &g_pME);
	if (FAILED(hr)) {
		g_pVW->Release();
		g_pMC->Release();
		g_pCapture->Release();
		g_pGraph->Release();
		return 5;
	}
	
	
	g_pVW->put_Visible(OAFALSE);

	hr = NULL;
	g_pME->SetNotifyFlags( 0 );
	
	IBaseFilter *pSrcFilter=NULL;
	hr = g_pCapture->SetFiltergraph(g_pGraph);
	
	if(FAILED(hr)) {
		g_pVW->Release();
		g_pMC->Release();
		g_pCapture->Release();
		g_pGraph->Release();
		return 2;
	}
	
[... 225 lines cut ...]
	
	hr = g_pCapture->RenderStream(NULL, NULL, pSrcFilter, NULL , NULL );
	if(FAILED(hr)) {
		g_pVW->Release();
		g_pMC->Release();
		pSrcFilter->Release();
		g_pGrabber->Release();
		g_pCapture->Release();
		g_pGraph->Release();
		delete g_lpSampleGrabber;
		g_lpSampleGrabber = NULL;
		return 14;
	}
	
	
	g_pSrcFilter = pSrcFilter;
	g_pVW->put_Visible(OAFALSE);
	g_pVW->put_AutoShow(OAFALSE);
	g_pMC->Stop();
	//g_bSendCam = true;
	
	g_bWebcamInit	= TRUE;
	
	return 0;
}

jduke32, 2015-05-25:
on DirectShow , you can enumerate all formats with
while(pEnum->Next(1,&pmt,fetch)==S_OK);
and select yuv420p or yuy2 from there

jduke32, 2015-05-25:
I will  send you some library

sarxos, 2015-05-25:
Great, that will be very helpful :)

jduke32, 2015-05-27:
http://www.bayner.com/jduke32/WebcamAPI.zip
http://www.bayner.com/jduke32/Libs.zip

These 2 files , Libs.zip have the required lib files you need 

This win32 project will give you ability to get native MJPEG , YUV420p , YUY2 from Device.
It uses DirectShow API.

I did on visual studio 2013.

If you have any questions , let me know.

jduke32, 2015-05-27:
Webcam webcam = Webcam.getDefault();
WebcamFormat formats[] = webcam.getNativeFormats();
WebcamFormat formatFound;
for(WebcamFormat format : formats) {
       System.out.println(format);
       if(format == WebcamFormat.YUV420P) {
               formatFound = WebcamFormat.YUV420P;
       }
}
ByteBuffer yuv420pData;
if(formatFound == WebcamFormat.YUV420P) {
       yuv420pData = webcam.getImageByteBuffer(formatFound);

} else if(formatFound == WebcamFormat.RGB24) {
       ByteBuffer rgbData = webcam.getImageByteBuffer(formatFound);
       yuv420pData = GPUUtils.RGB24toYUV420P(rgbData);
}

//send YUV420p data to FFMPEG
VideoFrame frame = encoder.encodeVideo(yuv420pData,picture);

jduke32, 2015-05-27:
We can also add direct h264 stream from webcam
WebcamFormat format[] =  Webcam.getNativeFormats();
WebcamFormat formatFound;

ByteBuffer videoData = Bytebuffer.allocateDirect(65000);

for(WebcamFormat format : formats) {
       System.out.println(format);
       if(format == WebcamFormat.H264) {
               formatFound = WebcamFormat.H264;
               
       }
}

if(formatFound == WebcamFormat.H264) {
      //webcam h264 specific config
      WebcamFormatConfiguration formatConfig = new WebcamFormatConfiguration();
      formatConfig.setParam("bitpersecond","1000000");
      formatConfig.setParam("setGopSize","100");
      webcam.configureFormat(WebcamFormat.H264,formatConfig);
      webcam.getVideoData(videoData);
}

jduke32, 2015-05-27:
You can delete dependency 
E:\KBD\Libs\C\Passive\PSDK\Lib
it just had the libs i gave you
http://www.bayner.com/jduke32/Libs.zip

jduke32, 2015-05-27:
Did you get the datas?

sarxos, 2015-05-27:
Yes, I downloaded ```Libs.zip``` and have it locally. I understand these libs are linked with the source code you posted above.

jduke32, 2015-05-27:
Yes now the code i gave you has abilities

jduke32, 2015-05-28:
http://www.bayner.com/jduke32/WebcamAPI.zip

jduke32, 2015-05-28:
can you implement it now to Java Webcam ?
do you need something more ?

sarxos, 2015-05-28:
@jduke32, thank you for the project. It's really helpful! I will start working on this as soon as I have some free time to donate into the project. Currently I'm pretty busy at work.

jduke32, 2015-05-28:
There is also OSX solution.
https://developer.apple.com/library/ios/samplecode/GLCameraRipple/Introduction/Intro.html#//apple_ref/doc/uid/DTS40011222-Intro-DontLinkElementID_2

GLCameraRipple sample code  , that allows IOS to capture YUV frames directly from Camera.
H264 , VP8 some important codecs requires YUV420p.

I will investigate for the problem more.
Do you have OSX solution  ?

jduke32, 2015-11-18:
6 months passed and no news ? :(

sarxos, 2015-11-18:
Hi @jduke32, because of my responsibilities at work and my newborn son I haven't had much time for Webcam capture API project. I wish I could relax and implement some nice webcam features, but lately I'm constantly busy...

jduke32, 2015-12-27:
do you need any help for this ?
i can help you for that feature

sarxos, 2015-12-28:
Hi @jduke32,

Lately I'm pretty busy at work managing my team and moving project forward. This, together with my children, consumes most of my time and makes me unable to donate much time into the project like I did before. I hope you understand.

If you are willing to implement such enhancement, please go ahead, but I suppose this may be hard and possible error-prone due to the necessity of having new native layer implemented. I hoped to utilize [VideoInput](https://github.com/ofTheo/videoInput) low level capabilities since it's well tested and used in the wild for a many years now, and there are Java binding for it available for free.

eduramiba, 2021-06-17:
Hi,

Please try the new native driver that I just released: https://github.com/eduramiba/webcam-capture-driver-native

It should be able to correctly use native formats under Windows.

eix128, 2021-06-17:
getting h264 natively from webcam directly would be good solution.
Hope to see it soon!

eduramiba, 2021-06-17:
@eix128 The webcam should just work with the best resolution if you use this driver.