sarxos/webcam-capture #348: Raspberry-pi camera board render hardware

charlie23k, 2015-05-25:
Thank´s 

I use your api sarxos with raspberry-pi camera board but i had bad results because take photos instead of video. low fps.

The api sarxos render via software but when you use with the raspberry the microchip is very reduced and for this reason the raspberry-pi need render video via hardware.

It´s possible to do render video via hardware using sarxos api with raspberrypi camera board ??

thank´s
best regards.

sarxos, 2015-05-25:
Hi @charlie23k,

The Webcam capture API has been created to have a uniform way to access frames from the camera device, not RAW video stream, and therefore it does not use h.264 video encoding acceleration, available on Pi, but formats available from ```v4l2``` device exposed by ```bcm2835-v4l2``` driver.

In regards to video rendering - the only place where image is being rendered is a ```WebcamPanel``` which reacts to the new frames and updates the image on screen. This solution uses acceleration features exposed by Java in DD/GDI pipeline and cannot go deeper. But still, as you said, this is software solution when it comes to reading images from camera because RAW data from camera buffer is being copied to ```BufferedImage``` of Java.

Maybe [FFmpeg](https://www.ffmpeg.org/) or [GStreamer](http://gstreamer.freedesktop.org/) is a better solution for your purpose if you are looking for a way to access raw video stream to be transcoded and sent to remote sink, or [recompiled VLC](http://www.oblivion-software.de/index.php?id=56&type=98) if you need video to be rendered directly on the Pi. Java is a bit tricky when it comes to hardware acceleration - e.g. you touch the image buffer once and the whole image is not accelerated any more (VRAM is not used).

closed 2015-05-27 by sarxos

zhangkeke, 2015-05-27:
.....

sarxos, 2015-05-27:
Just FYI, if you need to access RAW bytes that comes from camera (instead of having to deal with ```BufferedImage```) you can use:

```java
public ByteBuffer getImageBytes()
```

To get access to direct ```ByteBuffer``` which points to natively allocated memory, or:

```java
public void getImageBytes(ByteBuffer target)
```

To copy bytes into the ```target``` buffer (direct or non-direct depending on what type you've created).