sarxos/webcam-capture #349: BufferedImage sucks for realtime video

jduke32, 2015-05-27:
its better to mix webcam capture api with libgdx.
Libgdx texture is so fast enough to handle realtime webcam operations.
OpenGL , OpenCL mixing.
Texture texture;
webcam.getImage(texture);

jduke32, 2015-05-27:
i do it with;

    public void pushVideoFrameFromQueueToTexture(Texture texture, AtomicBoolean initTexture) {
        final VideoFrame picture = pictures.poll();
        if(picture != null) {
            /*long delay = millisecondsUntilTimeToDisplay(picture);
            try
            {
                if (delay > 0) {
                    Thread.sleep(delay);
                }
            }
            catch (InterruptedException e)
            {
                return;
            }*/
            //long timeStamp = picture.getTimeStamp();
            //System.out.println(l+" "+picture.getTimeStamp());
            final ByteBuffer pictureByteBuffer = picture.getData();
            //long timestamp = picture.getTimestamp();

            final ByteBuffer byteBuffer = pictureByteBuffer;

            final int textureObjectHandle = texture.getTextureObjectHandle();

            glBindTexture(GL_TEXTURE_2D, textureObjectHandle); //Bind texture ID
            //Send texel data to OpenGL
            //if (init)
            final int width = picture.getWidth();
            final int height = picture.getHeight();
            //glPixelStorei(GL_UNPACK_ALIGNMENT, 1);
            //GL11.glPixelStorei(GL11.GL_UNPACK_ALIGNMENT,1);
            if(!initTexture.get())
                GL11.glTexImage2D(GL20.GL_TEXTURE_2D, 0, GL11.GL_RGB, width , height , 0, GL11.GL_RGB , GL11.GL_UNSIGNED_BYTE, byteBuffer);
            else
                GL11.glTexSubImage2D(GL20.GL_TEXTURE_2D, 0, 0, 0, width , height , GL11.GL_RGB , GL11.GL_UNSIGNED_BYTE, byteBuffer);
            initTexture.set(true);
        }
    }

jduke32, 2015-05-27:
is there any faster way ?

jduke32, 2015-05-27:
AtomicBoolean initTexture <= this is checking for texture first time run or not.
AtomicBoolean initTexture <= initial value should be false

sarxos, 2015-05-27:
Hi @jduke32,

You can use:

```java
public ByteBuffer getImageBytes()
```

To get access to direct ```ByteBuffer``` which points to natively allocated memory, or:

```java
public void getImageBytes(ByteBuffer target)
```

To copy bytes into the ```target``` (direct or non-direct depending on what type you've created).

This above methods does not involve ```BufferedImage``` creation if ```WebcamDevice``` instance implements ```BufferAccess```, which is this is true for ```WebcamDefaultDevice```:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/ds/buildin/WebcamDefaultDevice.java#L33

But e.g. ```GStreamerDevice``` does not support buffer access:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-gstreamer/src/main/java/com/github/sarxos/webcam/ds/gstreamer/GStreamerDevice.java#L30

sarxos, 2015-05-27:
Just FYI, the ```ByteBuffer``` data is RGB24.

closed 2015-05-27 by sarxos