sarxos/webcam-capture #352: TEMP directory out of space disk

jduke32, 2015-06-02:
Hi , When i tried to use webcam-capture library.
I saw , everytime everyrun of webcam-capture library is producing files  to TEMP directory.
I was using TEMP directory as a ramdisk.

When i got out of space on ram disk TEMP directory.
I started to saw error shown below :

D:\Tools\JDK\1.8\x64\bin\java -agentlib:jdwp=transport=dt_socket,address=127.0.0.1:10890,suspend=y,server=n -Dlogback.statusListenerClass=ch.qos.logback.core.status.NopStatusListener -Djava.library.path=C:\Users\Kadir\IdeaProjects\WebcamTransmitter\nativeLib\windows\x64 -Dfile.encoding=windows-1252 -classpath "D:\Tools\JDK\1.8\x64\jre\lib\charsets.jar;D:\Tools\JDK\1.8\x64\jre\lib\deploy.jar;D:\Tools\JDK\1.8\x64\jre\lib\javaws.jar;D:\Tools\JDK\1.8\x64\jre\lib\jce.jar;D:\Tools\JDK\1.8\x64\jre\lib\jfr.jar;D:\Tools\JDK\1.8\x64\jre\lib\jfxswt.jar;D:\Tools\JDK\1.8\x64\jre\lib\jsse.jar;D:\Tools\JDK\1.8\x64\jre\lib\management-agent.jar;D:\Tools\JDK\1.8\x64\jre\lib\plugin.jar;D:\Tools\JDK\1.8\x64\jre\lib\resources.jar;D:\Tools\JDK\1.8\x64\jre\lib\rt.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\access-bridge-64.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\cldrdata.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\dnsns.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\jaccess.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\jfxrt.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\localedata.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\nashorn.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\sunec.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\sunjce_provider.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\sunmscapi.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\sunpkcs11.jar;D:\Tools\JDK\1.8\x64\jre\lib\ext\zipfs.jar;C:\Users\Kadir\IdeaProjects\WebcamTransmitter\target\classes;C:\Users\Kadir\.m2\repository\com\github\sarxos\webcam-capture\0.3.10\webcam-capture-0.3.10.jar;C:\Users\Kadir\.m2\repository\com\nativelibs4java\bridj\0.6.2\bridj-0.6.2.jar;C:\Users\Kadir\.m2\repository\org\slf4j\slf4j-api\1.7.2\slf4j-api-1.7.2.jar;C:\Users\Kadir\.m2\repository\org\bytedeco\javacpp\0.11\javacpp-0.11.jar;C:\Users\Kadir\.m2\repository\org\bytedeco\javacpp-presets\ffmpeg\2.6.1-0.11\ffmpeg-2.6.1-0.11.jar;C:\Users\Kadir\.m2\repository\org\bytedeco\javacpp-presets\ffmpeg\2.6.1-0.11\ffmpeg-2.6.1-0.11-windows-x86.jar;C:\Users\Kadir\.m2\repository\ch\qos\logback\logback-classic\1.1.3\logback-classic-1.1.3.jar;C:\Users\Kadir\.m2\repository\ch\qos\logback\logback-core\1.1.3\logback-core-1.1.3.jar;C:\Program Files (x86)\JetBrains\IntelliJ IDEA 14.1.3\lib\idea_rt.jar" service.WebcamCapture
Connected to the target VM, address: '127.0.0.1:10890', transport: 'socket'
java.lang.RuntimeException: Failed to initialize BridJ
	at org.bridj.Platform.initLibrary(Platform.java:370)
	at org.bridj.Platform.<clinit>(Platform.java:175)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.UnsatisfiedLinkError: no bridj in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1865)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at org.bridj.Platform.initLibrary(Platform.java:355)
	... 7 more
com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:879)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:856)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:834)
	at service.WebcamCapture.restart(WebcamCapture.java:171)
	at service.WebcamCapture.run2(WebcamCapture.java:244)
	at service.WebcamCapture.run(WebcamCapture.java:150)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.util.concurrent.FutureTask.get(FutureTask.java:192)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 8 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:118)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	... 1 more
Caused by: java.lang.UnsatisfiedLinkError: org.bridj.Platform.sizeOf_ptrdiff_t()I
	at org.bridj.Platform.sizeOf_ptrdiff_t(Native Method)
	at org.bridj.Platform.<clinit>(Platform.java:179)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more
Disconnected from the target VM, address: '127.0.0.1:10890', transport: 'socket'

Process finished with exit code -1

sarxos, 2015-06-02:
Hi @jduke32,

Indeed, but this is not Webcam Capture API library, but BridJ which is extracting native objects to ```/tmp``` to later link them into runtime. If there is no space left on ```/tmp``` it will raise ```UnsatisfiedLinkError``` due to native object missing.

However, this is just how BridJ is working and I doubt this can be changed easily. The best option I can see now it to check if ```/tmp``` size is sufficient for the extraction needs and display some nice exception saying _"No space left in /tmp"_ instead of this ugly-and-nothing-explaining error you caught. The best place to implement such feature is the BridJ itself, imho, since I would like to avoid adding such dependency in Webcam Capture API.

jduke32, 2015-06-02:
or java code automatically clear temp folder.
And there is 1 more thing.
Everytime extracting jni dlls to unique folder bad idea.
Extract once to same folder better idea.

sarxos, 2015-06-05:
I partially agree, but this is just how BridJ works. I cannot change this.

rahul189git, 2015-06-12:
The bridj only expects the absolute path of temp folder and when we pass in relative path, it throws error as below:

 Failed to initialize BridJ (java.lang.UnsatisfiedLinkError: Expecting an absolute path of the library: ..\app\tmp\BridJExtractedLibraries2194816147530680582\bridj.dll)

Please convert the temp path to absolute path.

sarxos, 2015-06-12:
Just to follow up with @rahul189git comment. This can be done by setting ```java.io.tmpdir``` property at the ```java``` command line, e.g.:

```
java -Djava.io.tmpdir=/your/app/path -cp libs/* -jar my.jar MyMain.java
```

This will not work in case of Applets.

rahul189git, 2015-06-15:
Thanks Sarxos. Ours is a big solution and currently it is passing relative path as jvm arguments. The change to pass absolute value will require  big changes, which is why I asked to change relative path to absolute one at bridj end itself. Such exceptions should not be thrown to end user as this can be handled at user end. Why the restriction to path only absolute path.

sarxos, 2015-06-15:
Hi @rahul189git,

> The change to pass absolute value will require big changes, which is why I asked to change relative path to absolute one at bridj end itself.

Olivier, who manage [BridJ project](https://github.com/nativelibs4java/BridJ) is really open for fixes and enhancement and I'm sure he will merge such change when you send him a pull request.

> a better solution should to separate bridj dll and jar and give both two independently as part of build

This is better solution for more advanced users, but in reality most of the people wants to have a single JAR with possibility to run it everywhere, on all systems, without a need of additional DLL/SO files being installed on the system, especially when this must be done by beginners or non-technical users, which actually knowns nothing about how the software is build.

> Let end user place the jar and dll himself in the lib directory him/herself. This way the temp folder handling will go away.

I think it's enough to put BridJ DLL/SO in the application working directory to bypass extraction from JAR, so from the functional point of view BridJ already supports it. I tested this once on my RasPi long time ago with BridJ 6.2 release (and it was working), but you have to verify this BridJ feature on BridJ 7.x. Same can be done with OpenIMAJGrabber DLL/SO.

rahul189git, 2015-06-15:
I am not aware of the process as I am new to git. You mean that should I make changes and inform Olivier. I can do that but then where will I get the latest build with this change.

sarxos, 2015-06-15:
@rahul189git,

>  am not aware of the process as I am new to git.

The process looks like the following:

1. Fork repository to your account,
2. Clone repository to your PC (using standard ```git clone```),
3. Change the code on your PC,
4. Commit and push to your forked repo,
5. Create pull request by clicking _Create pull request_ button on the commit (it appears whenever you push your changes to fork).

> You mean that should I make changes and inform Olivier.

Yes. The project owner will be notified about new pull request and will be able to merge your changes into his project code.

You can also create new issue which will describe what and how should be changed, but in such a case this still needs to be implemented by the project owner, thus introducing delays in the process if he or her does not have enough time to donate.

> where will I get the latest build with this change

This depend on the project. E.g. I used to release my JARs to Maven Central on every major release, and put all minor ones in my private repo which is also available for the public. 

Olivier may give you more details on his process, but I noticed that there are fewer major releases in BridJ and current HEAD can be found in the [Sonatype repository](https://oss.sonatype.org/content/repositories/snapshots/com/nativelibs4java/bridj/).

rahul189git, 2015-06-15:
Thanks a lot Sarxos. Very helpful.