sarxos/webcam-capture #353: Webcam close method sometimes not working

jduke32, 2015-06-02:
Hi , i tried webcam.close(); to close webcam.
But sometimes this method is not working.

I made this solution and it worked:
        while (webcam.isOpen())
            webcam.close();

sarxos, 2015-06-02:
hi @jduke32,

Thanks for the report. Do you have any errors from the console, or debug messages from logback? I'm asking because when you take a look [into the code](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/Webcam.java#L367) you will notice that first invocation of ```close()``` sets atomic ```open``` to ```false```, so all next invocations are simply ignored. The only place where this flag is reverted back to ```true``` is [line 381](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/Webcam.java#L381) where we ignore it because of ```InterruptedException``` being thrown from the background. The simplest way to check this is to enable debug logs and verify if it was closed or not and what is actually happening beneath the API surface.

jduke32, 2015-06-02:
this time my code hang on webcam.close();

jduke32, 2015-06-02:
i tried solution
WebcamCloseTask task = new WebcamCloseTask(
                webcam.getDriver(),
                webcam.getDevice() );
        try {
            task.close();
        } catch (InterruptedException e) {
        }

But its very unstable.
webcam.getimage returns null now...

jduke32, 2015-06-02:
yea , i was using thread.interrupt that cause a bug on webcam.capture i think.
Now again i am testing without any problem

sarxos, 2015-06-05:
Great. There were issues with multithreading on windows (native objects failure) so I had to implement this weird ```WebcamProcessor``` (to have native layer accessed by a single thread), but it seems to be failing in some edge cases.

I'm glad the issue is now resolved.

closed 2015-06-05 by sarxos