sarxos/webcam-capture #355: VlcjDriver with protocol RTSP

henriquem, 2015-06-11:
I have read about VLC and I checked there is an library called Caprica has support for support RTSP,
But I founded here the VlcjDriver which uses vlcj library from Caprica, Bingo!!

So In examples avaliable , How I could implements using Protocol RTSP ?

For Example using this url:

     rtsp://streaming1.osu.edu/media2/ufsap/ufsap.mov  

In VLC this address works fine!
I work with IPCamDriver, and I would add VlcjDriver for capture the image.
I Have an class  implemented the interface WebcamMotionListener 

     @Override
	public void motionDetected(WebcamMotionEvent event) {
		WebcamMotionDetector motion=(WebcamMotionDetector) event.getSource();
			BufferedImage captured = motion.getWebcam().getImage(); //came from RTSP !

How I register the RTSP:// address in VlcjDriver ?

Thks
Henrique

sarxos, 2015-06-15:
Hi @henriquem,

The bad news is the fact that this is not supported by default, but I have also a good one - I managed to craft small PoC working with the RTSP URI which should satisfy your needs for now. Another bad news is the fact it forces audio to be enabled (it hangs when I supply it with ```--no-audio``` option, I didn't manage to w/a this issue).

Here is the example code:

```java
import java.util.ArrayList;
import java.util.Arrays;

import javax.swing.JFrame;

import uk.co.caprica.vlcj.medialist.MediaListItem;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;
import com.github.sarxos.webcam.ds.vlcj.VlcjDriver;

public class WebcamPanelExample {

	static {
		String name = "mov";
		String rtsp = "rtsp://streaming1.osu.edu/media2/ufsap/ufsap.mov";
		Webcam.setDriver(new VlcjDriver(Arrays.asList(new MediaListItem(name, rtsp, new ArrayList<MediaListItem>()))));
	}

	public static void main(String[] args) throws InterruptedException {

		Webcam webcam = Webcam.getWebcams().get(0);
		webcam.setViewSize(WebcamResolution.VGA.getSize());

		WebcamPanel panel = new WebcamPanel(webcam);
		panel.setFPSDisplayed(true);
		panel.setImageSizeDisplayed(true);

		JFrame window = new JFrame("Webcam Panel");
		window.add(panel);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}
```

And the modified ```VlcjDevice``` class:

```java
package com.github.sarxos.webcam.ds.vlcj;

import java.awt.Dimension;
import java.awt.Transparency;
import java.awt.color.ColorSpace;
import java.awt.image.BufferedImage;
import java.awt.image.ComponentColorModel;
import java.awt.image.ComponentSampleModel;
import java.awt.image.DataBuffer;
import java.awt.image.DataBufferByte;
import java.awt.image.Raster;
import java.awt.image.WritableRaster;
import java.util.HashMap;
import java.util.Map;
import java.util.concurrent.atomic.AtomicBoolean;
import java.util.concurrent.atomic.AtomicReference;

import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import uk.co.caprica.vlcj.medialist.MediaListItem;
import uk.co.caprica.vlcj.player.MediaPlayer;
[... 420 lines cut ...]
		// convert memory to image

		Converter converter = Converter.getConverter(format.getWidth(), format.getHeight());
		BufferedImage image = converter.convert(buffers, format);
		imageRef.set(image);

		// calculate fps

		if (t1 == -1 || t2 == -1) {
			t1 = System.currentTimeMillis();
			t2 = System.currentTimeMillis();
		}

		t1 = t2;
		t2 = System.currentTimeMillis();

		fps.set((4 * fps.get() + 1000 / (t2 - t1 + 1)) / 5);
	}

	@Override
	public double getFPS() {
		return fps.get();
	}
}
```

sarxos, 2015-06-15:
@henriquem,

I will mark this issue as enhancement and either include RTSP support in existing VLCj driver, or create new one dedicated strictly for RTSP support (but the first option imho seems more elegant).

Of course all pull requests are really welcome here since I'm pretty busy at both work (projects, interviews, etc) and at home (my newborn child coming into this world in several days) and cannot donate so much time into the project as I did in the past,.

henriquem, 2015-06-17:
Thanks so much!!!!
Its works fine !!

sarxos, 2015-06-17:
@henriquem, please let me know if you find a way to disable audio in the stream. I've tried but didn't manage to fix this.

henriquem, 2015-06-18:
Hello @sarxos ,
Ok I will check about audio, and Congratulations for baby, send my greetings for baby hahaha
newborn is understandable the parents doesn't have time.
I only can do something when my son is sleeping! 

Look if you can help me, I forgot to told something wrong using rtsp+Motion Detection.
I keep the same implements for Motion Detection, when I used IP Cam, follow:

    detectorMoviment = new WebcamMotionDetector(Webcam.getWebcams().get(0));
    detectorMoviment.setInterval(500); // one check per 500 ms
    detectorMoviment.addMotionListener(new WebcamMotionListener(){
			@Override
			public void motionDetected(WebcamMotionEvent ev) {
                                //capture BufferedImage from RTSP and soon on 
			}
		});
    detectorMoviment.start(); //* In this point occur Exception

Exception at line detectorMoviment.start();

    Exception in thread "Thread-17" java.lang.NullPointerException
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:139)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:302)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:251)
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:236)
	at com.github.sarxos.webcam.WebcamMotionDetector.start(WebcamMotionDetector.java:233)

Something wrong happened at line 233, I'm using the version webcam-capture-0.3.11-snapshot.jar
It is easy to fix about RTSP  ?

dembogurski, 2017-02-08:
Hello I have a Chinesse camera Model HCN-M45L-P   and I don`t have the specification of this camera 
how a can find the end points of the camera, This only work in IE with the rtsp protocol 
with this code

<html>

<OBJECT classid="clsid:9BE31822-FDAD-461B-AD51-BE1D1C159921"
        codebase="http://downloads.videolan.org/pub/videolan/vlc/latest/win32/axvlc.cab"
        width="640" height="441" id="vlc" events="True">

    <param name="Src" value="rtsp://192.168.0.110" />
    <param name="ShowDisplay" value="True" />
    <param name="AutoLoop" value="False" />
    <param name="AutoPlay" value="True" />

    <embed id="vlcEmb"  type="application/x-google-vlc-plugin" quality="high" version="VideoLAN.VLCPlugin.2" autoplay="yes" loop="no" width="640" height="441"  target="rtsp://192.168.0.110" ></embed>
    
</html>   

but in java throws an exeption

  static {
        Webcam.setDriver(new IpCamDriver());
    } 
    public static void main(String[] args) throws MalformedURLException {
 
        //IpCamDevice ipCam = new IpCamDevice("sCamera", "http://192.168.0.110:6050/cgi-bin/guest/Video.cgi?media=jpeg", IpCamMode.PULL);
        //IpCamDevice ipCam = new IpCamDevice("sCamera", "http://192.168.0.110/cgi-bin/viewer/video.jpg?streamid=0", IpCamMode.PUSH);
        IpCamDevice ipCam = new IpCamDevice("sCamera", "http://admin:123456@192.168.0.110:80/video2.mjpg", IpCamMode.PULL);

-----------------------------

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot download image
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.getImagePullMode(IpCamDevice.java:389)

Thank you

sarxos, 2017-02-09:
Does you camera support MJPEG? If not then using IpCamDriver will give you
nothing.

For RTSP you have to use VlcjDriver.

dembogurski, 2017-02-09:
I don´t know if the camera suport or not because I only have this specs of the camera
http://www.netavr.com/uppic/20161122153833413341.pdf

For RTSP si suficient to use only this url?  rtsp://192.168.0.110

in the above example you have 
String rtsp = "rtsp://streaming1.osu.edu/media2/ufsap/ufsap.mov";

Can explain me what is the purposed of the above Class named VlcjDevice?   
How i can use it.

Than you very much.

sarxos, 2018-06-18:
Done in https://github.com/sarxos/webcam-capture/commit/a6a7a344854ebe7a6914af7c8bcfbf2abfbd040a

closed 2018-06-18 by sarxos

sarxos, 2018-06-18:
See https://github.com/sarxos/webcam-capture/issues/632 for more details.

nwthilina, 2019-07-16:
hi i have an uniview camera. I used ur code but it did not work. Can i have your sample code with required libraries.