sarxos/webcam-capture #357: JavaFX example hangs on Mac OS X Yosemite

defranke, 2015-06-21:
Hi,
when trying to launch the JavaFX example WebCamAppLauncher the programs hangs while searching for devices. I guess the problem is related to #318.

I get the following output:
```
17:39:07.222 [JavaFX Application Thread] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
17:39:07.229 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
```

and the output from jstack:
```
2015-06-21 17:41:57
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.40-b25 mixed mode):

"Attach Listener" #19 daemon prio=9 os_prio=31 tid=0x00007ff334aa1800 nid=0x7f17 waiting on condition [0x0000000000000000]
   java.lang.Thread.State: RUNNABLE

"atomic-processor-1" #17 daemon prio=5 os_prio=31 tid=0x00007ff335208800 nid=0xc403 runnable [0x00000001269f5000]
   java.lang.Thread.State: RUNNABLE
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices(Native Method)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$GetDevicesTask.handle(WebcamDefaultDriver.java:93)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

"webcam-discovery-service" #16 daemon prio=5 os_prio=31 tid=0x00007ff3351ff000 nid=0xc107 waiting on condition [0x00000001268f2000]
   java.lang.Thread.State: WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x0000000796c32fc0> (a java.util.concurrent.SynchronousQueue$TransferQueue)
	at java.util.concurrent.locks.LockSupport.park(LockSupport.java:175)
	at java.util.concurrent.SynchronousQueue$TransferQueue.awaitFulfill(SynchronousQueue.java:762)
	at java.util.concurrent.SynchronousQueue$TransferQueue.transfer(SynchronousQueue.java:695)
	at java.util.concurrent.SynchronousQueue.take(SynchronousQueue.java:924)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:70)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$GetDevicesTask.getDevices(WebcamDefaultDriver.java:79)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:124)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

"Thread-4" #15 daemon prio=5 os_prio=31 tid=0x00007ff3351ec800 nid=0x9103 runnable [0x0000000000000000]
   java.lang.Thread.State: RUNNABLE

"JavaFX Application Thread" #14 prio=5 os_prio=31 tid=0x00007ff3349e9000 nid=0x713 waiting on condition [0x00007fff57c14000]
   java.lang.Thread.State: TIMED_WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x0000000796ba5158> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.awaitNanos(AbstractQueuedSynchronizer.java:2078)
	at java.util.concurrent.ThreadPoolExecutor.awaitTermination(ThreadPoolExecutor.java:1465)
	at java.util.concurrent.Executors$DelegatedExecutorService.awaitTermination(Executors.java:675)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:113)
	- locked <0x0000000796b16f60> (a java.lang.Class for com.github.sarxos.webcam.Webcam)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:874)
	- locked <0x0000000796b16f60> (a java.lang.Class for com.github.sarxos.webcam.Webcam)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:847)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:826)
	at com.bazinga.WebCamAppLauncher.createTopPanel(WebCamAppLauncher.java:147)
	at com.bazinga.WebCamAppLauncher.start(WebCamAppLauncher.java:99)
	at com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$159(LauncherImpl.java:863)
	at com.sun.javafx.application.LauncherImpl$$Lambda$53/1742046688.run(Unknown Source)
	at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$172(PlatformImpl.java:326)
	at com.sun.javafx.application.PlatformImpl$$Lambda$49/362249184.run(Unknown Source)
	at com.sun.javafx.application.PlatformImpl.lambda$null$170(PlatformImpl.java:295)
[... 101 lines cut ...]
	at com.intellij.rt.execution.application.AppMain.main(AppMain.java:140)

"VM Thread" os_prio=31 tid=0x00007ff33481d800 nid=0x2703 runnable 

"GC task thread#0 (ParallelGC)" os_prio=31 tid=0x00007ff335800800 nid=0x1f03 runnable 

"GC task thread#1 (ParallelGC)" os_prio=31 tid=0x00007ff335801800 nid=0x2103 runnable 

"GC task thread#2 (ParallelGC)" os_prio=31 tid=0x00007ff335802000 nid=0x2303 runnable 

"GC task thread#3 (ParallelGC)" os_prio=31 tid=0x00007ff335802800 nid=0x2503 runnable 

"VM Periodic Task Thread" os_prio=31 tid=0x00007ff33583e800 nid=0x4903 waiting on condition 

JNI global references: 515
```

Searching for Webcams in a separate thread like this works without problems:
```java
new Thread(() -> {
    for(Webcam w : Webcam.getWebcams()) {
        System.out.println(w.getName());
    }
}).start();
```

Greg767, 2015-10-06:
Any solutions for this? I've tried to search for the webcams in a different thread, but its still freezing up.

sarxos, 2015-10-06:
@Greg767,

The issue is caused by the Apple limitation that allows some native code to run only on the main thread when using SWT, GWT or JavaFX.

Have you try starting your app with ```-XstartOnFirstThread``` JVM option? If this won't help you can also try w/a implemented by @phrack in his commit https://github.com/phrack/ShootOFF/commit/2144100b711083b3a3d401b17cbb702cef0706e5

Greg767, 2015-10-06:
Hey thanks for the quick reply. I just tried with -XstartOnFirstThread and it still hangs. I will look into that post.
By the was does it suppose to work with the built-in cam as well or it must have a USB cam?

sarxos, 2015-10-06:
@Greg767,

In regards to:

>  By the way does it suppose to work with the built-in cam as well or it must have a USB cam?

Yes. It support both build-in and UVC :)

Greg767, 2015-10-07:
OK so finally what worked for me is to switch to my Macbook to finalize the development and get the default cam in a Task, as per the code exemple. I didn't need to use -XstartOnFirstThread option.
Thanks a lot.

hendraanggrian, 2016-06-03:
I am having a similar issue and I can confirm that putting `-XstartOnFirstThread` did not solve the problem.

@Greg767 can you please elaborate your solution?

Greg767, 2016-06-03:
For me it worked without that property on a Mac in netbeans and I used this code:
`protected void initializeWebCam(final int webCamIndex) {
        takeImageBtn.setDisable(true);
        webcamDisposed = false;

        Task<Void> webCamTask = new Task<Void>() {

            @Override
            protected Void call() throws Exception {

                if (webCam != null) {
                    disposeWebCamCamera();
                }

                for (Webcam w : Webcam.getWebcams()) {
                    if(!w.getName().contains("FaceTime"))
                        webCam = w;
                }
                if(webCam == null)
                    webCam = Webcam.getDefault();
                Dimension[] sizes = webCam.getViewSizes();
                webCam.setViewSize(sizes[sizes.length - 1]);
                webCam.open();

                startWebCamStream();

                return null;
            }
        };

        Thread webCamThread = new Thread(webCamTask);
        webCamThread.setDaemon(true);
        webCamThread.start();

    }`

hendraanggrian, 2016-06-05:
@Greg767 thanks for sharing the code. I think the clear solution here is to run `webcam.getWebcams()` on a separate thread. Webcam on my Mac works now although I'm still figuring out why `webcam.close()` throws NullPointerException.

sarxos, 2016-06-06:
Hi @hendraanggrian,

Can you please elaborate more about this ```NullPointerException``` in ```close()```? Especially, can you provide stack trace?

hendraanggrian, 2016-06-06:
@sarxos I weirdly cannot reproduce the error, at this point everything works great with your library. Thanks!

sarxos, 2016-06-07:
Cool, in case it manifest again please do not hesitate to report it :)

james-d, 2017-03-16:
I found that in general the FX Application Thread is prone to hanging if you interact with the web cam much at all on that thread. E.g. `Webcam.close()` or `Webcam.getWebcams()` seem to cause deadlock. (Mac OS X Sierra on a Macbook Pro with built-in camera). I haven't tested much (honestly, I don't actually use the library, I was just answering a [StackOverflow question](http://stackoverflow.com/questions/42816859/run-javafx-and-swing-application-at-the-same-time)...), but I created a couple of wrapper classes that wrap the `Webcam` as a JavaFX service and a corresponding view. Not intended to be production quality, but if you are interested in adapting that as an example, feel free (just acknowledge, please). Gist at https://gist.github.com/james-d/f826c9f38d53628114124a56fb7c4557

sarxos, 2017-03-16:
Hi @james-d,

Thank you for this contribution. Indeed many people from Webcam Capture community reported that using this API on Mac is somehow tricky.

In regards to your code - since there are lambdas in your code I had to update ```pom.xml``` to specify Java source and target compatibility:

```xml
<plugin>
  <artifactId>maven-compiler-plugin</artifactId>
  <version>3.3</version>
  <configuration>
    <source>1.8</source>
    <target>1.8</target>
  </configuration>
</plugin>
```

Otherwise it gave me compilation error:

```plain
$ mvn clean install
[INFO] Scanning for projects...
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building WebCamTest 0.0.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ WebCamTest ---
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ WebCamTest ---
[WARNING] Using platform encoding (UTF-8 actually) to copy filtered resources, i.e. build is platform dependent!
[INFO] skip non existing resourceDirectory /home/sarxos/workspace/webcam-capture/webcam-capture-examples/webcam-capture-javafx-service/src/main/resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.3:compile (default-compile) @ WebCamTest ---
[INFO] Changes detected - recompiling the module!
[WARNING] File encoding has not been set, using platform encoding UTF-8, i.e. build is platform dependent!
[INFO] Compiling 3 source files to /home/sarxos/workspace/webcam-capture/webcam-capture-examples/webcam-capture-javafx-service/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /home/sarxos/workspace/webcam-capture/webcam-capture-examples/webcam-capture-javafx-service/src/main/java/WebCamView.java:[25,95] lambda expressions are not supported in -source 1.5
  (use -source 8 or higher to enable lambda expressions)
[ERROR] /home/sarxos/workspace/webcam-capture/webcam-capture-examples/webcam-capture-javafx-service/src/main/java/FXCamTest.java:[34,41] lambda expressions are not supported in -source 1.5
  (use -source 8 or higher to enable lambda expressions)
[INFO] 2 errors 
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 1.299 s
[INFO] Finished at: 2017-03-16T09:27:57+01:00
[INFO] Final Memory: 12M/291M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.3:compile (default-compile) on project WebCamTest: Compilation failure: Compilation failure:
[ERROR] /home/sarxos/workspace/webcam-capture/webcam-capture-examples/webcam-capture-javafx-service/src/main/java/WebCamView.java:[25,95] lambda expressions are not supported in -source 1.5
[ERROR] (use -source 8 or higher to enable lambda expressions)
[ERROR] /home/sarxos/workspace/webcam-capture/webcam-capture-examples/webcam-capture-javafx-service/src/main/java/FXCamTest.java:[34,41] lambda expressions are not supported in -source 1.5
[ERROR] (use -source 8 or higher to enable lambda expressions)
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
```

sarxos, 2017-03-16:
@james-d, I included your code as a new example. Feel free to modify it whenever you need.

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-javafx-service

Thanks!