sarxos/webcam-capture #358: Using Maven and webcam-capture

karijuana, 2015-06-24:
I haven't used maven in a very, very long time and had little clue of what I was even doing then. Is there a quick rundown anyone could give on managing the webcam-capture dependencies with Maven? I thought it wouldn't be a problem until I realized how many dependencies there actually are, and some of them are ancient ones that are not longer available by google search like jp.co.swiftinc.relax

closed 2015-06-26 by karijuana

sarxos, 2015-06-29:
Hi @karijuana,

It's enough to add this dependency to your ```pom.xml``` and Maven will handle whole stuff for you.

```xml
<dependency>
  <groupId>com.github.sarxos</groupId>
  <artifactId>webcam-capture</artifactId>
  <version>0.3.10</version>
</dependency>
```

However, you mentioned that ```webcam-capture``` JAR has many dependencies, but it really has only few ones:
- BridJ (one JAR, managed by Maven),
- SLF4J (also one JAR, managed by Maven).

Optionally you can add Logback (which are two JARs, also managed by Maven).

There are other drivers which has other dependencies, but I'm not aware of ```jp.co.swiftinc.relaxv``` being used in neither Webcam Capture API core nor any webcam driver (please correct me if I'm wrong). Maybe some examples are using it, but these can be ignored since in terms of dependencies only core module and drivers really matters.

sarxos, 2015-06-29:
Just FYI, I'm strongly recommend using Maven in all of the Java projects :) Not only Webcam Capture API. It is really a lifesaver when it comes to the dependencies management!