sarxos/webcam-capture #361: Control over camera focus, exposure, and other parameters

lessthanoptimal, 2015-06-25:
There doesn't appear to be anyway to control the cameras focus and exposure.  Native libraries such as v4l and libuvc provide access to those parameters as well as others, e.g. zoom, contrast, brightness, and others.  For computer vision applications control over focus is essential since often the auto focus is on crack and wants to focus on anything but what you want it to and constantly changes its mind.  It's worth nothing that not all webcams allow you to control those parameters, but many do.

sarxos, 2015-06-27:
Hi @lessthanoptimal,

Indeed, the Webcam Capture API does not expose such features to be controlled. This is because it acts as a very basic and unified API between the developer and different capturing frameworks (v4l, gstreamer, openimaj, jmf, vlc, opencv, etc) from which some may actually support tilt, zoom, pan, etc features, but most unfortunately does not provide such capabilities.

I thought about adding such feature in the past, but decided not to implement it due to various reasons (mostly due to lack of hardware that can be used while development).

I will keep this issue open and mark it as enhancement. Hopefully will come back to this when my newborn son gives me a little more free time to spare :)

lessthanoptimal, 2015-06-29:
Do you have a logitech webcam?  Most of them allow manual control of focus, or at least the ones I've been using.  Logitech HD Pro Webcam C920 is a well supported, allows control over focus, and produces clear sharp images.

For an API I would suggest the following functions:

void setAutoFocus(  boolean active ) throws NotSupportedException;
double getFocus();
void setFocus( double value );

The value would be from 0 to 1.  The actual hardware/API would only support some arbitrary scale from min to max value.  You can hide that scale from the user easily.

Congratulations on your newborn!  Hopefully he becomes self sufficient soon so you can get work on this ;-)

sarxos, 2015-06-30:
I have few UVC devices, will check if I can at least control focus.

In terms of API, I was rather thinking about the more abstract approach, i.e. to have ```WebcamControl``` interface such as:

```java
public interface WebcamControl {
  String getName();
  double getValue();
  void setValue(double value);
  // some other methods, depending on the needs
}
```

Some enum inside to keep predefined control names, like:

```java
public static enum WebcamControlType {
  FOCUS("focus"),
  ZOOM("zoom"),
  TILT("tilt"),
  PAN("pan");
  // etc
  public String getName() {
    return name;
  }
}
```

And some useful methods in ```Webcam``` class such as:

```java
public boolean isSupported(String ctrl);
public boolean isSupported(WebcamControlType ctrl);
public WebcamControl getControl(String ctrl);
public WebcamControl getControl(WebcamControlType ctrl);
```

With such approach people could write their own controls if the predefined ones are not enough for their needs. E.g. many IP cameras has its own API to control tilt/pan/zoom and there is no unified interface to access these properties, so I can imagine devs writing own ```WebcamControl``` impl to parametrize these differently in e.g. Foscam or Marmitek.