sarxos/webcam-capture #362: Webcam does not properly set custom view size

sepeterson, 2015-07-02:
I have a 1080p webcam connected via USB port. I can capture images from it just fine, but for some reason, the highest resolution it will capture images at is 640 x 480.

Here is the code I used to test it (a slightly modified version of the nonstandard resolution example):

		ArrayList<Webcam> cameraList = new ArrayList(Webcam.getWebcams());

		Dimension[] nonStandardResolutions = new Dimension[] {
			WebcamResolution.PAL.getSize(),
			WebcamResolution.HD720.getSize(),
			new Dimension(2000, 1000),
			new Dimension(1000, 500),
		};
		
		Webcam webcam = cameraList.get(1);
		webcam.setCustomViewSizes(nonStandardResolutions);
		webcam.setViewSize(WebcamResolution.HD720.getSize());
		webcam.open();

		BufferedImage image = webcam.getImage();
		
		System.out.println(webcam.toString());
		for(Dimension d: webcam.getCustomViewSizes()){
		System.out.println(d.toString());
		}
		System.out.println(image.getWidth() + "x" + image.getHeight());
	
The above code will print out:

		Webcam Microsoft LifeCam Studio 1
		java.awt.Dimension[width=768,height=576]
		java.awt.Dimension[width=1280,height=720]
		java.awt.Dimension[width=2000,height=1000]
		java.awt.Dimension[width=1000,height=500]
		640x480

I have tried this code with the resolution set to both 1280 x 720 and 1920 x 1080 and in both instances, 480p images were captured. Additionally, when using the Windows 8 "Camera" utility, it captures 1080p images just fine.

This code works fine when webcam is set to Webcam.getDefault(), the laptop's built-in webcam.

Thanks!

sarxos, 2015-07-03:
Hi @sepeterson,

The image of custom view size is not guaranteed to be returned when camera does not support it. Please take a look at javadoc for ```setCustomViewSizes(..)```, it says:

> Set custom resolution. If you are using this method you have to make sure that your webcam device can support this specific resolution.

On Windows (and only Windows), when your camera is not able to support custom resolution you've set, the library used by Webcam Capture on Windows ([videoInput](https://github.com/ofTheo/videoInput) by Theodore Watson) approximate the closest possible resolution which are valid for given webcam hardware, and set it instead of the one you choose. Unfortunately there is no information about this change, no callback, or anything, so Webcam Capture API on Java side is unable to react to such a change. The only information is the warning log being printed. I thought this is much better solution than throwing exception when requested image size does not match the one which is returned.

The reason of why Windows 8 is capturing the image in 1080p and Webcam Capture API is not, is probably the fact that this specific resolution comes with h.264 frames format which Windows utility can read natively (same as e.g. Skype), and Webcam Capture API is not able to read it because videoInput does not support such format.

eduramiba, 2021-06-17:
Hi,

Please try the new native driver that I just released: https://github.com/eduramiba/webcam-capture-driver-native

It should be able to correctly list all device resolutions.