sarxos/webcam-capture #363: BufferedImage method is not found in Framegrabber class

vijay0850, 2015-07-07:
I'm try with JavaCv So I created the needed class from our code in github 

For JavaCvDevice.Java I'm facing the following issue,

1. method getBufferedImage() cannot found in Frame class

I tried to convert the frame to buffer it also not working it shows 

2, incompatible types: Frame cannot be converted to BufferedImage

Code:-

public BufferedImage getImage() {

		if (!open) {
			throw new WebcamException("Cannot grab image - webcam device is not open");
		}

		try {
			return grabber.grab().getBufferedImage();
                                       
                       /* Buffer  = grabber.grab();

                            return Buffer; */
                    
		} catch (Exception e) {
			throw new WebcamException("Cannot grab image...");
		}

Due to this issue I'm not able to get the image from webcam So please help me to fix this thing.

Thanks in Advance

sarxos, 2015-07-07:
Hi, I guess they modified the API and removed this method or you are using
invalid dependency. Did you use Maven, or imported JARs manually? I
currently have very limited internet access and cannot look at this now,
but will come back to this (hopefully on weekend). Take care!

sarxos, 2015-07-13:
@vijay0850,

I see. you are using ```javacv-1.0``` and the code I was basing on was ```javacv-0.8```. I suppose I should migrate to newer version already. I will let you know when the code is ready. For the time being there is no other solution than downgrading dependency to 0.8 and use it together with existing solution.

closed 2015-07-13 by sarxos

sarxos, 2015-07-13:
Hi @vijay0850,

The code is ready and available in SNAPSHOT version from Sonatype repo (please make sure to make use of the newest JAR).

https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture-driver-javacv/0.3.11-SNAPSHOT/

commit 802311ff (2015-07-13): Upgrade JavaCV from 0.8 to 1.0, fixes #363
commit d50521dd (2015-07-13): Upgrade JavaCV from 0.8 to 1.0, fixes #363