sarxos/webcam-capture #366: Error web cam in jdk 1.6

abdulrosyid, 2015-07-13:
Java Plug-in 1.6.0
Using JRE version 1.6.0 Java HotSpot(TM) Client VM
User home directory = C:\Documents and Settings\tri.sunu.wad
----------------------------------------------------
c:   clear console window
f:   finalize objects on finalization queue
g:   garbage collect
h:   display this help message
l:   dump classloader list
m:   print memory usage
o:   trigger logging
p:   reload proxy configuration
q:   hide console
r:   reload policy configuration
s:   dump system and deployment properties
t:   dump thread list
v:   dump thread stack
x:   clear classloader cache
0-5: set trace level to <n>
----------------------------------------------------
Exception in thread webcam-updater-thread-1
java.lang.IllegalMonitorStateException
	at java.util.concurrent.locks.ReentrantLock$Sync.tryRelease(Unknown Source)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer.release(Unknown Source)
	at java.util.concurrent.locks.ReentrantLock.unlock(Unknown Source)
	at java.util.concurrent.LinkedBlockingQueue.take(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.getTask(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Exception in thread atomic-processor-1
java.lang.IllegalMonitorStateException
	at java.util.concurrent.locks.ReentrantLock$Sync.tryRelease(Unknown Source)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer.release(Unknown Source)
	at java.util.concurrent.locks.ReentrantLock.unlock(Unknown Source)
	at java.util.concurrent.LinkedBlockingQueue.take(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.getTask(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

i'm run applet in jdk 1.7 its work but if i'm run in jdk 1.6 cannot work, can u help me ?

sarxos, 2015-07-13:
Hi @abdulrosyid,

Have you added:

```xml
<param name="separate_jvm" value="true" />
```

Into you ```<applet>``` tag? Please see this example:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-applet/src/main/html/index.html

The issue is caused by the fact that applet runtime has already shutdown executor services (this is done automatically by jvm), and it tries to use them again after the restart.

Please reopen the ticket if the above flag does not help resolving described issue.

abdulrosyid, 2015-07-13:
I already added but still error

sarxos, 2015-07-14:
Can you prepare some example I can run to reproduce the issue?

sarxos, 2015-07-20:
Hi @abdulrosyid, can you please prepare small example which will demonstrate the issue? The source code can be uploaded to github or any other service.

abdulrosyid, 2015-07-23:
u can see my code in https://github.com/abdulrosyid/webcam/blob/master/appMain.java,
if i try upgrade my jre from 1.6 to 1.7 no error but if i try run in 1.6 have error

sarxos, 2015-07-23:
@abdulrosyid, in your source code I do not see ```Applet.destroy()```. Please see:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-applet/src/main/java/WebcamAppletExample.java

In this example ```destroy()``` closes webcam and invokes ```WebcamCaptureFrameworkDestroyer.destroy()``` which invoked ```Webcam.shutdown()```. The last one is protected so this is why  ```WebcamCaptureFrameworkDestroyer``` is put in the ```com.github.sarxos.webcam``` package.

Have you tried doing similar thing in your code?

abdulrosyid, 2015-07-23:
yes i have tried to do that, but still error just in jre 1.6

sarxos, 2015-07-23:
Just FYI, I can take a look at this on weekend, but are you aware that Oracle will not support 1.6 any more? Public updates for Java 1.6 stopped at February 2013. Please note that even Java 1.7 is pretty old now with 1.8 being in the wild for some time already. The end date of public updates for Java 1.7 was at April 2015 (source: http://www.oracle.com/technetwork/java/javase/eol-135779.html).

Therefore, it's my opinion, investigating Java 1.6 issues nowadays, is a waste of time if this is fixed in 1.7 release (which is pretty old now as well).