sarxos/webcam-capture #368: launch video frame from a JPanel

mhamed306, 2015-07-16:
Hi,

First, I want to congratulate you for the API !

I am planning to use your webcam-capture API for a specific usecase : 

I have a JPanel that offers users the possibility to load a profile image from disk. When done, the image is displayed on the JPanel.
I want it to offer also the possibility to display live stream video from a connected webcam, so I added a JButton that calls your WebcamPanelExample() class to make a primary test.
Unfortunatly, the frame stays on backgroud (which is normal), when I changed the WebcamPanelExample() class to extends from a JDialog instead of a JFrame, I cannot have the live stream video anymore !!! I think it is a matter of modality coming from the JDialog.

Can you give me a simple way to fix this problem ? I mean having the live stream video window as modal so I can see the video and take a capture to save it in a image for example...

Thanks a lot.

sarxos, 2015-07-16:
Hi @mhamed306, can you please post a simple code sample demonstrating the issue you've described?

mhamed306, 2015-07-17:
Hi @sarxos ,

I prepared for you a little quick&dirty project where I test your API, sorry...

Can you send me an email to mhamed306@hotmail.com so I can send you the project ?

Thanks for all ;)

sarxos, 2015-07-17:
@mhamed306, the email has been sent.

mhamed306, 2015-07-17:
@sarxos ,

thanks for your quick responding, but I found what was the problem ! I just changed some little things : 

1- Change the WebcamViewerExample() class to extend from a JPanel instead of JFrame
2- Create a JDialog (to have a modal window) and add to it the WebcamViewerExample() object
3- Call the JDialog object into a thread (this is what allowed me to display the camera video stream)

Thanks for all ;)

sarxos, 2015-07-17:
Great :) I'm glad it works. The Swing can be tricky sometimes. Take care!

closed 2015-07-17 by sarxos