sarxos/webcam-capture #369: error webcam java.lang.reflect.InvocationTargetException

abdulrosyid, 2015-07-23:
java.lang.reflect.InvocationTargetException
	at java.awt.EventQueue.invokeAndWait(Unknown Source)
	at java.awt.EventQueue.invokeAndWait(Unknown Source)
	at webcamsdd.appMain.init(appMain.java:78)
	at com.sun.deploy.uitoolkit.impl.awt.AWTAppletAdapter.init(Unknown Source)
	at sun.plugin2.applet.Plugin2Manager$AppletExecutionRunnable.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.ExceptionInInitializerError
	at com.github.sarxos.webcam.Webcam.getDriver(Unknown Source)
	at com.github.sarxos.webcam.Webcam.getDiscoveryService(Unknown Source)
	at com.github.sarxos.webcam.Webcam.getWebcams(Unknown Source)
	at com.github.sarxos.webcam.Webcam.getDefault(Unknown Source)
	at com.github.sarxos.webcam.Webcam.getDefault(Unknown Source)
	at com.github.sarxos.webcam.Webcam.getDefault(Unknown Source)
	at webcamsdd.appMain$1.run(appMain.java:84)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$200(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.security.AccessControlException: access denied ("java.util.PropertyPermission" "webcam.debug" "read")
	at java.security.AccessControlContext.checkPermission(Unknown Source)
	at java.security.AccessController.checkPermission(Unknown Source)
	at java.lang.SecurityManager.checkPermission(Unknown Source)
	at sun.plugin2.applet.AWTAppletSecurityManager.checkPermission(Unknown Source)
	at java.lang.SecurityManager.checkPropertyAccess(Unknown Source)
	at java.lang.System.getProperty(Unknown Source)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.<clinit>(Unknown Source)

There is no error when running it on my desktop, but when I run it on my webpage this is all I get.

can u help me to solve this error

sarxos, 2015-07-23:
@abdulrosyid, have you signed your applet?

abdulrosyid, 2015-07-23:
how to signed my applet ? i use editor netbeans.

sarxos, 2015-07-23:
You can do that from command line by using ```jarsigner``` tool (there are thousands of examples in google) or use Maven (as I did) to bind ```maven-jarsigner-plugin``` to your Maven build. The Maven POM to build my applet example (if you decide to use it) can be found here:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-applet/pom.xml

If this is the first time you hear about project called Maven please read more about it because the first approach may be overwhelming.

Generally, you may copy my project and put your sources inside. Netbeans supports Maven very well, but since this is not my IDE of choice (I use Eclipse) I cannot give you details on how to integrate Maven in your Netbeans project. From the console it's enough if you have ```pom.xml``` in a directory and you run ```mvn clean package``` command there.

abdulrosyid, 2015-07-23:
okay thx's sarxos, i want to ask again about applet, why must signed applet ?

sarxos, 2015-07-23:
@abdulrosyid, this is because you have to prove you are you and not some random hacker from the depths of Internet. Without applets signing everyone could do literally everything on a remote computer whenever victim access dedicated web page which is a host of the malicious applet. This is also the reasons why Sun/Oracle dropped self-signed applets and they do not intend to support it any more without a valid RSA signature which can be purchased from a certificate vendors. The price for such signature varies between several bucks up to the hundreds per year. There are some really cheep ones, but they may work only on some very limited set of platforms, so please consider yourself warned when you decide to purchase such certificate. Also, please note that code signing certificate is something different than regular SSL/TLS certificate that are used on a web page.

For further reading please see:

http://mrbool.com/what-is-self-signed-applet/28604
http://docs.oracle.com/javase/7/docs/technotes/guides/jweb/security/rsa_signing.html

abdulrosyid, 2015-07-24:
thx's sarxos for your explain.

closed 2015-08-07 by sarxos