sarxos/webcam-capture #370: Warnings and error

gpotter2, 2015-07-23:
Hey !

First, thanks a lot, this API is just GREAT !!!!!!

Then, on linux (i'm using ubuntu) the bridj lib says that you must do:
"execstack -c" on the bridj.so files..... (i did it and i've got now this: http://www.filedropper.com/bridj-062)

Finally, while doing this, on linux only (it was going well on windows):
Webcam webcam = Webcam.getDefault();

I got the message:
Not enough args for null OpenIMAJGrabber.startSession(int, int, double)

Thanks a lot for correcting it, this is a great project !!!!!

sarxos, 2015-07-23:
Hi @gpotter2,

Both this messages are not errors but simple warnings and comes from BridJ.

1. The _Not enough args_ message has been fixed and removed in https://github.com/nativelibs4java/nativelibs4java/issues/328 (BridJ 0.6.3)
2. The _execstack -c_ message has been removed in https://github.com/nativelibs4java/nativelibs4java/issues/468 (BridJ 0.7)

Webcam Capture API still uses older 0.6.2 because newer BridJ deprecated Raspberry Pi support. If you want to get rid of these warnings just upgrade BridJ to newer one but please note that you may experience issues if you are on Raspberry Pi.

gpotter2, 2015-07-23:
Ok, thanks :)

closed 2015-07-23 by gpotter2